espressif/esp-idf #15485: Setting minimal build on or COMPONENT main causes the build to not include the component test sub-directory

i400s, 2025-02-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2119-g0461e2ff88

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The build system should auto-include a components test directory and the c files within.

### What is the actual behavior?

Build system doesn't include components test directory or doesn't find the c files within.

### Steps to reproduce.

1. Create a new project from the unit_test example.
2. Compile/flash/monitor the device.
3. No tests are run: 0 tests 0 failures 0 ignored.
4. Comment out the idf_build_set_property(MINIMAL_BUILD ON) line.
5. Compile/flash/monitor the device.
6. Tests are run: 3 Tests 1 Failures 0 Ignored 
...

### Build or installation Logs.

```plain

```

### More Information.

The example program contains the minimal requirements to show the error and the workaround of not trimming the build.

My actual application/components code can't be compiled using an untrimmed build as a workaround because it is using some mocks and this causes the code to fail to compile with an error because not all drivers are mocked.

Another workaround using the example program:
It is possible to copy the test_mean.c from the components/testable/test/ directory to the application test/main/ and amend the CMakeLists.txt file to add the file in the SRCS section, add testable to the PRIV_REQUIRES section, but only if WHOLE_ARCHIVE is added. If WHOLE_ARCHIVE is not included then it again fails to pick up the tests (0 tests 0 failures 0 Ignored).

There is another warning/error if changing from a physical device to "linux" which is partially related to this issue but I will raise that in a separate bug report after a few more tests.

fhrbata, 2025-02-28:
Hello @i400s,

thank you very much for highlighting the issue with the minimal build and test components. You are correct that when the minimal build is enabled, the components specified with `set(TEST_COMPONENTS ...)` are not included in the build.
The following patch seems to fix the issue.
```diff
diff --git a/tools/cmake/project.cmake b/tools/cmake/project.cmake
index cfb95b95cec2..a8f3cea063fa 100644
--- a/tools/cmake/project.cmake
+++ b/tools/cmake/project.cmake
@@ -504,7 +504,7 @@ function(__project_init components_var test_components_var)
             message(WARNING "The MINIMAL_BUILD property is disregarded because the COMPONENTS variable is defined.")
             set(minimal_build OFF)
         else()
-            set(COMPONENTS main)
+            set(COMPONENTS main ${TEST_COMPONENTS})
             set(minimal_build ON)
         endif()
     else()

```

i400s, 2025-02-28:
Thank you. Will apply the change manually to my clone as an interim solution.

fhrbata, 2025-02-28:
Hello @i400s, thank you very much and apologies for the inconvenience. I've already submitted an internal merge request to address this. Let's see what others think. Please let me know if you encounter any other issues related to this.

i400s, 2025-02-28:
Have just tested. 

idf_build_set_property(MINIMAL_BUILD ON) now works as expected and includes the code under the component_name/test/ directory.

However it doesn't work if set(COMPONENTS main) is specified (which I think was the old way of creating a minimal build).

Using the code you showed, if I amend my project CMakefile.txt to have set(COMPONENTS main ${TEST_COMPONENTS}) it works. I'm guessing because although part of the esp_idf code knows about TEST_COMPONENTS the part that "builds" doesn't if its not included in the COMPONENTS list.

Now I understand the process a bit more an obvious fix would be to always append to COMPONENTS if TEST_COMPONENTS has a list (as apposed to the proposed change within the conditional, line 504)... but I'm guessing people who may have found this issue with the old method (and understand CMake) may have made the change I did to their project CMakefile... so I'm not sure what the implications of appending the TEST_COMPONENTS would make if the COMPONENTS already has the same items. eg. (main a_component mytest_component a_component mytest_component).

I'm still very new at this (c and cmake), so I will leave this in your capable hands.

fhrbata, 2025-03-01:
Hello @i400s ,

> idf_build_set_property(MINIMAL_BUILD ON) now works as expected and includes the code under the component_name/test/ directory.

Great, thank you for confirming.

> However it doesn't work if set(COMPONENTS main) is specified (which I think was the old way of creating a minimal build).

Setting the COMPONENTS variable is still a valid method. The MINIMAL_BUILD property is just a shortcut for configuring the COMPONENTS to include the main component only, and with the recent fix, it also includes TEST_COMPONENTS.

> Using the code you showed, if I amend my project CMakefile.txt to have set(COMPONENTS main ${TEST_COMPONENTS}) it works. I'm guessing because although part of the esp_idf code knows about TEST_COMPONENTS the part that "builds" doesn't if its not included in the COMPONENTS list.

Yes, you got it right. Here's how it currently works: During the build configuration, all available components, including test components, are added to a list of components to be built. This includes components that might not be necessary for your project. This approach was intentionally chosen to make it easier for users to transition from the previous build system version without worrying about component dependencies. To avoid building unnecessary components and to speed up the process, you can set the COMPONENTS variable to specify which components should be explicitly included in the build, rather than including all found components. When set, the list of components involved in the build is [trimmed](https://github.com/espressif/esp-idf/blob/master/tools/cmake/project.cmake#L522) to those listed in the COMPONENTS variable and all their direct or indirect dependencies, which may significantly speeding up the build. The issue with TEST_COMPONENTS is that they are only added to the build if they are also [included](https://github.com/espressif/esp-idf/blob/master/tools/cmake/project.cmake#L540) in the list of components to be built. This is true only if COMPONENTS  variable is not set at all, or if the TEST_COMPONENTS are also included in COMPONENTS list. 

> Now I understand the process a bit more an obvious fix would be to always append to COMPONENTS if TEST_COMPONENTS has a list (as apposed to the proposed change within the conditional, line 504)... [...]

You're right, and I was considering this as well. The issue might be that using the COMPONENTS in this way has been part of the build system for a long time, and as you mentioned, people might depend on this behavior. I can imagine having the TEST_COMPONENTS set, but modifying what gets built with the COMPONENTS variable. Honestly, I'm not entirely certain, but my thought was to avoid altering the existing behavior. The minimal build is relatively new and only available on the master branch, and it doesn't offer the same flexibility as setting the COMPONENTS variable.

> I'm still very new at this (c and cmake), so I will leave this in your capable hands.

I'm not so sure about my 'capable' hands, but I'll give it my best shot 😃 . Thank you for bringing the issues you've faced to our attention.

closed 2025-03-06 by espressif-bot