espressif/esp-idf #15486: Permission Control functionalities esp32s3

plamendc, 2025-02-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.7

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32s3

### Power Supply used.

USB

### What is the expected behavior?

I am trying to implement a memory protection functionally for my project. I have several partitions: SSBL, factory and application. During normal operation the code first goes into SSBL than boots the factory/application.

### What is the actual behavior?

It works like that, but when I try to implement the PMS functionality to select partitions and set access control for each of them, I can't boot any partition following SSBL. The memory protection algorithm is located in my SSBL.

### Steps to reproduce.

This is my SSBL code:

/*
 * We arrive here after the ROM bootloader finished loading this second stage bootloader from flash.
 * The hardware is mostly uninitialized, flash cache is down and the app CPU is in reset.
 * We do have a stack, so we can do the initialization in C.
 */
void __attribute__((noreturn)) call_start_cpu0(void)
{
    tReprogFlag* pReprogFlag = (tReprogFlag*) REPROGRAMMING_STRUCT_BASE_ADDR;
    esp_image_metadata_t img_meta;
    uint32_t calculatedCrcValue = 0;
    bool noMainApp = false;
    bool verifiedFAC  = false;
    bool verifiedOTA[2] = {false, false};

    /* 1. Hardware initialization */
    if (bootloader_init() != ESP_OK) {
        bootloader_reset();
    }

#ifdef CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP
    // If this boot is a wake up from the deep sleep then go to the short way,
    // try to load the application which worked before deep sleep.
    // It skips a lot of checks due to it was done before (while first boot).
    bootloader_utility_load_boot_image_from_deep_sleep();
[... 340 lines cut ...]
E (398) PCU: read FLASH ACE ATTR REG: FF
E (402) PCU: FLASH ACE ADDR REG: 60026044, start address: 620000
E (409) PCU: FLASH ACE SIZE REG: 60026054, size: F
E (414) PCU: FLASH ACE ATTR REG: 60026034, attr: FF
E (420) PCU: read FLASH ACE ADDR REG: 620000
E (425) PCU: read FLASH ACE SIZE REG: F
E (429) PCU: read FLASH ACE ATTR REG: FF
E (434) PCU: Flash memory regions configured with full access for Secure, Non-Secure, and SPI1.
I (443) SSBL: Try to load image: -1 (-1: factory | 0/1: main app)
I (450) esp_image: segment 0: paddr=00610020 vaddr=3c070020 size=2a618h (173592) map
E (480) esp_image: invalid segment length 0xfeebdaed
E (481) boot: Factory app partition is not bootable
I (481) esp_image: segment 0: paddr=00010020 vaddr=3c1b0020 size=7ffe8h (524264) map
E (549) esp_image: invalid segment length 0xfeebdaed
E (549) boot: OTA app partition slot 0 is not bootable
E (549) esp_image: image at 0x310000 has invalid magic byte (nothing flashed here?)
E (557) boot: OTA app partition slot 1 is not bootable
E (563) boot: No bootable app partitions in the partition table
```

### More Information.

I followed the specifications specified in section 15.5 in this document: https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf

The whole idea was to implement memory protection that while I am in one partition in the code, to disable unwanted access to the other partitions. Perhaps, I am doing something wrong during the configuration of the PMS or perhaps, this functionality is not applicable for my use case.

sachin-billore, 2025-03-05:
Hi @plamendc ,

1.	Can you confirm that start address and size of the all the flash partitions are 64Kb aligned? From the code snippet it looks like SSBL partition size is not (it’s 48KB).
2.	All the partition size defined are in numbers (SSBL_SIZE = 1, ….) and same is configured in FLASH_ACE_n_SIZE register. Can you please confirm it? If this is the case, it will be invalid configuration.
3.	Are you configuring all the valid permission for Secure World? I think entire code will be executing in secure world, it is not switching to Non-secure world.
4.	Are you configuring and enabling the interrupt for access violation for external memory (Section 15.6.3 in TRM)? If not, can you read control and addr registers (0x88 and 0x8c), to find the location and type of access violation?

georgi-pachov, 2025-03-24:
Hi, I will be taking over this problem.
@sachin-billore The size of the SSBL is indeed not 64kB. The 64kB section ends on a reserved space in the partition table. 
Is it a problem that we protect the partition data and the OTA data?

[image]

sachin-billore, 2025-03-26:
Hi @georgi-pachov, It will not be a problem to protect the partition data and OTA data as long as the application which is accessing them is executing in the right privilege level (Secure World or Non-Secure World) as per the configuration in FLASH_ACE_ATTR.
If you could explain how this application trying to use the flash PMS setting then I may explain you in the right context.

plamendc, 2025-03-30:
Hi @sachin-billore 

**Can you confirm that start address and size of the all the flash partitions are 64Kb aligned? From the code snippet it looks like the SSBL partition size is not (it’s 48KB).**
Yeah, currently I have two partitions SSBL and PROD that are not of a size 64Kb. I have read in the documentation that the size has to be aligned to 64kB. Does this also apply for their starting addresses?

**All the partition size defined are in numbers (SSBL_SIZE = 1, ….) and same is configured in FLASH_ACE_n_SIZE register. Can you please confirm it? If this is the case, it will be invalid configuration.**
Currently, in the FLASH_ACE_n_SIZE register I set the variable that is the multiples of 64Kb of the corresponding partition. In other words in the case of:
#define OTA_0_SIZE 48 // (3072 * 1024) // 3072KB
- 48 * 64Kb = 3072Kb

**Are you configuring all the valid permission for Secure World? I think entire code will be executing in secure world, it is not switching to Non-secure world.**
I haven't configured the secure world functionality, therefore, it is operating in default mode. However, I have configured the access criteria for both modes to full access.

**Are you configuring and enabling the interrupt for access violation for external memory (Section 15.6.3 in TRM)? If not, can you read control and addr registers (0x88 and 0x8c), to find the location and type of access violation?**
The interrupt for access violation for external memory is not configured or enabled/disabled by the code, so it should be in the default state.  I did add reading the suggested registers as follows:

    /*** MEMORY PROTECTION ***/
    hal_mem_protect_enable(bootIndex);
    /*** MEMORY PROTECTION ***/
    /* 7. Load the app image for booting */
    ESP_LOGI(TAG, "Try to load image: %d (-1: factory | 0/1: main app)", bootIndex);
    bootloader_utility_load_boot_image(&bs, bootIndex);

    uint32_t value = REG_READ(SYSCON_SPI_MEM_PMS_CTRL_REG);
    ESP_LOGE("PCU", "read SYSCON_SPI_MEM_PMS_CTRL_REG: %X", value);
    value = REG_READ(SYSCON_SPI_MEM_REJECT_ADDR_REG);
    ESP_LOGE("PCU", "read SYSCON_SPI_MEM_REJECT_ADDR_REG: %X", value);

However, I got the same response on the monitor. I believe I just resets after "bootloader_utility_load_boot_image(&bs, bootIndex);" and never got to the next two lines.
 
This is the response I get:

[image]

sachin-billore, 2025-04-15:
Hi @plamendc ,

>> Yeah, currently I have two partitions SSBL and PROD that are not of a size 64Kb. I have read in the documentation that the size >> has to be aligned to 64kB. Does this also apply for their starting addresses?
Yes, start address and length both should be 64KB align.

>> Currently, in the FLASH_ACE_n_SIZE register I set the variable that is the multiples of 64Kb of the corresponding partition. In other >> words in the case of:
If you are ensuring that size is multiple of 64KB then should be fine.

>> I haven't configured the secure world functionality, therefore, it is operating in default mode. However, I have configured the 
>> access criteria for both modes to full access.
In this case CPU will remain in World0.

>> The interrupt for access violation for external memory is not configured or enabled/disabled by the code, so it should be in the >> default state. I did add reading the suggested registers as follows:
I think execution did not reach till this code and reset before that.

>> However, I got the same response on the monitor. I believe I just resets after "bootloader_utility_load_boot_image(&bs, 
>> bootIndex);" and never got to the next two lines.
I see in the log that size is not multiple of 64kb. And register is not configured as you explain in point 2 in your previous response.

[image]

It is configured with value of OTA_0_SIZE (which is 0x30) and not as 48 * 64Kb = 3072Kb. Could you please check this again?

plamendc, 2025-05-14:
Hi @sachin-billore,

Thank you for getting back to me. I now understand that the starting address and length should be aligned to 64Kbs, meaning segments with smaller lengths or not aligned starting addresses to 64Kbs won't work.

Apart from that, I still don't understand what is the root cause of the resets I experience. 

Regarding this:

**It is configured with value of OTA_0_SIZE (which is 0x30) and not as 48 * 64Kb = 3072Kb. Could you please check this again?**

As far as my understanding, the value for this register is provided in multiples of 64Kb, so for 48 * 64Kb = 3072Kb I need to set this register to 48 or 0x30, am I correct?

sachin-billore, 2025-06-05:
>> As far as my understanding, the value for this register is provided in multiples of 64Kb, so for 48 * 64Kb = 3072Kb I need to set >> this register to 48 or 0x30, am I correct?
Yes you are correct. It should be set as multiple of 64K, i.e., 40 or 0x30.

sachin-billore, 2025-07-08:
Hi @plamendc Closing this issue if you still need help please feel free to reopen or create new issue.