espressif/esp-idf #15490: unit_test only adds test for the alphabeticly first component for TARGET=linux on macOS

morgana2313, 2025-02-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When adding a second component `testable2` only the alphabeticly first component (testable) is included in the unit tests. `testable2` is ignored when compiled for TARGET=linux on macos.

https://github.com/morgana2313/esp-idf/commit/cfca9dee8f4dc8d9e3e7f0a0429be09634c095bf

ESP-Marius, 2025-03-03:
Thanks for reporting this.

Probably this is because we didnt account for the difference between the GNU and the Darwin linker when applying `whole-archive"` for these test components.

I've attached a patch which I think should fix this. 

[0001-fix-linux_target-fixed-TEST_COMPONENTS-not-working-o.txt](https://github.com/user-attachments/files/19048947/0001-fix-linux_target-fixed-TEST_COMPONENTS-not-working-o.txt)

morgana2313, 2025-03-04:
Hello Marius, thank you for your prompt reply!

The patch indeed fixes the problem for TARGET=linux on macOS.

Thanks, Jasper