espressif/esp-idf #15492: nimble ble spp client not subscribing to server

tommasoclini, 2025-02-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have been sent a fix for this via email, but I see that it has not been fixed here, in `examples/bluetooth/nimble/ble_spp/spp_client`:
```
static void
ble_spp_client_set_handle(const struct peer *peer)
{
    const struct peer_chr *chr;
    chr = peer_chr_find_uuid(peer,
                             BLE_UUID16_DECLARE(GATT_SPP_SVC_UUID),
                             BLE_UUID16_DECLARE(GATT_SPP_CHR_UUID));
    attribute_handle[peer->conn_handle] = chr->chr.val_handle;
    // subscribe event; conn_handle=1 attr_handle=12 reason=1 prevn=0 curn=1 previ=0 curi=
    printf("attribute_handle %x\n", attribute_handle[peer->conn_handle]);
    const struct peer_dsc *dsc;
    dsc = peer_dsc_find_uuid(peer,
                             BLE_UUID16_DECLARE(GATT_SPP_SVC_UUID),
                             BLE_UUID16_DECLARE(GATT_SPP_CHR_UUID),
                             BLE_UUID16_DECLARE(BLE_GATT_DSC_CLT_CFG_UUID16));
    if (dsc == NULL) {
        MODLOG_DFLT(ERROR, "Error: Peer lacks a CCCD for the subscribable characterstic\n");
    }
    uint8_t value[2];
    value[0] = 1;
    value[1] = 0;
    ble_gattc_write_flat(peer->conn_handle, dsc->dsc.handle,
                            value, sizeof(value), NULL, NULL);
}
```

rahult-github, 2025-02-28:
Hi @tommasoclini  ,  this is application level change.  The SPP client / server demo example just shows how SPP as a BLE service can be implemented in IDF.  For all other application changes, you are free to make them at your end as per your use case. 

For information, we maintaing Bleprph/ blecent as a detailed example present in IDF which showcases majorly all features.

tommasoclini, 2025-02-28:
> Hi @tommasoclini  ,  this is application level change.  The SPP client / server demo example just shows how SPP as a BLE service can be implemented in IDF.  For all other application changes, you [...]

Ok but I think it would be nice to have the client subscribe to the server as the default, because not everyone knows how to do that, and it's probably expected to happen in the example.

tommasoclini, 2025-03-04:
Why do you not want to add this to the client code? It is literally wrong as of now.

Shreeyash-17, 2025-03-04:
Agreed, we are adding it up internally, and should soon show up.

tommasoclini, 2025-03-04:
> Agreed, we are adding it up internally, and should soon show up.
Thank you

rahult-github, 2025-03-26:
Change is [merged](https://github.com/espressif/esp-idf/commit/030cdf98076cd70e919f974bce2bad0ce0769bcc).

closed 2025-03-26 by rahult-github