espressif/esp-idf #15493: Adjust PSRAM timing tuning point via on-chip temperature sensor in real-time

baldhead69, 2025-02-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-S3 revision v0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

S3-WROOM-2-N32R8V

### Power Supply used.

External 3.3V

### What is the expected behavior?

SPIRAM clock adjust by temperature reading.

### What is the actual behavior?

Crashing at run time when cpu start.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

Icarus113, 2025-03-17:
Hi, it's because this is an experimental feature, now only some devkits with flash chips from limited vendors have passed our internal test. (This is why the error log shows `MSPI Timing: The flash model has not been verified support this feature, please contact espressif business support`)

@baldhead69

Alvin1Zhang, 2025-03-26:
Thanks for reporting, feel free to reopen.

closed 2025-03-26 by Alvin1Zhang

baldhead69, 2025-05-20:
> Hi, it's because this is an experimental feature, now only some devkits with flash chips from limited vendors have passed our internal test. (This is why the error log shows `MSPI Timing: The flash [...]

Flash chip or psram chip ?

Is there a list of memories or development modules that can already use this feature ?

Thanks