espressif/esp-idf #15495: 当tcp连接离线成功后，串口无法接收数据

jankens, 2025-02-28:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.3

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

我的程序通过串口与PC通讯。并创建TCP socket 连接。
当TCP socket连接和PC连线成功后，TCP socket 通讯正常。但是串口只能发，不能收了，当断开TCP socket 连接后，串口接收恢复正常。

我的串口初始化做法参考[peripherals/uart/uart_async_rxtxtasks]中的写法，task已经设置为最高优先级：
 xTaskCreate(uart_receive_task, "uart_task", 2048+1024, (void *)NULL, configMAX_PRIORITIES, NULL);

static void uart_receive_task(void *param)
{
    static uint8_t buffer[UART_BUFFER_SIZE];

    xTaskCreate(uart_send_task, "uart_send_task", 2048, (void *)NULL, configMAX_PRIORITIES-1, NULL);

    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_DEFAULT,
    };
    int intr_alloc_flags = 0;

#if CONFIG_UART_ISR_IN_IRAM
    intr_alloc_flags = ESP_INTR_FLAG_IRAM;
#endif

    ESP_ERROR_CHECK(uart_driver_install(UART_PORT_NUM, sizeof(buffer) * 2, sizeof(buffer) * 2, 0, NULL, intr_alloc_flags));
    ESP_ERROR_CHECK(uart_param_config(UART_PORT_NUM, &uart_config));
    ////ESP_ERROR_CHECK(uart_set_pin(UART_PORT_NUM, ECHO_TEST_TXD, ECHO_TEST_RXD, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));

    while(1) {
        int n = uart_read_bytes(UART_PORT_NUM, buffer, sizeof(buffer), 20 / portTICK_PERIOD_MS);
        if(n > 0) {
            handle_uart_data(buffer, n);
        }
    }
}

请问有什么办法可以解决这个串口接收的问题？谢谢！

### What is the actual behavior?

tcp socket连接成功后，串口无法接收

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

jankens, 2025-03-08:
请问有人遇到相同的问题吗？有没有解决办法？

wanckl, 2025-03-14:
@jankens TCP socket 用的wifi还是有线连接？是不是UART引脚复用了

closed 2025-04-17 by Alvin1Zhang