espressif/esp-idf #15497: 6cc2c71 causes default I2S Slave mode config to stop working

ftab, 2025-02-28:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

6cc2c71

### Espressif SoC revision.

ESP32-P4 v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I2S is configured and works correctly

### What is the actual behavior?

Gives error: sample rate or mclk_multiple is too large for the current clock source

### Steps to reproduce.

1. Configure I2S as slave RX with 48khz sample rate, 32 bit data, stereo
2. Attempt to run on recent master esp-idf including commit 6cc2c71

Using esp-adf i2s_stream:

```
    i2s_stream_cfg_t i2s_cfg =
        I2S_STREAM_CFG_DEFAULT_WITH_PARA(I2S_NUM_1, 48000, I2S_DATA_BIT_WIDTH_32BIT, AUDIO_STREAM_READER);
    i2s_cfg.chan_cfg.role = I2S_ROLE_SLAVE;
    stream_reader = i2s_stream_init(&i2s_cfg);
```

### Debug Logs.

```plain
E (10845) i2s_std: i2s_std_calculate_clock(63): sample rate or mclk_multiple is too large for the current clock source
E (10846) i2s_std: i2s_std_set_clock(80): clock calculate failed
E (10847) i2s_std: i2s_channel_init_std_mode(240): initialize channel failed while setting clock
E (10848) I2S_STREAM_IDF5.x: I2S stream init failed
```

### More Information.

I reverted 6cc2c71 and my setup works again

@L-KAYA what am I doing wrong with the new version that it suddenly doesn't like the clock config?

L-KAYA, 2025-03-21:
Hi @ftab, thanks for reporting, may I ask what clock source are you using for I2S? On P4, the default clock source of I2S is XTAL, which is 40MHz. I calculated the MCLK frequency in your case, it's about 24MHz, more than half of the source clock, it 
actually works in wrong clock division, which might be risky for slave role reading or writing.

And the commit you mentioned is kind of improvement of the check, which will be more strict to guarantee it reads/writes correctly.

You can try to use `I2S_CLK_SRC_APLL` as the I2S source clock to solve this issue.

ftab, 2025-03-21:
@L-KAYA I am using APLL for the master clock (TDM8 32 bit 48khz output). Which means I can't also use APLL for the slave clock (stereo 32 bit 48khz input). Unless I'm mistaken.

L-KAYA, 2025-03-21:
@ftab Although APLL frequency can't be changed after it is used, another I2S port will still try to work at the current APLL frequency, if the current ALL frequency can satisfy the latter I2S port (i.e., as long as the latter one requires slower or same APLL frequency), then they can both use the APLL source.

In your case, the slave role needs `24.576 MHz` MCLK, meanwhile the TDM master also requires `24.576 MHz`(assume `mclk_multiple = 512`), it is exactly same as the previous one, so you can register the slave role first and then the master role.

ftab, 2025-03-24:
@L-KAYA here's what I get when I set up both using APLL (output first):

W (9637) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 768
I (9639) AUD: Register stream_writer to pipeline
I (9640) AUD: Start codec chip
I (9648) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (9649) AUD: Create i2s stream to read data from A2B
W (9650) i2s_common: APLL is occupied already, it is working at 73728027 Hz while the expected frequency is 49152000 Hz
W (9652) i2s_common: Trying to work at 73728027 Hz...

Does that make sense? I thought mclk multiple 768 would mean a 36,864,000 Hz clock

It seems to work.

L-KAYA, 2025-03-31:
@ftab It is correct, here is how the driver calculate:
- For the master TDM8, 32bit @48KHz:` APLL freq = 48K * 768 * 2(minimum source to mclk divider) = 73.728 MHz`, then the actual APLL was set to 73728027 Hz initially.
- For the slave TDM2(or STD), 32bit @48KHz: it tried to set `APLL freq = 48K * 32bit * 2 slots * 8(minimum mclk to bclk divider for slave role) * 2(minimum source to mclk divider) = 49.152 MHz`. But since the previous APLL freq was set, the driver will adjust the souce to mclk divider, it will finally adjust to: `APLL freq = 48K * 32bit * 2 slots * 8 * 3 = 73.728 MHz`

ftab, 2025-04-02:
@L-KAYA thanks! Last question I can think of, why is it 73728027 Hz specifically and not exactly 73728000?

L-KAYA, 2025-04-03:
The APLL frequency is generated from the XTAL with some coefficients. See the ESP32 TRM about APLL (P4 is same):

[image]

There are four coefficients needed to generate a specific APLL frequency. Here is the general steps:

1. [Step 1](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2s/i2s_common.c#L546) Once we choose APLL as the I2S source clock, the driver will calculate the expected APLL frequency with the default minimum divider, (i.e., APLL freq = MCLK * 2). 
2. [Step 2](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2s/i2s_common.c#L557) Check whether the APLL has been used. If used by other I2S ports or peripherals, then calculated the source to mclk divider `mclk_div` again by the known APLL frequency. Otherwise we need to calculate the APLL coefficients.
3. [Step 3](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/port/esp32p4/rtc_clk.c#L484) Calculate the proper APLL coefficients to get the most closed APLL freq
4. [Step 4](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/port/esp32p4/rtc_clk.c#L535) Calculate the real APLL according to the coefficients found by step3.

For summary, APLL frequency is not a fixed frequency, it is generated by XTAL with 4 arguments, it won't be 100% accurate.

closed 2025-04-03 by ftab