espressif/esp-idf #15502: ESP32 compilation error triggered by recommended by third part PSRAM type

baruchli, 2025-03-01:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Arduino IDE 2.3.4

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I use SparkFun Thing Plus - ESP32-S3 board with next SW environment:
Arduino IDE: 2.3.4
Installed boards support:
Arduino ESP32 Boards v2.0.18 - arduino.5 by Arduino
esp32 v3.2.0-RC1 by Espressif

If I choose PSRAM: QSPI PSRAM option according to [Hookup guide](https://docs.sparkfun.com/SparkFun_Thing_Plus_ESP32-S3/arduino_example/#board-settings).

I had tried to compile one of examples provided by Sparkfun [RGB_LED.ino](https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3/blob/main/Firmware/RGB_LED/RGB_LED.ino) 

Expected behaviour: Successful compilation 

### What is the actual behavior?

Actual behavior:
Next compilation error:

> In file included from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/sys/reent.h:34,
> from …\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/reent.h:13,
> from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/stdio.h:60,
> from…\Arduino15\packages\esp32\hardware\esp32\3.2.0-RC1\cores\esp32/Arduino.h:27,
> from …\AppData\Local\arduino\sketches\F2D0A6956F0C0D1ADC2714C276F620B6\sketch\RGB_LED.ino.cpp:1:
> …\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/lock.h:9:10: fatal error: sdkconfig.h: No such file or directory
> 9 | #include “sdkconfig.h”
> | ^~~~~~~~~~~~~
> compilation terminated 

### Steps to reproduce.

Install from scretche next environment:

1. Arduino IDE: 2.3.4
2. Installed boards support:
3. Arduino ESP32 Boards v2.0.18 - arduino.5 by Arduino
4. esp32 v3.2.0-RC1 by Espressif

- Import into Arduino IDE [FastLED library](https://github.com/FastLED/FastLED)
- Clone Spakrfun [Thing Plus ESP32-S3 repository](https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3)
- Open [RGB_LED.ino](https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3/blob/main/Firmware/RGB_LED/RGB_LED.ino) file and try to compile it. You will get compilation error.

### Build or installation Logs.

```plain
In file included from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/sys/reent.h:34,
from …\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/reent.h:13,
from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/stdio.h:60,
from…\Arduino15\packages\esp32\hardware\esp32\3.2.0-RC1\cores\esp32/Arduino.h:27,
from …\AppData\Local\arduino\sketches\F2D0A6956F0C0D1ADC2714C276F620B6\sketch\RGB_LED.ino.cpp:1:
…\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/lock.h:9:10: fatal error: sdkconfig.h: No such file or directory
9 | #include “sdkconfig.h”
| ^~~~~~~~~~~~~
compilation terminated
```

Alvin1Zhang, 2025-03-03:
Thanks for reporting.

baruchli, 2025-03-05:
> Thanks for reporting.

You welcome. I will be glad to provide any information that will help to mend the issue.