espressif/esp-idf #15503: menuconfig no option for altering save location

hyansuper, 2025-03-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-683-g0ba53566fa

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I just install esp-idf using the windows installation. with release/v5.4 version. 

and I have read [configuration guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/kconfig/project-configuration-guide.html) on `idf.py menuconfig`.

### What is the actual behavior?

but in menuconfig, when I hit [S] to save sdkconfig, I can't alter save location

also, the [D] for save minimal config can't alter save path

### Steps to reproduce.

idf.py menuconfig
[O] to load default config
then [S] will save to the previuosly entered default config path, and overwrite the default config file

### Build or installation Logs.

```plain

```

BitsForPeople, 2025-03-02:
Why do you want to change the 'save path' of a project's sdkconfig file?

Every project has its own sdkconfig file, which stores the Kconfig settings for this project, and this project-specific file is what you edit via idf.py menuconfig.

Alvin1Zhang, 2025-03-03:
Thanks for reporting.

dobairoland, 2025-03-03:
HI @hyansuper. I transfered this to feature request and we will consider it. Changing the sdkconfig file locations or names from menuconfig were never supported. I'm not sure if it ever will be.

hyansuper, 2025-03-03:
@dobairoland you don't necessarily have to support this feature, if default location is fix to sdkcobfig, and not overwrite other files.

Honza0297, 2025-03-03:
Hi @hyansuper and thank you for reaching us! Currently, "our" menuconfig truly does not support changing the location where to save files (older versions did support it). It was removed on purpose, because most of the time (like 95 % of it) , the use case is  just changing several values and wanting to save them back to `sdkconfig`. 
I will consider how to support it for use cases like yours (something like "save as" option).

hyansuper, 2025-03-03:
I am sure in v5.1 there was an input text field prompt when you hit [D] or [S], just like [O]