espressif/esp-idf #15505: esp32c3 calibration seems not working when BLE only with ESP-IDF after 5.1.4

yuki-sato, 2025-03-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.6

### Espressif SoC revision.

esp32-c3

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

my own

### Power Supply used.

External 3.3V

### What is the expected behavior?

We use HCI command to control BLE and collect iBeacons.
We found that 10% of our esp32-c3 report low-level rssi. 
It should be around 45 but it reports around 80.
Downgrading ESP-IDF solve this problem

### What is the actual behavior?

It reports low-level RSSI of ble packet and missing rate is bad.

If you turn Wi-Fi ON, then BLE RSSI seems good range(45-55).
I suspect Calibration is not working under BLE only situation with ESP-IDF 5.1.4 and later.

### Steps to reproduce.

1. Build with ESP-IDF v5.1.4( or 5.1.6 )
2. Run Only BLE
3. Collect RSSI data. it seems low.
4. Turn on Wi-Fi and BLE
5. Then you can see correct RSSI levels.

This occurs not every esp32-c3. We found it around 10% of esp32-c3.

### Debug Logs.

```plain
iBeacon logs. last one is RSSI of iBeacon ESP32 got.

Only BLE

2025-03-03 16:37:19.170, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -72
2025-03-03 16:37:19.290, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -72
2025-03-03 16:37:19.290, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -72
2025-03-03 16:37:19.290, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -50
2025-03-03 16:37:19.291, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -50
2025-03-03 16:37:19.529, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -71
2025-03-03 16:37:19.529, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -71
2025-03-03 16:37:19.600, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -46
2025-03-03 16:37:19.600, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -50
2025-03-03 16:37:19.769, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -72
2025-03-03 16:37:19.770, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -72
2025-03-03 16:37:19.830, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -50
2025-03-03 16:37:19.830, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -50

BLE+Wi-Fi (same device)

2025-03-03 16:38:18.369, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -49
2025-03-03 16:38:19.497, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -42
2025-03-03 16:38:19.627, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -35
2025-03-03 16:38:19.627, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -42
2025-03-03 16:38:19.939, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -43
2025-03-03 16:38:19.939, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -43
2025-03-03 16:38:20.007, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -45
2025-03-03 16:38:20.008, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -71
2025-03-03 16:38:20.418, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -50
2025-03-03 16:38:20.419, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -50
2025-03-03 16:38:20.556, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -38
2025-03-03 16:38:20.557, 9f8a7ae3-5d01-49e4-9329-312032d434e2, 1, 100, -43
```

### More Information.

We tries changing calibration procedure like  changing `CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE` on/off and calling `esp_phy_erase_cal_data_in_nvs()` before starting BLE. But these never helped.

Only "Turning On Wi-Fi" or "Downgrading ESP-IDF" works.

esp-zhp, 2025-03-13:
@yuki-sato 
1. Is there significant interference in your environment? (If you have a shielding box, try testing inside it.)  
2. What type of antenna are you using?  
3. When performing a BLE scan, can you lock the scan to a specific channel? By default, scanning occurs on channels 37, 38, and 39, and the RSSI of advertisements can vary considerably across different channels. Comparing RSSI results on the same channel will yield more accurate measurements.
4-Is the iBeacon device's position fixed?

yuki-sato, 2025-03-13:
@esp-zhp Thank you for you reply.

1. Unfortunately I don't have one.
2. I'm using ESP32-C3-MINI. And It's pattern antenna
3. I will check how to lock with ESP-IDF
4. Of-course. Fixed for both test. I just changed the software on ESP32C3. And this is not only once. I tried several times. I changed the location and devices but same result.

esp-zhp, 2025-03-17:
3-for adv,
you can fix advertising channel by:

[image]

yuki-sato, 2025-03-17:
@esp-zhp Thank you for your support.
Unfortunately our software is not using bluedroid. We are using Bluetooth HCI to control bluetooth.
So your suggested mask can not be used with.

i'm figuring out how to fix with Bluetooth HCI.

Alvin1Zhang, 2025-03-26:
Thanks for reporting, feel free to reopen.

yuki-sato, 2025-03-26:
OK I'm still investigating. I will reopen when I prepared the data with fixed adv channnel.