espressif/esp-idf #15506: Allow mocking individual driver components the same way real driver components can be individually specified.

i400s, 2025-03-03:
### Is your feature request related to a problem?

Component mocking the "driver" for linux or "on chip" is only partially implemented and is also only available as a "get all" option.

I have a component I wish to test using mocks. The component requires i2c and gpio mocks as its an i2c component that also has gpio "interrupt" processing. The current mocking only mocks the old style of i2c.

My current workaround for testing my component is either to patch the esp-idf driver mock, or to have individual mocks within my testing application directory eg: `my_test_app/components/esp_driver_gpio` and `my_test_app/components/esp_driver_i2c`

Both solutions work, but neither is great. Additionally, when I eventually make the code public via github then the patched version won't work as only my system will be patched as required... and having multiple mock drivers within my testing application isn't the best solution; especially as I have multiple components all requiring i2c and gpio (mcp9808, sd3031, etc.) so each physical device would need the same duplicated mock directories.

### Describe the solution you'd like.

1) The global "driver" component should also mock the new style i2c_master and i2c_slave headers which it currently does not. The simplest fix is as follows:
```
diff --git a/tools/mocks/driver/CMakeLists.txt b/tools/mocks/driver/CMakeLists.txt
index f8647eda7c..d5a3db05fa 100644
--- a/tools/mocks/driver/CMakeLists.txt
+++ b/tools/mocks/driver/CMakeLists.txt
@@ -1,6 +1,6 @@
 # NOTE: This kind of mocking currently works on Linux targets only.
 #       On Espressif chips, too many dependencies are missing at the moment.
-message(STATUS "building DRIVER MOCKS (only SPI master, I2C, RMT, USB-serial and GPIO driver)")
+message(STATUS "building DRIVER MOCKS (only SPI master, I2C, I2C new style, RMT, USB-serial and GPIO driver)")
 
 idf_component_get_property(original_driver_dir driver COMPONENT_OVERRIDEN_DIR)
 
@@ -11,6 +11,8 @@ set(include_dirs
     "${IDF_PATH}/components/esp_driver_spi/include"
     "${IDF_PATH}/components/esp_driver_rmt/include/driver"
     "${IDF_PATH}/components/esp_driver_rmt/include"
+    "${IDF_PATH}/components/esp_driver_i2c/include/driver"
+    "${IDF_PATH}/components/esp_driver_i2c/include"
     "${original_driver_dir}/i2c/include/driver"
     "${IDF_PATH}/components/esp_driver_usb_serial_jtag/include/driver"
     "${original_driver_dir}/i2c/include"
@@ -28,5 +30,7 @@ idf_component_mock(INCLUDE_DIRS ${include_dirs}
     ${IDF_PATH}/components/esp_driver_rmt/include/driver/rmt_tx.h
     ${IDF_PATH}/components/esp_driver_rmt/include/driver/rmt_common.h
     ${IDF_PATH}/components/esp_driver_rmt/include/driver/rmt_encoder.h
+    ${IDF_PATH}/components/esp_driver_i2c/include/driver/i2c_master.h
+    ${IDF_PATH}/components/esp_driver_i2c/include/driver/i2c_slave.h
     ${original_driver_dir}/i2c/include/driver/i2c.h
     ${IDF_PATH}/components/esp_driver_usb_serial_jtag/include/driver/usb_serial_jtag.h)
```
2) A more complete solution would be to have individual mocks for the various driver subsystems and then have the "driver" component pull those individual mocks in (see 4). 

3) With the more complete solution if mock "driver" (`set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/tools/mocks/driver/)`) is selected then using individual sub components should work for mocks in the same way it works for real drivers... specifying `REQUIRES esp_driver_gpio esp_driver_i2c` at a component level should pull in the subsystem mocked drivers but it currently fails with `#include Mockgpio.h not found`.

4) There should be the option to specify individual driver components to be mocked eg:  ` set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/tools/mocks/esp_driver_i2c/)`

i400s, 2025-03-04:
I just wanted to add. My patch has only been tested to work for my specific instance. I've not checked that it doesn't break some other part of the idf. I'm still learning how these things work, and while I know it should in theory be possible to run tests on check ins (github mentions something about tests/runners being disabled on the fork) I have no idea how to set that up on a local system or even if espressif have a totally different system of checks that run prior to pushing to github.

ESP-Marius, 2025-03-28:
@i400s thanks for the suggestions in both issues. What you are requesting does make sense!

It is unlikely we'll have capacity/time to improve the mocking situation right now. But I will keep both feature requests in our backlog for whenever we do have time to look at it!

i400s, 2025-04-09:
Hi @ESP-Marius thanks for your reply.

No rush as my workaround works and isn't to horrible. 

My apologies for the late acknowledgement.