espressif/esp-idf #15510: W5500 Ethernet SPI timeouts when ILI9342C LCD has been used

jlmurdoch, 2025-03-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

M5Stack CoreS3 SE

### Power Supply used.

USB

### What is the expected behavior?

Both W5500 (esp_eth) and ILI9342C LCD (esp_lcd) running on same SPI bus should work with no issues.

### What is the actual behavior?

When the ILI9342C LCD is accessed, even just once (e.g. `esp_lcd_panel_draw_bitmap()`), the W5500 generates the following every second thereafter:
```
E (14112) w5500.mac: w5500_send_command(208): send command timeout
E (14112) w5500.mac: emac_w5500_start(376): issue OPEN command failed
E (14112) w5500.mac: emac_w5500_set_link(467): w5500 start failed
E (14112) esp_eth: eth_on_state_changed(131): ethernet mac set link failed
E (14122) w5500.phy: w5500_update_link_duplex_speed(88): change link failed
E (14132) w5500.phy: w5500_get_link(112): update link duplex speed failed
```

### Steps to reproduce.

1. Take attached code, compile and upload onto a M5Stack CoreS3 SE with W5500 v1.2 Module attached.
2. Witness `send command timeout`
3. Comment out `esp_lcd_panel_draw_bitmap()` line.
4. Recompile & upload
5. Witness no errors. Link is detected if a cable is attached.

[lcd_w5500_coexistance.c.txt](https://github.com/user-attachments/files/19061847/lcd_w5500_coexistance.c.txt)

### Debug Logs.

```plain
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4200b9c5
--- 0x4200b9c5: s_test_psram at /home/jmurdoch/esp/v5.4/esp-idf/components/esp_psram/esp_psram.c:479

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (26) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Mar  3 2025 21:43:30
I (27) boot: Multicore bootloader
I (28) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=153cch ( 86988) map
I (104) esp_image: segment 1: paddr=000253f4 vaddr=3fc95c00 size=02cd8h ( 11480) load
I (107) esp_image: segment 2: paddr=000280d4 vaddr=40374000 size=07f44h ( 32580) load
I (115) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=3bcd0h (244944) map
I (159) esp_image: segment 4: paddr=0006bcf8 vaddr=4037bf44 size=09c04h ( 39940) load
I (168) esp_image: segment 5: paddr=00075904 vaddr=600fe100 size=0001ch (    28) load
I (176) boot: Loaded app from partition at offset 0x10000
I (176) boot: Disabling RNG early entropy source...
I (186) esp_psram: Found 8MB PSRAM device
I (186) esp_psram: Speed: 40MHz
I (187) cpu_start: Multicore app
I (1862) esp_psram: SPI SRAM memory test OK
I (1870) cpu_start: Pro cpu start user code
I (1870) cpu_start: cpu freq: 160000000 Hz
I (1871) app_init: Application information:
I (1871) app_init: Project name:     lcd_w5500_coexistance
I (1876) app_init: App version:      1
I (1879) app_init: Compile time:     Mar  3 2025 21:48:54
I (1885) app_init: ELF file SHA256:  5e2b75fc8...
I (1889) app_init: ESP-IDF:          v5.4-dirty
I (1893) efuse_init: Min chip rev:     v0.0
I (1897) efuse_init: Max chip rev:     v0.99 
I (1901) efuse_init: Chip rev:         v0.2
I (1905) heap_init: Initializing. RAM available for dynamic allocation:
I (1912) heap_init: At 3FC99C10 len 0004FB00 (318 KiB): RAM
I (1917) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (1922) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1927) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (1933) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1940) spi_flash: detected chip: generic
I (1943) spi_flash: flash io: dio
W (1946) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (1959) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (1965) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (1972) main_task: Started on CPU0
I (1982) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1982) main_task: Calling app_main()
I (12002) esp_eth.netif.netif_glue: 48:xx:xx:xx:xx:xx
I (12002) esp_eth.netif.netif_glue: ethernet attached to netif
I (12012) LCD_W5500_TEST: Ethernet Started
I (12012) ili9341: LCD panel create success, version: 2.0.0
Reset LCD
Initialise LCD
Turn on LCD
LCD Panel: Change
E (14112) w5500.mac: w5500_send_command(208): send command timeout
E (14112) w5500.mac: emac_w5500_start(376): issue OPEN command failed
E (14112) w5500.mac: emac_w5500_set_link(467): w5500 start failed
E (14112) esp_eth: eth_on_state_changed(131): ethernet mac set link failed
E (14122) w5500.phy: w5500_update_link_duplex_speed(88): change link failed
E (14132) w5500.phy: w5500_get_link(112): update link duplex speed failed
E (16112) w5500.mac: w5500_send_command(208): send command timeout
E (16112) w5500.mac: emac_w5500_start(376): issue OPEN command failed
E (16112) w5500.mac: emac_w5500_set_link(467): w5500 start failed
E (16112) esp_eth: eth_on_state_changed(131): ethernet mac set link failed
E (16122) w5500.phy: w5500_update_link_duplex_speed(88): change link failed
E (16132) w5500.phy: w5500_get_link(112): update link duplex speed failed
```

### More Information.

The M5Stack CoreS3 SE and W5500 v1.2 module SPI setup is as below. 

- SPI
  - MISO    35
  - MOSI    37
  - SCLK    36 
- LCD
  - CS      3
  - DC      35
- W5500
  - CS      9
  - RST     7
  - INT     14

Other observations:
- Incidentally, the same challenge, but less verbose is witnessed on the following BSP issue: [M5Stack CoreS3 and W5500 Error during network initialization](https://github.com/espressif/esp-bsp/issues/374). Note that the pinouts are the same as above. 
- I can run an `esp_eth_ioctl()` with `ETH_CMD_G_MAC_ADDR` to return the MAC address, so the W5500 SPI is functional within `app_main()`, so it may be a esp_timer issue with SPI.
- With alternate code, I can access another SPI device with no issues (SX127x-based radio)

kostaond, 2025-03-10:
@jlmurdoch thank you for reporting the issue. We will obtain hardware to try to reproduce and I'll be back to you. I presume we will be able since I've already consulted with LCD display driver maintainer and he confirmed me that similar issue has been observed with display when it was used with SPI SD card reader...

kostaond, 2025-05-07:
@jlmurdoch I finally had time to have a look on this issue and I think I found reason it doesn't work... The problem is the W5500 SPI MISO is GPIO 35 but at the same time GPIO 35 is used as LCD DC. As such, the LCD driver pulls MISO always to low and W5500 cannot transmit any data back to ESP32. I'll try to discuss with LCD driver maintainer if it was possible to use ILI9342 in "3-line Serial Interface" (without DC). It should be possible per  [datasheet](https://www.displayfuture.com/Display/datasheet/controller/ILI9342.pdf) (Section 7.6.1).

erikveg, 2025-11-07:
@kostaond : I'm experiencing the same problem. However, when I try it with the arduino examples, it works. So I don't believe it is a hardware issue.

closed 2025-12-04 by Alvin1Zhang