espressif/esp-idf #15511: 使用片外FLASH FATFS性能测试

Fire3612, 2025-03-04:
### Describe the solution you'd like.

我实际测的片外Flash fatfs写入速度大概在7KB/S，还能有提升吗

Fire3612, 2025-03-04:
240Mhz，SPI frequency 80Mhz

adokitkat, 2025-03-07:
Hello, can you try to run performance benchmark example and copy the result here? It's in `examples/storage/perf_benchmark` folder .

Fire3612, 2025-03-12:
Problem sloved 。   This is new version     
static wl_handle_t wl_handle;
        const esp_vfs_fat_mount_config_t mount_config = {
            .max_files = 40,
            .format_if_mount_failed = true,
            .allocation_unit_size = 20 * 1024};
        esp_err_t err = esp_vfs_fat_spiflash_mount_rw_wl(MOUNT_PATH, "fatfs", &mount_config, &wl_handle);          allocation_unit_size used to be 512

closed 2025-03-12 by Fire3612