espressif/esp-idf #15515: esp32s3 + esp-idf-5.2.1 pdm录音不正常

crl6, 2025-03-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

去年的问题还没解决。https://github.com/espressif/esp-idf/issues/13714
做过的尝试，更换不同开发板，使用不同gpio，使用不同pdm麦克风，今天新增测试DM4737-223，效果是一样的。

1. 请问你们测试pdm录音功能，有使用过哪款pdm麦克风吗，可以给个型号或者链接吗？
2. 你们测试pdm录音是保存成文件在电脑上看波形还是直接串口打印数据？

crl6, 2025-03-04:
连线图：

[image]

crl6, 2025-03-05:
sdk自带的pdm rx例子是录制单声道，并且是用串口打印的。

麻烦提供一个验证ok的例子（录音立体声保存成文件，在电脑播放声音正常的，并且波形也是正常的）。

crl6, 2025-03-08:
@L-KAYA  您好。

我看master代码和5.2.1代码有差别，请问5.2.1版本有测试过我说的立体声录音，在电脑上看波形正常吗？

另外你给的例子，代码也是单声道的。请确保你们自己有实际测试过立体声录音。

#define NUM_CHANNELS        (1) // For mono recording only!

#if SOC_I2S_SUPPORTS_PDM2PCM
        .slot_cfg = I2S_PDM_RX_SLOT_PCM_FMT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO),
#else
        .slot_cfg = I2S_PDM_RX_SLOT_RAW_FMT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO),
#endif

意思是esp32S3 pdm录音只支持单声道吗？

L-KAYA, 2025-03-10:
[i2s_s3_pdm_stereo.patch](https://github.com/user-attachments/files/19162632/i2s_s3_pdm_stereo.patch)

@crl6 您好，我看了一下S3 PDM Stereo下的寄存器配置，找到了可能出问题的代码，修复后双声道音频采样正常，麻烦测试一下这个patch对您是否有效，非常感谢！

crl6, 2025-03-11:
@L-KAYA 您好，还是一样。

### 一、另外我改成只用单声道录音发现也有问题:
.slot_mask = I2S_PDM_SLOT_RIGHT 的时候录音正常。
.slot_mask = I2S_PDM_SLOT_LEFT 的时候录音波形不正常，都在0db线以下（把mic板2个mic的clk和data分开，只留左mic连接esp32s3 问题还是一样）。

波形对比如下图：

[image]

麻烦您帮忙测试一下 .slot_mask = I2S_PDM_SLOT_LEFT 的时候录音波形是否正常。

### 二、个人觉得可能得原因：
1. esp32s3 的 GPIO matrix功能可能有影响。
2. esp32s3的  GPIO毛刺过滤器 功能可能有影响。
3. esp32s3的sdk哪里做了处理，感觉数值被处理过。

L-KAYA, 2025-03-12:
这是我测的数据，上面是只有左声道的，下面是双声道的，双声道里的左声道确实有不知道哪里进来的噪音

[image]

## 测试例程和音频：

[i2s_wifi_recorder.zip](https://github.com/user-attachments/files/19202309/i2s_wifi_recorder.zip)

## 用法：

1. 修改 `sdkconfig.defaults` 中的wifi 名字和密码；
2. 修改 `i2s_record_main.c` 中的ip地址；
3. 电脑端运行 `python tcp_sever.py` 启动 tcp 主机；
4. 烧录程序到 S3, 等录音完成；

crl6, 2025-03-12:
@L-KAYA 您好。
1. 我说的不是杂音。我没有遇到杂音问题。

2. 我说的是波形全部在0.0 这条中线以下。请花点时间仔细看看我的描述，特别是图片，以及https://github.com/espressif/esp-idf/issues/13714 这个问题。

3. 看你的截图波形也是有问题的，波形不在0.0中线对称分布，全在0.0这条线上面。

crl6, 2025-03-12:
@L-KAYA  
1. 你发的 [i2s_wifi_recorder.zip](https://github.com/user-attachments/files/19202309/i2s_wifi_recorder.zip) 和 [这个 record 例程](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/i2s_recorder) 没有本质区别。

2. 我自己尝试过修改很多参数，如：dma_desc_num，dma_frame_num等sdk开放的参数我都排列组合试过N遍了。也更换过不同厂家mic。都是有问题。之前用gpio 10 12，发现波形全部在0.0中线以下。使用你推荐的i2s_recorder 这个例子使用的是gpio 4 5,发现有部分波形能超过0.0中线。不知道你i2s_wifi_recorder 使用的是哪2个gpio。

我自己测试的2款在rk3588量产的mic板 + 你手上的一款。一共三款都有同样的问题，可以确定是esp32s3硬件或者sdk有问题，不可能是mic板有问题。

crl6, 2025-03-17:
@L-KAYA  您好，请问贵司有人在看这个问题吗？

L-KAYA, 2025-03-17:
@crl6 跟数字的同事确认了一下，S3上的PDM RX还没支持高通滤波，所以这个低频直流分量确实会存在，后面的芯片像 ESP32-P4 才会带可配置的高通滤波

crl6, 2025-03-17:
@L-KAYA 您说的 “S3上的PDM RX还没支持高通滤波”  这里的高通滤波指的是s3 硬件上的高通滤波器吗？ 

如果s3硬件上没有高通滤波器，能否让软件同学实现软件高通滤波器？ 

我们最开始用s2，为了让产品支持pdm功能，特地换成了s3。不想再换p4了。

L-KAYA, 2025-03-19:
Hi @crl6 ，是的，指的是硬件高通滤波器。如果要软件实现高通滤波器，您可以自己实现一个简单的一阶 IIR 高通滤波器，示例代码如下:

```c
#include <math.h>

// 滤波器结构体
typedef struct {
    float b0, b1;    // 分子系数
    float a1;        // 分母系数
    float x1;        // 输入延迟单元
    float y1;        // 输出延迟单元
} IIR_HPF_1st;

// 初始化滤波器系数
// fs: 采样频率(Hz)
// fc: 截止频率(Hz)
void IIR_HPF_1st_Init(IIR_HPF_1st* filter, float fs, float fc) {
    // 使用双线性变换进行设计
    const float omega_c = 2.0f * M_PI * fc;
    const float T = 1.0f / fs;
    
    // 预畸变校正
    const float warped_omega = (2.0f/T) * tanf(omega_c * T / 2.0f);
    
    // 模拟原型系数（一阶RC高通）
    const float alpha = 1.0f / (1.0f + 1.0f/(warped_omega * T));
    
    // 转换为数字滤波器系数
    filter->b0 = 1.0f / (1.0f + 1.0f/(warped_omega * T));
    filter->b1 = -filter->b0;
    filter->a1 = (1.0f - 1.0f/(warped_omega * T)) / (1.0f + 1.0f/(warped_omega * T));

    // 初始化延迟单元
    filter->x1 = 0.0f;
    filter->y1 = 0.0f;
}

// 执行滤波计算
float IIR_HPF_1st_Process(IIR_HPF_1st* filter, float input) {
    // 计算输出
    float output = filter->b0 * input 
                 + filter->b1 * filter->x1
                 - filter->a1 * filter->y1;

    // 更新延迟单元
    filter->x1 = input;
    filter->y1 = output;

    return output;
}

/******************** 使用示例 ********************/
int main() {
    IIR_HPF_1st hpf;
    const float fs = 8000.0f;  // 8kHz采样率
    const float fc = 200.0f;   // 200Hz截止频率

    // 初始化滤波器
    IIR_HPF_1st_Init(&hpf, fs, fc);

    // 模拟处理流程
    while(1) {
        float input = /* 获取输入采样值 */;
        float filtered = IIR_HPF_1st_Process(&hpf, input);
        /* 使用滤波后的数据 */
    }

    return 0;
}
```

另外您也可以考虑 FIR 滤波器，计算量比 IIR 大，但是相位线性。类似的，还可以选择更复杂、计算量更大的频域滤波，通过 FFT 变换到频率，再通过窗函数滤除无用频段，最后由 IFFT 返回时域。

如果由我们添加的话，不会直接集成到驱动中，会以独立组件的形式在 [component registry](https://components.espressif.com/) 里提供，由于为这些这些滤波器设计通用化接口比较花时间，所以进程会比较慢，如果您着急，可以先参考上面提供的数字滤波器设计思路尝试软件上对对PDM接收到的数据进行滤波。