espressif/esp-idf #15520: touch button can‘t support single click.

iotsystem, 2025-03-05:
### Is your feature request related to a problem?

touch button can‘t support single click.

L-KAYA, 2025-03-07:
Hi @iotsystem, could you provide some details about this feature request? Like, what kind of touch driver API you are using? The legacy touch driver, the new touch driver or touch element library?

And all of these API should support the single click, do you actually mean it's a bug?

Looking forward to your reply, thank you!

iotsystem, 2025-03-10:
typedef enum {
    TOUCH_BUTTON_EVT_ON_PRESS,         //!< Button Press event
    TOUCH_BUTTON_EVT_ON_RELEASE,       //!< Button Release event
    TOUCH_BUTTON_EVT_ON_LONGPRESS,     //!< Button LongPress event
    TOUCH_BUTTON_EVT_MAX
} touch_button_event_t;
Touch buttons support the three enumeration types mentioned above, but I cannot distinguish whether it is a single click or a long press. For example, if I want to trigger a long press, it will definitely trigger a press event. If possible, it is recommended to add single click, double click, and long press events

L-KAYA, 2025-03-10:
@iotsystem Thank you for your advice! The current touch element library indeed does not directly support `double click` and `interval of two clicks`. But it can distinguish single click and lone press by registering multiple event and their callbacks, for example:

```
void button_event_handler(touch_button_handle_t handle, touch_button_message_t *message, void *arg)
{
    if (message->event == TOUCH_BUTTON_EVT_ON_PRESS) {
        // Single press
    }
    if (message->event == TOUCH_BUTTON_EVT_ON_LONGPRESS) {
        // Long press
    }
}

// ......
// Register both TOUCH_ELEM_EVENT_ON_PRESS and TOUCH_ELEM_EVENT_ON_LONGPRESS events
touch_button_subscribe_event(button_handle, TOUCH_ELEM_EVENT_ON_PRESS | TOUCH_ELEM_EVENT_ON_LONGPRESS, &button_channel_num);
touch_button_set_longpress(button_handle, 300);
touch_button_set_dispatch_method(button_handle, TOUCH_ELEM_DISP_CALLBACK));
touch_button_set_callback(button_handle, &button_event_handler);
```

And as for the double click, it can be implemented in the application, like, starting a timer when the first press triggered, and if the second press triggered within the expected time, then double click event happened, otherwise just reset the press count on expire of the timer.

We'll consider to add double click feature if there are more requests on this feature.

iotsystem, 2025-03-11:
Well, thank you very much.

closed 2025-03-11 by iotsystem