espressif/esp-idf #15526: Short config desc has wrong bDescriptorType

alex-k8, 2025-03-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

I expect the enumeration and parsing to succeed successfully when connecting a USB Keyboard

### What is the actual behavior?

Instead, I get the error `Short config desc has wrong bDescriptorType`

### Steps to reproduce.

1. Build and upload the HID Host or the USB Host Lib Example code
2. Connect the NuPhy Air75 HE keyboard
3. Check the logs

### Debug Logs.

```plain
[0;32mI (6795) CLASS: Deregistering Client[0m
[0;32mI (6795) USB host lib: No more clients[0m
[0;32mI (6795) USB host lib: All devices marked as free[0m
[0;32mI (6895) USB host lib: End of the example[0m
[0;32mI (6895) main_task: Returned from app_main()[0m
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x21d4
load:0x403c8700,len:0xed8
load:0x403cb700,len:0x3468
entry 0x403c8974
I (26) boot: ESP-IDF v5.5-dev-1942-g1160a86ba0 2nd stage bootloader
I (27) boot: compile time Mar  3 2025 22:28:53
I (27) boot: Multicore bootloader
D (29) bootloader_flash: non-XMC chip detected by SFDP Read (00), skip.
D (35) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
[... 882 lines cut ...]
Kernel Name              : \Device\00000196
Device ID                : HID\VID_19F5&PID_6120&MI_02&COL05\7&1C15E5C5&0&0004
Hardware IDs             : HID\VID_19F5&PID_6120&REV_0104&MI_02&Col05 HID\VID_19F5&PID_6120&MI_02&Col05 HID\VID_19F5&UP:0001_U:000C HID_DEVICE_UP:0001_U:000C HID_DEVICE
Driver KeyName           : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}\0053 (GUID_DEVCLASS_HIDCLASS)
Driver Inf               : C:\windows\inf\input.inf
Legacy BusType           : PNPBus
Class                    : HIDClass
Class GUID               : {745a17a0-74d3-11d0-b6fe-00a0c90f57da} (GUID_DEVCLASS_HIDCLASS)
Enumerator               : HID
Location Info            : -
Address                  : 5
Manufacturer Info        : (Standard system devices)
Capabilities             : 0xE0 (SilentInstall, RawDeviceOK, SurpriseRemovalOK)
Status                   : 0x0180200A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
EnhancedPowerMgmtEnabled : 0
Power State              : D0 (supported: D0, D1, D2, D3, wake from D0, wake from D1, wake from D2)

             ++++++++++++++ HID Information +++++++++++++++
Manufacturer             : NuPhy
Product                  : NuPhy Air75 HE
Serial Number            : 13BC609A0594
UsagePage                : 0x01 (Generic Desktop Controls)
Usage                    : 0x0C (unk)
```

roma-jam, 2025-03-05:
Hi @alex-k8 ,

could you try to increase Reset recovery delay to 500 ms and try again? 

This parameter could be found in menuconfig: Component config → USB-OTG → Hub Driver Configuration → Root Port configuration.

alex-k8, 2025-03-06:
Hey @roma-jam 

Thanks for the suggestion, however it seems to be doing the same thing. Judging by these two lines and the timestamp, I suppose I did set the value correctly, right?
D (3295) ENUM: [0:0] CHECK_SHORT_DEV_DESC OK[0m
D (3825) ENUM: [0:0] SECOND_RESET OK[0m

```
[0;33mW (4155) USB host lib: To shutdown example, remove all USB devices and press button again.[0m
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x21d4
load:0x403c8700,len:0xed8
load:0x403cb700,len:0x3468
entry 0x403c8974
I (26) boot: ESP-IDF v5.5-dev-1942-g1160a86ba0 2nd stage bootloader
I (27) boot: compile time Mar  3 2025 22:28:53
I (27) boot: Multicore bootloader
D (29) bootloader_flash: non-XMC chip detected by SFDP Read (00), skip.
D (35) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (41) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (48) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (54) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (60) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (67) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (73) boot: chip revision: v0.2
I (76) boot: efuse block revision: v1.3
D (80) boot.esp32s3: magic e9
D (82) boot.esp32s3: segments 03
D (85) boot.esp32s3: spi_mode 02
D (88) boot.esp32s3: spi_speed 0f
D (91) boot.esp32s3: spi_size 01
I (94) boot.esp32s3: Boot SPI Speed : 80MHz
I (98) boot.esp32s3: SPI Mode       : DIO
I (102) boot.esp32s3: SPI Flash Size : 2MB
D (106) boot: Enabling RTCWDT(9000 ms)
I (109) boot: Enabling RNG early entropy source...
D (114) bootloader_flash: rodata starts from paddr=0x00008000, size=0xc00, will be mapped to vaddr=0x3c000000
V (123) bootloader_flash: after mapping, starting from paddr=0x00000000 and vaddr=0x3c000000, 0x10000 bytes are mapped
D (134) boot: mapped partition table 0x8000 at 0x3c008000
D (139) flash_parts: partition table verified, 4 entries
I (144) boot: Partition Table:
I (147) boot: ## Label            Usage          Type ST Offset   Length
D (153) boot: load partition table entry 0x3c008000
D (158) boot: type=1 subtype=2
I (161) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (167) boot: load partition table entry 0x3c008020
D (172) boot: type=1 subtype=1
I (175) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (181) boot: load partition table entry 0x3c008040
D (186) boot: type=0 subtype=0
I (189) boot:  2 factory          factory app      00 00 00010000 00100000
I (195) boot: End of partition table
D (198) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (204) esp_image: reading image header @ 0x10000
D (209) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (215) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (222) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (228) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (235) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (241) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (248) esp_image: image header: 0xe9 0x06 0x02 0x01 40375400
V (253) esp_image: loading segment header 0 at offset 0x10018
D (259) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (265) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
V (272) esp_image: segment data length 0xcdf4 data starts 0x10020
V (278) esp_image: MMU page size 0x10000
V (281) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3c020020
[... 221 lines cut ...]
D (3825) ENUM: [0:0] SECOND_RESET_COMPLETE OK[0m
D (3825) ENUM: [0:0] SET_ADDR OK[0m
D (3825) USBH: Processing actions 0x4[0m
D (3825) USBH: Default pipe device 0[0m
V (3825) ENUM: 0x3fceb324   00 05 01 00 00 00 00 00                           |........|[0m
D (3835) ENUM: Assign address (dev_addr=1)[0m
D (3845) ENUM: [0:0] CHECK_ADDR OK[0m
D (3855) ENUM: [0:0] SET_ADDR_RECOVERY OK[0m
D (3855) ENUM: [0:0] GET_FULL_DEV_DESC OK[0m
D (3855) USBH: Processing actions 0x4[0m
D (3855) USBH: Default pipe device 1[0m
V (3855) ENUM: 0x3fceb324   80 06 00 01 00 00 12 00  12 01 00 02 00 00 00 40  |...............@|[0m
V (3865) ENUM: 0x3fceb334   f5 19 20 61 04 01 01 02  03 01                    |.. a......|[0m
D (3875) ENUM: [0:0] CHECK_FULL_DEV_DESC OK[0m
D (3885) ENUM: Selected bConfigurationValue=1[0m
D (3885) ENUM: [0:0] SELECT_CONFIG OK[0m
D (3885) ENUM: [0:0] GET_SHORT_CONFIG_DESC OK[0m
D (3895) USBH: Processing actions 0x4[0m
D (3895) USBH: Default pipe device 1[0m
V (3905) ENUM: 0x3fceb324   80 06 00 02 00 00 08 00  00 00 00 00 00 00 00 00  |................|[0m
[0;31mE (3905) ENUM: Short config desc has wrong bDescriptorType[0m
[0;31mE (3915) ENUM: [0:0] CHECK_SHORT_CONFIG_DESC FAILED[0m
D (3925) ENUM: [0:0] CANCEL OK[0m
D (3925) HUB: Disabling root port[0m
```

roma-jam, 2025-03-06:
Hi @alex-k8, 

thanks for trying this, sometimes devices need a bit more time after reset to be able to return the Configuration Descriptor. But this is not our case. 

May I ask you to change the logic of getting short descriptor in the file [esp-idf/components/usb/enum.c, line 370](https://github.com/espressif/esp-idf/blob/master/components/usb/enum.c#L370)? 

Just replace the whole case with this: 

```
    case ENUM_STAGE_GET_SHORT_CONFIG_DESC: {
        // Get a short config descriptor at descriptor index
        USB_SETUP_PACKET_INIT_GET_CONFIG_DESC((usb_setup_packet_t *)transfer->data_buffer, desc_index, 255  /* ENUM_SHORT_DESC_REQ_LEN */);
        transfer->num_bytes = sizeof(usb_setup_packet_t) + usb_round_up_to_mps(255 /* ENUM_SHORT_DESC_REQ_LEN */, ctrl_ep_mps);
        // IN data stage should return exactly ENUM_SHORT_DESC_REQ_LEN bytes
        p_enum_driver->single_thread.expect_num_bytes = 0 /* sizeof(usb_setup_packet_t) + ENUM_SHORT_DESC_REQ_LEN */;
        break;
    }
```

And share the output.  You can share only the lines after tne `ENUM: [0:0] Start processing, device address 0`, everything else is not relevant right now. 

Thank you.

alex-k8, 2025-03-06:
Thanks @roma-jam 

Still getting the same issue. It did return more bytes (zeros) so I suspect the change was successful, but I'm still getting the `Short config desc has wrong bDescriptorType` error. Doesn't seem to be related to the number of bytes, but the values of them (?)

```
D (3276) ENUM: [0:0] Start processing, device address 0[0m
D (3276) ENUM: [0:0] GET_SHORT_DEV_DESC OK[0m
D (3276) USBH: Processing actions 0x4[0m
D (3276) USBH: Default pipe device 0[0m
V (3286) ENUM: 0x3fceb324   80 06 00 01 00 00 08 00  12 01 00 02 00 00 00 40  |...............@|[0m
D (3296) ENUM: [0:0] CHECK_SHORT_DEV_DESC OK[0m
D (3826) ENUM: [0:0] SECOND_RESET OK[0m
D (3826) ENUM: [0:0] SECOND_RESET_COMPLETE OK[0m
D (3826) ENUM: [0:0] SET_ADDR OK[0m
D (3826) USBH: Processing actions 0x4[0m
D (3826) USBH: Default pipe device 0[0m
V (3826) ENUM: 0x3fceb324   00 05 01 00 00 00 00 00                           |........|[0m
D (3836) ENUM: Assign address (dev_addr=1)[0m
D (3846) ENUM: [0:0] CHECK_ADDR OK[0m
D (3856) ENUM: [0:0] SET_ADDR_RECOVERY OK[0m
D (3856) ENUM: [0:0] GET_FULL_DEV_DESC OK[0m
D (3856) USBH: Processing actions 0x4[0m
D (3856) USBH: Default pipe device 1[0m
V (3856) ENUM: 0x3fceb324   80 06 00 01 00 00 12 00  12 01 00 02 00 00 00 40  |...............@|[0m
V (3866) ENUM: 0x3fceb334   f5 19 20 61 04 01 01 02  03 01                    |.. a......|[0m
D (3876) ENUM: [0:0] CHECK_FULL_DEV_DESC OK[0m
D (3886) ENUM: Selected bConfigurationValue=1[0m
D (3886) ENUM: [0:0] SELECT_CONFIG OK[0m
D (3886) ENUM: [0:0] GET_SHORT_CONFIG_DESC OK[0m
D (3896) USBH: Processing actions 0x4[0m
D (3896) USBH: Default pipe device 1[0m
V (3906) ENUM: 0x3fceb324   80 06 00 02 00 00 ff 00  00 00 00 00 00 00 00 00  |................|[0m
V (3906) ENUM: 0x3fceb334   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|[0m
V (3916) ENUM: 0x3fceb344   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|[0m
V (3926) ENUM: 0x3fceb354   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|[0m
V (3936) ENUM: 0x3fceb364   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|[0m
V (3946) ENUM: 0x3fceb374   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|[0m
V (3956) ENUM: 0x3fceb384   00 00 00                                          |...|[0m
[0;31mE (3966) ENUM: Short config desc has wrong bDescriptorType[0m
[0;31mE (3966) ENUM: [0:0] CHECK_SHORT_CONFIG_DESC FAILED[0m
D (3976) ENUM: [0:0] CANCEL OK[0m
D (3976) HUB: Disabling root port[0m
```

roma-jam, 2025-03-06:
Hi @alex-k8 , thanks for the info. 

Yes, the device returns appropriate amount of bytes (91, which is 0x5B - right wTotalLengh value for Config Descriptor), but  the questions is why the data is invalid. 

Do you have any chance to capture the Wireshark log while attaching this device to the PC Host?

alex-k8, 2025-03-06:
@roma-jam unfortunately I don't own the device, I'm getting the logs from my customer who is using my custom board. I can assist him in getting the logs if you can send me some high level instructions.

From my understanding, the short config response is used to store the wTotalLength value, and then that's used in the full config request? Couldn't we hardcode the 0x005B to see if it proceeds and works as expected after that step?

roma-jam, 2025-03-06:
@alex-k8 , 

> short config response is used to store the wTotalLength value, and then that's used in the full config request?

yes, there is no information about the full length, so we request first 8 bytes (short), get the full length and then use this value to request the whole Configuration Descriptor.

> Couldn't we hardcode the 0x005B

We could, indeed, but that doesn't make any real sense, as this will be the same as request the maximum - 0xff (or 255) which we just tried. 

> if you can send me some high level instructions

I don't think that I will provide any better than the Wireshark themselves: https://wiki.wireshark.org/CaptureSetup/USB
Anyway, the process in oversimplified way should be: install the software, select the capture source (usb), attach the device, wait till it appears in system, save log.

Then, it is better to filter the log, but if that is not possible by any reasons we can filter it on our side and get the information we need. So, both (filtered / non-filtered) are OK. 

If you need anything more precise, let me know.

alex-k8, 2025-03-06:
@roma-jam 

> yes, there is no information about the full length, so we request first 8 bytes (short), get the full length and then use this value to request the whole Configuration Descriptor.

Oh I see, so it's actually passing the GET_SHORT_CONFIG_DESC and the issue is in the full descriptor?

Thanks for the link, I will see what I can do and get back to you soon!

roma-jam, 2025-03-06:
@alex-k8, 

in case to prevent any misunderstandings - there are no requirements for any additional hardware to capture the usb log with Wireshark: only the device itself and PC Host. 

Feel free to ask any questions in case they appear. 

Thanks for the understanding and collaboration.

alex-k8, 2025-03-06:
Yup, thanks @roma-jam . I had used it once before to figure out what the computer sends to turn on the caps lock LED on keyboards, so I'll figure it out.

Thank you for your help.

alex-k8, 2025-03-22:
Hey @roma-jam , I managed to get some Wireshark logs. Unfortunately they are not filtered, but the logs with ID 1.17.0 should be the relevant ones. Let me know if you need anything else from me. Github wouldnt allow me to upload the .pcapng file, so I zipped it.

[log (1).zip](https://github.com/user-attachments/files/19405752/log.1.zip)

roma-jam, 2025-03-22:
@alex-k8, 

Thanks, log from the Host was from the High-speed port.....S3 has a Full-speed. That might be relevant. 
Is there any chance to verify the device in the other Full-speed port (not esp)? 

In the log the request configuration descriptor is done in two steps: first request is 9 bytes only, then - the `wTotalLength` (which is 91 or 0x5b). 

We can try to do the same:  change short amount from 8 to 9 bytes and check whether it helps.

To do that, it is possible just to change the define in `enum.c`: 

`#define ENUM_SHORT_DESC_REQ_LEN                     9 // Number of bytes to request when getting a short descriptor (just enough to get bMaxPacketSize0 or wTotalLength)`

alex-k8, 2025-03-23:
Hey @roma-jam thanks for the quick response.

Not sure what you mean by `other full-speed port`? You mean a USB 2.0 port on the computer? Unfortunately my customer who has the UHK only has USB 3.0 ports.

I did some tests by changing the ENUM_SHORT_DESC_REQ_LEN, but it didn't seem to help unfortunately. Below are the logs

```
D (3276) ENUM: [0:0] Start processing, device address 0[0m
D (3276) ENUM: [0:0] GET_SHORT_DEV_DESC OK[0m
D (3276) USBH: Processing actions 0x4[0m
D (3276) USBH: Default pipe device 0[0m
V (3286) ENUM: 0x3fceb324   80 06 00 01 00 00 09 00  12 01 00 02 00 00 00 40  |...............@|[0m
V (3296) ENUM: 0x3fceb334   f5                                                |.|[0m
D (3296) ENUM: [0:0] CHECK_SHORT_DEV_DESC OK[0m
D (3836) ENUM: [0:0] SECOND_RESET OK[0m
D (3836) ENUM: [0:0] SECOND_RESET_COMPLETE OK[0m
D (3836) ENUM: [0:0] SET_ADDR OK[0m
D (3836) USBH: Processing actions 0x4[0m
D (3836) USBH: Default pipe device 0[0m
V (3836) ENUM: 0x3fceb324   00 05 01 00 00 00 00 00                           |........|[0m
D (3846) ENUM: Assign address (dev_addr=1)[0m
D (3856) ENUM: [0:0] CHECK_ADDR OK[0m
D (3866) ENUM: [0:0] SET_ADDR_RECOVERY OK[0m
D (3866) ENUM: [0:0] GET_FULL_DEV_DESC OK[0m
D (3866) USBH: Processing actions 0x4[0m
D (3866) USBH: Default pipe device 1[0m
V (3866) ENUM: 0x3fceb324   80 06 00 01 00 00 12 00  12 01 00 02 00 00 00 40  |...............@|[0m
V (3876) ENUM: 0x3fceb334   f5 19 20 61 04 01 01 02  03 01                    |.. a......|[0m
D (3886) ENUM: [0:0] CHECK_FULL_DEV_DESC OK[0m
D (3896) ENUM: Selected bConfigurationValue=1[0m
D (3896) ENUM: [0:0] SELECT_CONFIG OK[0m
D (3896) ENUM: [0:0] GET_SHORT_CONFIG_DESC OK[0m
D (3906) USBH: Processing actions 0x4[0m
D (3906) USBH: Default pipe device 1[0m
V (3916) ENUM: 0x3fceb324   80 06 00 02 00 00 09 00  00 00 00 00 00 00 00 00  |................|[0m
V (3916) ENUM: 0x3fceb334   00                                                |.|[0m
[0;31mE (3926) ENUM: Short config desc has wrong bDescriptorType[0m
[0;31mE (3936) ENUM: [0:0] CHECK_SHORT_CONFIG_DESC FAILED[0m
D (3936) ENUM: [0:0] CANCEL OK[0m
D (3946) HUB: Disabling root port[0m
```

roma-jam, 2025-03-24:
@alex-k8 , 
thanks for the additional information. 

> You mean a USB 2.0 port on the computer?

Yes, some device (maybe some old laptop, but I guess it should be really old, so maybe some other embedded device) with the port, that supports Full-speed. Because the keyboard has the usb-c connected and the behavior on Full-speed it quite strange: it reports the right amount of bytes, but no data.

Based on the Wireshark log, there is no specific in the interaction.

Also, could be the power problem. So let's exclude it: is there any chance to try connect the keyboard via external hub with the external power supply?

alex-k8, 2025-03-24:
Hi @roma-jam thank you.

I got some logs from a usb2 port, not sure if they are too much different. ID should be 1.2.0.

Unfortunately no power hub is available. But my PCB operates both in power mode (through the type c) and through battery. Same issue on both.

I read some things as well, but keep in mind this is the Air75HE and not the Air75. There is a switch that toggles between Windows/Linux/Mac, but unfortunately again it's the same issue on all. It doesn't seem to change anything, only some key bindings probably. There is no power switch.

Thank you for taking time to look at this.

[log (2).zip](https://github.com/user-attachments/files/19438633/log.2.zip)

alex-k8, 2025-04-03:
Hey @roma-jam, any luck? Any more ideas? Did you get a chance to look at the logs? Thanks!

roma-jam, 2025-04-04:
Hi @alex-k8 

sorry for the delay, the log is indeed the same from anything that might help. Thanks for the log, but unfortunately no new information for us. 

I would recommend to try to find external usb hub and attach the device via it. The better option that the hub should be powered externally. 

Otherwise, is there a chance to try to connect the keyboard to any esp dev kit?

alex-k8, 2025-04-13:
Hey @roma-jam 

Tried with an external hub (unfortunately not externally powered, I couldnt find one), and it's doing exactly the same.

Since I don't have direct access to the keyboard, sadly no, I can't try an esp dev kit. Any other ideas? Thanks!

roma-jam, 2025-05-23:
@alex-k8 , 

sorry for the late response. 

If the device doesn't support USB 1.1, that explains why it returns all zeroes on correct request. Unfortunately, there is not so much we can do here, because the S3 has exactly the USB 1.1 peripheral. 

Yes, P4 has two peripheral and one of them is USB 2.0 peripheral, so the device should work there without any problems.

@chipweinberger, 

> sounds like a similar issue here: https://github.com/espressif/esp-idf/issues/15304

not really, in 15304 the device doesn't respond anything, and here the device respond exactly the right amount of bytes, but all of them are zeroes.

roma-jam, 2025-06-04:
I'll close the issue, because there is nothing wrong with the the USB Host Library. 

If the device doesn't support USB 1.1 the behavior could be any.  

Feel free to re-open if there will be anything similar on P4.

closed 2025-06-04 by roma-jam