espressif/esp-idf #15528: How to separate the console port from the log port

hsw-li, 2025-03-06:
### Is your feature request related to a problem?

开发板：esp thread border router/zigbee gateway v1.2
version： 5.5.0
project：esp-thread-br
开机启动的时候，启动log和console都是一个串口。会刷新大量的log日志
我希望将开机启动log和console分开，一个输出到uart1,一个通过uart0来交互
目前我借助menuconfig的console配置选项，打开了第二辅助输出串口，但是开机启动的log还会同时输出到两个串口，我希望日志只能输出到一个串口uart1上，uart0只做console,只会输出我进行command后的信息
期望实现图2的效果

[image]

### Describe the solution you'd like.

他应该不是一下新功能，根据我的了解，似乎再几年前的esp-idf版本上实现过这种功能。但是我在5.5.0上没有配置成功，是我的配置方式有问题吗？又或者在issue中有人已经给出了解决方法。请贴个链接给我

我尝试过检索issue,有相关的信息，有人提到过类似的问题，但是和我想要的功能似乎不太一致。
请原谅，我的英语不太能支撑我在issue信息中找到对应的方法，或许是我漏掉了。

Alvin1Zhang, 2025-03-06:
Thanks for reporting.

hsw-li, 2025-03-07:
> Thanks for reporting.

你好，目前的5.5.0版本有办法实现这个console和log分开指向不同串口的功能吗？

KonstantinKondrashov, 2025-04-04:
I'm closing this issue for now. Feel free to reopen it if needed or leave a comment if you have any further questions.

closed 2025-04-04 by KonstantinKondrashov