espressif/esp-idf #15529: Device target linux causes unit testing order to reverse from on chip order

i400s, 2025-03-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.5-dev-2182-ga6c3a9cbbb

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

When running a unit test program it should call the TEST_CASE's in the order in which they are entered.

While individual unit tests should be entirely standalone (setting up and deleting their own dependencies) it is often the case that each test might build on the previous one (not data wise, logic wise). A group of tests might first test the i2c bus can be created. A second test might create an i2c bus and add a device. A third test might create an i2c bus, add a device, and check the device is present... etc.

This is how it works when you compile "on chip".

### What is the actual behavior?

When monitoring a unit test program that is compiled to run "on chip" the TEST_CASE's are run in the order in which they are entered.

When the same program is compiled to run on the host the TEST_CASE's are run in reverse order.

### Steps to reproduce.

1. Create a new project using the unit_test as the base
2. Compile the unit_test "test" to run on chip
3. Run program
4. Output is in logical "entered" order
5. Compile the unit_test "test" to run on linux
6. Run program
7. Output is in reversed order

### Build or installation Logs.

```plain

```

### More Information.

Output on esp32c6 after pressing enter:
```(1)     "Mean of an empty array is zero" [mean]
(2)     "Mean of a test vector" [mean]
(3)     "Another test case which fails" [mean][fails]
```

Output on linux after pressing Enter:
```(1)     "Another test case which fails" [mean][fails]
(2)     "Mean of a test vector" [mean]
(3)     "Mean of an empty array is zero" [mean]```

i400s, 2025-03-06:
Hi @KonstantinKondrashov 

I did wonder about my wording of that bit. What I meant was standalone tests going a bit further each time. So test 1 creates the thing = pass. Test 2 creates the thing and then uses it with wrong parameters = fail = pass. Then test 3 creates the thing and uses it correctly = pass. So each test is testing a part of the same thing, all created from scratch each time with no interdependence on previous tests.

Anyways. Thanks for looking into this. Much appreciated.

KonstantinKondrashov, 2025-04-22:
Hi @i400s!
There is a fix for this issue. Could you apply the fix (remember to set the CONFIG_UNITY_TEST_ORDER_BY_FILE_PATH_AND_LINE=y)?
[38616.patch](https://github.com/user-attachments/files/19848056/38616.patch)
Thanks.

i400s, 2025-04-22:
Downloaded clean esp-idf and applied patch.

Using my own project (not the unit test example):
With device set to "linux" and with the config not set the order is "reversed" as it was before and with the config set the order is now in coded sequence order.
With device set to "esp32c6" the config setting makes no difference, its always in coded sequence order.

Using the unit test example project:
With device set to "linux" and with the config not set the order is "reversed" as it was before and with the config set the order is now in coded sequence order.
With device set to "esp32c6" the config setting makes no difference, its always in coded sequence order.

So I can confirm the patch fixes the order issue for "linux".

One point of note, I noticed your patch sets up a pytest_unittest_order.py in addition to the coding modifications. I was unable to test that as I have no idea how to run it but I'm guessing it has something to do with your CI/CD?

KonstantinKondrashov, 2025-04-22:
@i400s 
Yes, the test in the patch is for our CI testing only. Thank you for the comprehensive testing it. 

This is not the final version because it is the review stage. I created the Kconfig option and disabled it by default just in case if it breaks the order that other users are already used to. Maybe we will make it enabled by default, I do not know yet.

This issue will be closed when it is merged, I hope soon.
(We will not backport it to the previous esp-idf versions).

closed 2025-04-27 by espressif-bot

i400s, 2025-04-28:
My thanks to everyone for the help on this. Much appreciated.