espressif/esp-idf #15531: WiFi and LWIP layer got lost - ESP32 Zombie State – WiFi & BLE Stack Corruption Causing Illusory Connectivity

filzek, 2025-03-07:
using: commit d4aa25a38e9c8659a5e5f4704d31ff532d807de0 (HEAD -> release/v5.4, origin/release/v5.4)
this problem still happen for a long time since sdk 4.

Today we made a new discovery about the WiFi and BLE stacks in the ESP32 ESP-IDF. Under a certain catastrophic sequence of events—still to be fully reproduced—the system enters what we’re calling a ZOMBIE state.

### **What Is a Zombie State?**
BLE and WiFi appear dead:
The BLE stack has died, and the WiFi stack is non-responsive.

**WiFi stack illusion:**
Even though the WiFi stack believes it is connected to the access point (AP) and still holds a valid IP address, it is not truly connected. The LWIP HTTP server stops responding, and while the WiFi AP briefly shows the device as connected, it soon no longer does so.

**Misleading connection status:**
The ESP32 continues to think it is connected to WiFi, even though it is not.

### Our Findings and Workarounds
**Handler Code:**
We implemented code in the event handlers to destroy the WiFi layer upon disconnection. However, because the WiFi stack still thinks it is connected, the handler is only called when the WiFi router is rebooted.

**Temporary Fix:**
We were able to restore functionality by simply restarting the WiFi. This allowed us to connect to the unresponsive HTTP server since the device was not truly connected to WiFi.

**Logging for Analysis:**
We captured debug logs to help us fully understand the problem. It appears that the WiFi layer has a significant issue in detecting when it has corrupted itself—believing it’s connected and has a valid IP when it actually isn’t.

**Additional Logging:**
To gather more insight, we added a function to log errors during the WIFI_EVENT_STA_DISCONNECTED event. This helps us identify when the connection status is inconsistent (i.e., when the WiFi router restarts and the zombie state is active).

**Next Steps**
We now need to determine a reliable way to detect this condition so that we can programmatically perform a WiFi stop, start, and reconnect sequence to prevent the zombie state from persisting.

We need that the driver when get lost has a mechanism that tell us it, so we can do something later.

How can we find out trully tha this happen so far?

beibeisi, 2025-03-14:
Hello, @filzek 
You can use the `MQTT KeepAlive`  to detect whether the ESP32 has entered a Zombie State. The ESP32 can monitor the `MQTT_EVENT_DISCONNECTED` event to reconnect ap. Additionally, could you provide the logs and packet captures  when the issue occurs? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

filzek, 2025-03-14:
We currently do not have a Wi-Fi monitoring-capable device, but we have ordered one, and it is expected to arrive next week.

The issue occurs in field devices, making it extremely difficult to reproduce in a controlled lab environment. While we cannot currently capture logs, we do know that the problem is affecting approximately 1% or more of our deployed devices, with over 300,000 units already in the field.

From our testing, we observed that out of 30 devices running for 4 days, at least 1 device encountered this issue.

Once the monitoring device arrives next week, we will attempt to reproduce the issue and capture 802.11 frames from all affected devices. We plan to set up a Kali Linux system for full packet capture, logging all MAC addresses. If a device enters the "zombie mode," we should be able to analyze the captured frames and provide further insights.

closed 2025-11-26 by Alvin1Zhang