espressif/esp-idf #15532: Why gpio34 can't receive input level?

Ares-bit, 2025-03-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**board:**     ESP32-WROOM-32
**ide:**          ESP-IDF
**version:**   v5.2

**expect:**
I connect a button between VIN and GPIO34（input mode, posedge trigger）, and connect a LED light to GPIO27(output mode). I expect that when I press the button, GPIO34 should give an interrupt and set GPIO27 to high level to turn on the LED light. 

**fact:**
I don't press the button, GPIO34 continuely give interrupts by itself.

**question:**
Why this would happen? I don't even press the button, GPIO34 should have no input, where does it receive input? What should I do to stop GPIO34 interrupt?

songruo, 2025-03-07:
To debug such issue, please first disconnect any external hardware from the board. Make sure no continuous GPIO interrupt will gets triggered with the devkit itself.

How did you initialize the IO and register the interrupt? Did you take the [generic_gpio](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/gpio/generic_gpio/main/gpio_example_main.c) example as a reference?

songruo, 2025-03-10:
@Ares-bit I tried your code on release/v5.2 with an ESP32 devkit. I could not reproduce, I didn't see the interrupt keep getting triggered. Please provide more information, or check your hardware.

songruo, 2025-03-12:
Maybe try with removing the devkit from the bread board? Or even you can use a wire to connect IO34 to GND.

Ares-bit, 2025-03-12:
> Maybe try with removing the devkit from the bread board? Or even you can use a wire to connect IO34 to GND.

Thx, I remove the devkit from bread board, it stopped. The reason is GPIO34 is too sensitive, but when I touch it with myu finger, it get triggered. Is this normal?

Ares-bit, 2025-03-13:
> IO34 is an input only GPIO, therefore, the internal pullup/pulldown resistor does not take effect. You should ensure the IO is pulled to GND in your external circuit when the button is not pressed.

So I can't use 34-39 as input right? I can't pull IO34 to GND  when the button is not pressed, because they are wired on bread board.

songruo, 2025-03-14:
IO34-39 can be used as inputs, but you need external pullup / pulldown to avoid the pin floating. Well, since you are probably just doing a simple demo, I guess the easiest way for you is to change to an IO that has internal pullup / pulldown, so that you don't need to take care of it externally.

Ares-bit, 2025-03-14:
> IO34-39 can be used as inputs, but you need external pullup / pulldown to avoid the pin floating. Well, since you are probably just doing a simple demo, I guess the easiest way for you is to change to an IO that has internal pullup / pulldown, so that you don't need to take care of it externally.

If I have to use GPIO34 in a project, I should connect it to GND? But if I do this, how to connect GPIO34 to another peripheral?

songruo, 2025-03-17:
I mean you should make sure when the button is not pressed, the IO is pulled to GND. Something like the following:

```
      3.3V

         |

         |

        [Button]

         |

         |-------> GPIO34 (ESP32)

         |

        [10kΩ]

         |

        GND

 
```

closed 2025-03-19 by songruo