espressif/esp-idf #15533: I2S fails to start when CONFIG_I2S_ISR_IRAM_SAFE is OFF and PSRAM is enabled

me-no-dev, 2025-03-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Any

### Power Supply used.

USB

### What is the expected behavior?

`i2s_register_channel()` to return a valid handle

### What is the actual behavior?

Channel handle is allocated in PSRAM instead of internal memory, which then causes `gdma_register_tx_event_callbacks()` to fail with "user context not in internal RAM"

### Steps to reproduce.

Start I2S on ESP32-S3 with PSRAM enabled and `CONFIG_I2S_ISR_IRAM_SAFE=n`

### Debug Logs.

```plain
E (541) gdma: gdma_register_tx_event_callbacks(468): user context not in internal RAM
E (542) i2s_common: i2s_init_dma_intr(802): Register tx callback failed
E (544) i2s_std: i2s_channel_init_std_mode(243): initialize dma interrupt failed
```

### More Information.

Reference: https://github.com/espressif/arduino-esp32/issues/11004

Bug is also present on IDF 5.3 and IDF 5.4

wanckl, 2025-03-11:
@me-no-dev  You can also disable `CONFIG_GDMA_ISR_IRAM_SAFE`

@L-KAYA how do you think about

slashinpdx, 2025-03-14:
Getting the same error(s) on a XIAO ESP32-S3 Sense, when trying to capture audio and video for RTSP.  Forgive my ignorance, but how does one disable CONFIG_GDMA_ISR_IRAM_SAFE in an Arduino project?  I've spent a couple of hours fruitless with google.

closed 2025-03-24 by espressif-bot