espressif/esp-idf #15535: SD Card example (SDSPI) not working correctly with ESP32-S3

donkiejkong, 2025-03-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have compiled and trying running the sd card example on the WAVESHARE ESP32-S3-GEEK (also detected as ESP32-S3-USB-OTG in arduino studio) board.

IDF verson: 5.4.0
Readme file says the default PINs should be compatible with my board.

However it returns following error:
I (282) example: Initializing SD card
I (282) example: Using SPI peripheral
I (282) example: Mounting filesystem
I (292) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (502) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (502) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (502) gpio: GPIO[8]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (502) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.

Fragments of the source code:

`
void app_main(void)
{
    esp_err_t ret;

    // Options for mounting the filesystem.
    // If format_if_mount_failed is set to true, SD card will be partitioned and
    // formatted in case when mounting fails.
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
#ifdef CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .format_if_mount_failed = true,
#else
        .format_if_mount_failed = false,
#endif // EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };
    sdmmc_card_t *card;
    const char mount_point[] = MOUNT_POINT;
    ESP_LOGI(TAG, "Initializing SD card");

    // Use settings defined above to initialize SD card and mount FAT filesystem.
    // Note: esp_vfs_fat_sdmmc/sdspi_mount is all-in-one convenience functions.
    // Please check its source code and implement error recovery when developing
    // production applications.
    ESP_LOGI(TAG, "Using SPI peripheral");

    // By default, SD card frequency is initialized to SDMMC_FREQ_DEFAULT (20MHz)
    // For setting a specific frequency, use host.max_freq_khz (range 400kHz - 20MHz for SDSPI)
    // Example: for fixed frequency of 10MHz, use host.max_freq_khz = 10000;
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();

    // For SoCs where the SD power can be supplied both via an internal or external (e.g. on-board LDO) power supply.
    // When using specific IO pins (which can be used for ultra high-speed SDMMC) to connect to the SD card
    // and the internal LDO power supply, we need to initialize the power supply first.
#if CONFIG_EXAMPLE_SD_PWR_CTRL_LDO_INTERNAL_IO
    sd_pwr_ctrl_ldo_config_t ldo_config = {
        .ldo_chan_id = CONFIG_EXAMPLE_SD_PWR_CTRL_LDO_IO_ID,
    };
    sd_pwr_ctrl_handle_t pwr_ctrl_handle = NULL;

    ret = sd_pwr_ctrl_new_on_chip_ldo(&ldo_config, &pwr_ctrl_handle);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to create a new on-chip LDO power control driver");
        return;
    }
    host.pwr_ctrl_handle = pwr_ctrl_handle;
#endif

    
    spi_bus_config_t bus_cfg = {
        .mosi_io_num = PIN_NUM_MOSI,
        .miso_io_num = PIN_NUM_MISO,
        .sclk_io_num = PIN_NUM_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 4000,
    };

 /*
 spi_bus_config_t bus_cfg = {
        .mosi_io_num = GPIO_NUM_35,
        .miso_io_num = GPIO_NUM_37,
        .sclk_io_num = GPIO_NUM_36,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 4000,
    };
*/

    //ESP_LOGE(TAG, "USING PIN_NUM_MOSI=%d PIN_NUM_MISO=%d PIN_NUM_CLK=%d PIN_NUM_CS=%d\n", PIN_NUM_MOSI, PIN_NUM_MISO, PIN_NUM_CLK, PIN_NUM_CS);

    ret = spi_bus_initialize(host.slot, &bus_cfg, SDSPI_DEFAULT_DMA);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize bus.");
        return;
    }

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
    /*
    slot_config.gpio_cs = PIN_NUM_CS;
    slot_config.host_id = host.slot;
    */

    ESP_LOGI(TAG, "Using SPI peripheral");

    //sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_DEFAULT();
    

    ESP_LOGI(TAG, "Mounting filesystem");
    ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                     "If you want the card to be formatted, set the CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                     "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
#ifdef CONFIG_EXAMPLE_DEBUG_PIN_CONNECTIONS
            check_sd_card_pins(&config, pin_count);
#endif
        }
        return;
    }
    ESP_LOGI(TAG, "Filesystem mounted");
`
 
The function that fails: esp_vfs_fat_sdspi_mount
Why does it happen? How to fix it? (i've tried different PINs without any success, my SD card works correctly, tried with 3 different brand and sizes 4GB, 8GB, 32GB)

wanckl, 2025-03-10:
@donkiejkong Hi, it should transaction issues, as log says `Make sure SD card lines have pull-up resistors in place.`

> my SD card works correctly, tried with 3 different brand and sizes

So how do you try those card, with card reader or other esp dev kit? the card can work on sdio mode or sdspi mode, they have different hardware requirement

igrr, 2025-03-10:
Could you please mention the brand and the model of the SD card? We are currently seeing this issue (`sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108`) on a several SD card models.

donkiejkong, 2025-03-10:
sure:
- TRANSCENDENT 4GB MICROSD 
- KINGSTON Canvas Select Plus micro SDXC 64GB
- KINGSTON Canvas Select Plus micro SDXC 32GB

all those cards work in my ATMEGA boards and with microsd adapters. So am I missing in the code that needs to be configured or what?

igrr, 2025-03-10:
I can confirm we also see this issue on Canvas Select Plus cards. Not exactly sure of the root cause, we'll update the issue when we find the reason.

donkiejkong, 2025-03-10:
do you have any WORKING cards / models (meaning those that work 100%) that i can use for testing?

donkiejkong, 2025-03-12:
Same problem with SANDISK ULTRA 32GB SDXC UHS-I

adokitkat, 2025-03-17:
Sandisk microSD 32GB either Ultra or Extreme/Extreme Pro (sorry I don't remember which one) work for me.

adokitkat, 2025-03-17:
This is the example provided directly by Waveshare, could you try that as well (maybe with different IDF versions if the one you're using is not working)? 

[waveshare_usb_msc_wireless_disk_example.zip](https://github.com/user-attachments/files/19285971/waveshare_usb_msc_wireless_disk_example.zip)

donkiejkong, 2025-03-24:
> This is the example provided directly by Waveshare, could you try that as well (maybe with different IDF versions if the one you're using is not working)?
> 
> [waveshare_usb_msc_wireless_disk_example.zip](https://github.com/user-attachments/files/19285971/waveshare_usb_msc_wireless_disk_example.zip)

thank you! this seem to be working, i've one unrelated question, can this device work both as HID (mouse) and MASS STORAGE (SD) at the same time?

closed 2025-05-02 by adokitkat