espressif/esp-idf #15538: RCP not working over USB

MiguelCan13, 2025-03-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I have a pi 3 acting as a thread border router, essentially the expected behavior is the ESP32H2 will communicate with this, simply being able to ping it.

### What is the actual behavior?

When I try to build blink it seems to fail. Ive tried turning off minimul configuration but some http web server fails. I turned minimum config off, on, deleted it. I apologize, I am new to this.

### Steps to reproduce.

1. idf.py build
2.Error

### Build or installation Logs.

```plain

```

nichonaugle, 2025-03-08:
I have tried to replicate this and have the same issue. The application doesn't seem to allow communication over the USB or USB-to-UART lines from the host machine... @MiguelCan13

gytxxsy, 2025-06-24:
Now we have a temporary patch to support USB-based ot_rcp:
https://github.com/orgs/openthread/discussions/11627#discussioncomment-13558288

chshu, 2025-07-16:
It's supported since [51bcec7](https://github.com/espressif/esp-idf/commit/51bcec7ac67efff9e915239fa07fb1909b002d80).

closed 2025-07-16 by chshu