espressif/esp-idf #15540: Not exist method to switch modes to DIO, QUAD, Octal for SPI Slave.

ArthurKoba, 2025-03-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone. At the moment I am designing a heterogeneous system consisting of several ESP32 and ESP32-S3 microcontrollers for my PhD thesis. The future system implies constant transfer of a significant amount of data, so I decided to use SPI for communication. When I started studying the documentation and examples, I was able to organize data sending of all SPI Master submodes without any problems, namely standard duplex, and half-duplex modes DIO (2-bit), QUAD (4-bit) and OCTAL (8-bit). I have an oscilloscope, so I made sure that all modes work correctly.

However, when I tried to configure data reception in SPI Slave mode, only the standard full duplex mode worked correctly. According to the documentation, as well as examples, I could not find information on how to start the DIO, QUAD and OCTAL modes. Since I don't know much about the esp-idf framework yet, I double-checked everything several times and ended up searching the internet, where I found 3 threads on the esp32 forum where other engineers reported this problem:

1. [ESP32S2 - Quad SPI Slave / QPI Example + Notes](https://www.esp32.com/viewtopic.php?t=26207)
2. [SPI QUAD MODE](https://www.esp32.com/viewtopic.php?t=43548)
3. [SPI Dual mode, Half-Duplex using DMA - very strange results [IDFGH-4017]](https://www.esp32.com/viewtopic.php?t=17436)

When I read these topics, I realized that there is no solution at the moment, but I decided to check the source codes of the framework for additional verification. In the file `esp-idf/components/esp_driver_spi/include/driver.h` among the flags of the SPI slave transaction are only `SPI_SLAVE_TXBIT_LSBFIRST`, `SPILAVE_RXBIT_LSBFIRST`, `SPI_SLAVE_BIT_LSBFIRST`, `SPI_SLAVE_NO_RETURN_RESULT`, but the flag responsible for the DIO, QUAD and OCTAL mode, as is done in the SPI Master. For analogy, when sending data to the SPI Master in the transaction flags, the following flags can be used to switch the mode - `SPI_TRANS_MODE_DIO`, `SPI_TRANS_MODE_QIO`, `SPI_TRANS_MODE_OCT`. I also studied the initial codes of the functions responsible for initializing the SPI bus, the SPI slave device, as well as sending data receiving (spi_slave_transmit) and could not find something responsible for switching the SPI Slave mode. It is strange that this functionality is still missing, I think this can lead to problems for companies that decide to build systems based on ESP32 microcontrollers that participate in the exchange of large amounts of data.

Unfortunately, at the current time I do not have sufficient knowledge and experience to independently develop this code and contribute, but I am ready to help with testing, writing documentation and support in the implementation of this functionality.

wanckl, 2025-03-10:
@ArthurKoba Hi, thank you for detail description, pls see following reasons:
- For master, it is actively action the read or write.
- For slave, It is passive, however if setting DIO/QIO, it must not a full duplex transaction, then now master assert this device, should slave inout or output?

So you can see if you want a slave working as DIO/QIO, it acutally working as half-duplex, then you can find the flags you want in `spi_slave_hd` drivers. attention that this mode the master should send resbonsible `cmd` before starting a transaction. 

refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/spi_slave_hd.html

ArthurKoba, 2025-03-22:
@wanckl, Hi and thanks for your reply. When it comes to DIO/QIO and also Octal SPI I think it is always half-duplex. And when a programmer wants to program a chip using them, he must explicitly specify in the code which device is a master or a slave, and which one will send and receive data. Changes in these parameters can already be implemented by protocols, for example ESP SPI Slave HD or other proprietary ones. And the problem is that I don't mind using ESP SPI Slave HD, but it is not supported on ESP32 [[1]](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_spi_slave_protocol.html) which I need to use Bluetooth A2DP. That's why I'm looking for an option to use DIO/QIO and also Octal SPI without SPI Slave HD protocol in slave mode.

Thank you for your link, I will try to search for the logic of using DIO/QIO modes in the source codes of spi_slave_hd myself. But I think that I will not be able to independently, quickly and correctly implement software that will implement the necessary modes without ESP SPI Slave HD Mode Protocol.

wanckl, 2025-03-24:
@ArthurKoba Yeah, you are right, if app side already gurantee the direction, it may doable for slave driver using multi line mode. however driver don't support it till now.

we can have a check later.

For now If you will trying by yourselves, you can refer to `spi_ll_master_set_line_mode()`

ArthurKoba, 2025-03-24:
@wanckl, thanks for the tip. Once I finish working on my protocol, I'll see what I can do.

wanckl, 2025-07-07:
@ArthurKoba Not familiar about BLE, but is something chip related you did for your audio?

For peripheral side the esp-idf have unify API that you can change chip without any sense

wanckl, 2025-07-08:
@ArthurKoba So now you can close this issue if no other problem, thanks

closed 2025-07-09 by ArthurKoba

SerfinHub, 2025-08-01:
Hi @ArthurKoba, could you share an example of SPI octal master code for the ESP32-S3?

wanckl, 2025-08-04:
@SerfinHub Just config all the pins then using transaction flag `SPI_TRANS_MODE_OCT`

SerfinHub, 2025-08-04:
@wanckl  that about reading? I found that when I choose SPI_TRANS_MODE_OCT, it reports that it can only work in half-duplex mode. There’s no information about this in the documentation. ESP-IDF seems working that way. I’m trying to achieve high-speed OSPI communication with an FPGA and I’m wondering how to make it faster. I hope it could work similar to 4-wire SPI.

ArthurKoba, 2025-08-05:
@SerfinHub Hi, unfortunately I haven't saved the project in which I tested these modes. You asked about reading, if we are talking about esp32-s3 in octal slave spi mode - this is my problem, which we discussed here. Reading can be configured only in single-wire mode, or in octal mode but when using spi_slave_hd.