espressif/esp-idf #15541: esp_reset_reason() not working when set log level none (CONFIG_LOG_DEFAULT_LEVEL_NONE=y)

dzungpv, 2025-03-09:
I use esp_reset_reason() to detect crash on production firmware.
In my code I check the reason like ESP_RST_PANIC or ESP_RST_INT_WDT and switch device to safemode with limited hardware and peripheral install, because secure boot and flash encryption chip make it bricked if it enter boot loop, and there is no way to flash it.

It work fine if I set debug level from INFO to VERBOSE, it it will stop working when set:

```
CONFIG_LOG_DEFAULT_LEVEL_NONE=y
```

I don't know if this is a bug or by design.

KonstantinKondrashov, 2025-03-10:
Hi @dzungpv!
There is a constructor (esp_reset_reason_init) that runs at startup. This function reads the reset reason from the hardware register and then clears it, storing the value in the static variable s_reset_reason. The esp_reset_reason() API then retrieves this value, ensuring it remains accessible throughout the application runtime. The reset reason is only read from the register once at startup and then the register is cleared. The reset reason applies specifically to PRO_CPU_NUM (on dual-core chips).

https://github.com/espressif/esp-idf/blob/master/components/esp_system/port/soc/esp32/reset_reason.c#L59-L71

Possible reason could be that your application accesses the reset reason API very early in startup, before the reset reason contractor is called.

dzungpv, 2025-03-10:
As init comment. It work correctly if debug level other than NONE. I run it very early, about 1 seconds after device boot.

KonstantinKondrashov, 2025-03-10:
To exclude other issues that may cause this issue, you may try to read the register directly via `esp_rom_get_reset_reason(PRO_CPU_NUM)` to confirm that you are calling the API that is not ready yet.

I am wondering if your code works with this fix again?
```c
    s_reset_reason = ESP_RST_UNKNOWN;
// ...

esp_reset_reason_t esp_reset_reason(void)
{
    return s_reset_reason != ESP_RST_UNKNOWN? s_reset_reason  : esp_rom_get_reset_reason(PRO_CPU_NUM);
}
```

dzungpv, 2025-03-10:
> To exclude other issues that may cause this issue, you may try to read the register directly via `esp_rom_get_reset_reason(PRO_CPU_NUM)` to confirm that you are calling the API that is not ready yet.
> 
> I am wondering if your code works with this fix again?
> 
>     s_reset_reason = ESP_RST_UNKNOWN;
> // ...
> 
> esp_reset_reason_t esp_reset_reason(void)
> {
>     return s_reset_reason != ESP_RST_UNKNOWN? s_reset_reason  : esp_rom_get_reset_reason(PRO_CPU_NUM);
> }

Yes, It seem to work, but esp_reset_reason() also work on single core CPU like S2 C3 and C6, FYI

closed 2025-03-10 by dzungpv