espressif/esp-idf #15543: LEDC: Failed to configure timer: ESP_ERR_INVALID_STATE

ravencarcass, 2025-03-09:
Hello, 
I've a problem with the LEDC. I've copied the timer configuration from a simple example but gave me some troubles. While testing it gave it this error; 
```
E (635) ledc: ledc_timer_del(736): LEDC is not initialized
E (645) LEDC: Failed to configure timer: ESP_ERR_INVALID_STATE
```

The piece of code that results in a error 
```
    ledc_timer_config_t timer;
        timer.speed_mode         = LEDC_MODE;
        timer.timer_num          = LEDC_TIMER;
        timer.freq_hz            = LEDC_FREQUENCY;
        timer.duty_resolution    = LEDC_TIMER_8_BIT;
        timer.clk_cfg            = LEDC_AUTO_CLK;
    esp_err_t retT = ledc_timer_config(&timer);

    if (retT != ESP_OK) {
        ESP_LOGE("LEDC", "Failed to configure timer: %s", esp_err_to_name(retT));
        return;
    }
    
    ledc_channel_config_t channel;
        channel.gpio_num       = LEDC_OUTPUT_IO;
        channel.speed_mode     = LEDC_MODE;
        channel.channel        = LEDC_CHANNEL;
        channel.intr_type      = LEDC_INTR_DISABLE;
        channel.timer_sel      = LEDC_TIMER;
        channel.duty           = LEDC_DUTY;
        channel.hpoint         = 0;
        esp_err_t ret = ledc_channel_config(&channel);

    if (ret != ESP_OK) {
        ESP_LOGE("LEDC", "Failed to configure channel: %s", esp_err_to_name(ret));
        return;
    }
```

wanckl, 2025-03-10:
@ravencarcass  see the log, you calling `ledc_timer_config` -> `ledc_timer_del(736)`, menas you set the `timer_conf->deconfigure`    

review your code, `ledc_timer_config_t timer;` define it without clean all to zero can leader to the resoult.    

I guess at first you copy the code the following lines should decorated in a `{}`, 
```c
    ledc_timer_config_t timer = {
        timer.speed_mode         = LEDC_MODE,
        timer.timer_num          = LEDC_TIMER,
        timer.freq_hz            = LEDC_FREQUENCY,
        timer.duty_resolution    = LEDC_TIMER_8_BIT,
        timer.clk_cfg            = LEDC_AUTO_CLK,
    };
```

This way the compiler will help clean the unused members.

ginkgm, 2025-03-20:
Issue closed due to lack of response. Please feel free to reopen it when you still have problems!

closed 2025-03-20 by ginkgm