espressif/esp-idf #15544: Modify SPI flash driver to be compatible with stacks in PSRAM

jonsmirl, 2025-03-09:
### Is your feature request related to a problem?

This originally came up on esp-matter.  https://github.com/espressif/esp-matter/issues/1249

It would really be useful if the spi_flash driver had a menu config option telling it to switch onto it's own dedicated stack in internal RAM before doing anything.

Use case is a system running Lua. Lua uses a LOT of stack while running, too much to be in internal memory for larger Lua programs. Lua allocates all of its local variables on the stack. To get a lot of room it is easy to give Lua a PSRAM based stack, but then none of the flash calls work any more - like the file system. This is not so simple to fix in the application. Everything which touches flash has to be fixed not to use PSRAM stacks. All calls to NVM, file systems, OTA, etc...   Much easier if the spi_flash driver just switched onto it's own stack when called.

### Describe the solution you'd like.

Menuconfig option in the spi_flash driver telling it to use a private stack for internal operations.  Alternatively, check the stack pointer in the spi_flash component and if it is in PSRAM switch stacks automatically.

### Describe alternatives you've considered.

Fixing this problem at every call site in the user app. This is a lot of work and error prone.

BitsForPeople, 2025-03-09:
Sounds like you'd want to create a dedicated "flash-write" FreeRTOS task, with stack in interal RAM, to which you can send 'commands' from other tasks.

jonsmirl, 2025-03-09:
I know that technique, but now consider Lua.  I have to search all over the Lua engine, locate everywhere it can use flash (like all file system calls) and then make another task and figure out how to send all of the data over to it, get the answers and errors back and maintain task synchronization.  But I didn't write the Lua engine and I don't maintain it.

BitsForPeople, 2025-03-09:
I'd argue that it makes more sense to adapt the Lua engine to the ESP32 than to adapt the IDF to the Lua engine.

jonsmirl, 2025-03-09:
Why should I have to be on a specific stack type before calling the spi_flash module?  Lua is just one example of this, anything which needs large stacks would hit it. You can hit it from LVGL too if you enable file system support and use a PSRAM stack.

I am continuously running out of internal RAM and my own code only uses 8KB of it. I am always fighting with all of the various components to the THEM to stop using internal RAM. Right now I have a 80KB Lua stack in internal RAM that I can't move to PSRAM.

ginkgm, 2025-03-20:
Hi @jonsmirl ,

We have considered making such change. The change is likely to be another spi_flash task to delegate the operations that will affect the cache. But this won't happen very soon. Thanks for your suggestion.

Any other opinions are welcome below. We are willing to hear your suggestions!

jonsmirl, 2026-02-12:
## Our workaround: `--wrap` + `esp_execute_shared_stack_function()`

Since this hasn't been addressed yet in ESP-IDF, here's the workaround we built. It's been running in production on ESP32-S3 (with PSRAM) running Matter + Lua + LVGL. The approach is completely transparent to application code — no call-site changes needed.

### Why the ESP32-C61 made this critical

The ESP32-C61 is especially punishing without this fix. It has ~320KB internal SRAM, a single RISC-V core, and only 2 GDMA pairs. Running a Matter stack + WiFi + BLE + our application consumes nearly all internal RAM. We have 5+ FreeRTOS tasks (Lua scripting, attribute logging, WebRTC/TURN, BLE multicast, memory monitoring) that each need 4-8KB of stack. That's 30-40KB of internal RAM just for task stacks — memory we simply don't have. The C61 does have PSRAM, but moving any task stack to PSRAM immediately breaks all flash operations from that task (SPIFFS, NVS, OTA, partition reads). Without this workaround the C61 firmware wouldn't fit in memory.

### How it works

We use GCC's `--wrap` linker flag to intercept all 7 `esp_partition_*` functions. Each wrapper checks the current stack pointer — if it's in internal RAM, it passes through to the real function with zero overhead. If it's in PSRAM, it swaps to a shared 4KB internal RAM stack using ESP-IDF's own `esp_execute_shared_stack_function()`.

### CMakeLists.txt setup (only when PSRAM is enabled)

```cmake
# Flash stack swap: intercept esp_partition_* so PSRAM-stack tasks can do flash I/O
# by transparently switching to a shared internal RAM stack during the operation.
if(CONFIG_SPIRAM)
    target_link_libraries(${COMPONENT_LIB} INTERFACE
        "-Wl,--wrap=esp_partition_read"
        "-Wl,--wrap=esp_partition_read_raw"
        "-Wl,--wrap=esp_partition_write"
        "-Wl,--wrap=esp_partition_write_raw"
        "-Wl,--wrap=esp_partition_erase_range"
        "-Wl,--wrap=esp_partition_mmap"
        "-Wl,--wrap=esp_partition_munmap"
    )
endif()
```

### Implementation (flash_stack_swap.c)

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "esp_partition.h"
#include "esp_memory_utils.h"
#include "esp_attr.h"
#include "esp_expression_with_stack.h"

/* Shared internal RAM stack — 4KB is plenty for esp_partition -> esp_flash -> SPI HAL path */
#define FLASH_STACK_WORDS 1024
static DRAM_ATTR StackType_t s_flash_stack[FLASH_STACK_WORDS] __attribute__((aligned(16)));

/* Mutex serializes all PSRAM-task flash operations (protects shared args + stack) */
static SemaphoreHandle_t s_arg_mutex;
static DRAM_ATTR StaticSemaphore_t s_arg_mutex_buf;

/* Lock required by esp_execute_shared_stack_function() */
static SemaphoreHandle_t s_stack_lock;
static DRAM_ATTR StaticSemaphore_t s_stack_lock_buf;

/* Arguments passed via DRAM struct (PSRAM inaccessible during flash ops) */
static DRAM_ATTR struct {
    const esp_partition_t *partition;
    size_t offset;
    void *dst;
    const void *src;
    size_t size;
    esp_err_t ret;
    /* mmap fields */
    esp_partition_mmap_memory_t mmap_memory;
    const void **mmap_out_ptr;
    esp_partition_mmap_handle_t *mmap_out_handle;
    esp_partition_mmap_handle_t munmap_handle;
} s_args;

/* Declare real functions the linker resolves */
extern esp_err_t __real_esp_partition_read(const esp_partition_t *, size_t, void *, size_t);
[... 54 lines cut ...]

3. **Zero overhead for internal-RAM tasks** — The fast path is just reading SP and calling `esp_ptr_external_ram()`. Tasks with normal stacks pay essentially nothing.

4. **4KB shared stack** — The `esp_partition → esp_flash → SPI HAL` call path uses under 2KB of stack. 4KB provides comfortable margin.

5. **Single mutex** — Serializes all PSRAM-task flash operations. Fine for our use case since flash ops are already serialized by the SPI bus.

### What this enables

With this wrapper, we moved 5+ task stacks from internal RAM to PSRAM, recovering ~40KB of internal SRAM:
- Lua engine task (8KB → PSRAM) — calls SPIFFS for file loading
- Attribute logger task (4KB → PSRAM) — directly calls `esp_partition_write`/`esp_partition_erase_range`
- WebRTC/TURN task (8KB → PSRAM) — indirect NVS access
- Memory monitor task (4KB → PSRAM)
- BLE multicast task (4KB → PSRAM)

### Why this should be in ESP-IDF

This works, but it's fragile in two ways:

1. **Incomplete coverage** — We only wrap `esp_partition_*`. If some component calls `esp_flash_read()` directly (bypassing the partition layer), it would crash. The proper fix belongs inside `esp_flash` or `spi_flash` where ALL flash operations funnel through.

2. **Architecture-specific SP read** — Our `stack_in_psram()` uses inline asm. ESP-IDF already knows how to check this internally.

A Kconfig option like `CONFIG_SPI_FLASH_AUTO_STACK_SWAP` that does this inside the flash driver would fix it properly for everyone. The check could go in `esp_flash_default_host_driver_init()` or the `esp_flash_chip_driver_t` operation functions where all flash I/O converges.