espressif/esp-idf #15545: sdcard erase does not erase

dizcza, 2025-03-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.5-143-gdb1c67632a

### Espressif SoC revision.

ESP32 rev 3.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO T8 v1.7

### Power Supply used.

USB

### What is the expected behavior?

There are three issues:

1. `sdmmc_full_erase` happens immediately and thus I doubt it does the job.
2. After it "finishes", the SD card is no longer functional: `fopen` results in an error until I restart. Upon restart, the SD card mount fails and I need to format it prior mounting. Once it's formatted, the mounting is resumed and successful.
3. Although I explicitly call the `esp_vfs_fat_sdmmc_mount` function, the `is_mmc` is set to `0`. Shouldn't it be `1`?

```c

esp_err_t sdcard_erase() {
	if (m_card == NULL) {
		return ESP_ERR_INVALID_STATE;
	}
	ESP_LOGI(TAG, "Erasing SD card...");
	esp_err_t err = sdmmc_full_erase(m_card);
	if (err == ESP_OK) {
		ESP_LOGI(TAG, "SD card erased");
	} else {
		ESP_LOGW(TAG, "SD card erase failed %s", esp_err_to_name(err));
	}
	return err;
}

static void sdcard_mount_finish(esp_err_t err) {
    if (err == ESP_OK) {
        ESP_LOGI(TAG, "SD card mounted at %s", sdcard_mount_point);
        sdmmc_card_print_info(stdout, m_card);
        printf("MMC: %d\n", m_card->is_mmc);

        /* Print free memory size info */
        printf("Free: %lluMB\n", sdcard_get_free_bytes() / (1024 * 1024));
        if (m_card->csd.sector_size != SDCARD_SECTOR_SIZE_DEFAULT) {
            ESP_LOGW(TAG, "SD sector size %d != %d", m_card->csd.sector_size, SDCARD_SECTOR_SIZE_DEFAULT);
        }
    } else {
        ESP_LOGW(TAG, "sdcard_mount failed (%s)",
                esp_err_to_name(err));
    }
}

/**
 * Initialize and mount the SD card with MMC interface
 *
 * @return err code
 */
esp_err_t sdcard_mount_mmc() {
    if (m_card != NULL) {
        ESP_LOGI(TAG, "SD card already mounted");
        return ESP_OK;
    }

    ESP_LOGI(TAG, "Mounting SD MMC...");

	esp_vfs_fat_sdmmc_mount_config_t mount_config = SDCARD_MOUNT_CONFIG_DEFAULT();

    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

    // Use 1-line SD mode
    slot_config.width = 1;

    // Enable internal pullups on enabled pins. The internal pullups
    // are insufficient however, please make sure 10k external pullups are
    // connected on the bus. This is for debug / example purpose only.
    // When an SD card is operating in SPI mode or 1-bit SD mode,
    // the CMD and DATA (DAT0 - DAT3) lines of the SD bus must be
    // pulled up by 10 kOhm resistors.
    slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP;

    esp_err_t err = esp_vfs_fat_sdmmc_mount(sdcard_mount_point, &host,
            &slot_config, &mount_config, &m_card);

    sdcard_mount_finish(err);

	return err;
}
```

### What is the actual behavior?

1. The SD card is fully erased (all sectors are filled with zero).
2. The SD card is functional upon the sdmmc_full_erase finish.
3. The `is_mmc` is set to `1`.

### Steps to reproduce.

```c
sdcard_mount_mmc();
sdcard_erase();
```

### Debug Logs.

```plain
I (23:10:36.292) sdcard: Mounting SD MMC...
I (23:10:36.344) sdcard: SD card mounted at /sd
Name: SD32G
Type: SDHC/SDXC
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 29580MB
CSD: ver=2, sector_size=512, capacity=60579840 read_bl_len=9
SSR: bus_width=1
MMC: 0
Free: 29572MB
I (23:10:51.538) sdcard: Erasing SD card...
I (23:10:51.539) sdmmc_cmd: sdmmc_erase_sectors arg 0
I (23:10:51.596) sdcard: SD card erased
```

### More Information.

Should I use `sdmmc_write_sectors` to erase the SD card instead of `sdmmc_full_erase`? Doesn't make sense...

dizcza, 2025-03-09:
There are really no examples and no description notes on how to do a complete erase of an SD card. The API does not work as intended.

igrr, 2025-03-10:
Hi @dizcza, do you intend to format the card or to erase the filesystem?

I am guessing it's the former. We have added the function [esp_vfs_fat_sdcard_format](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/storage/fatfs.html#_CPPv425esp_vfs_fat_sdcard_formatPKcP12sdmmc_card_t) for this purpose, please give it a try.

sdmmc_full_erase is a low-level function of the SDMMC protocol layer. If you call it after initializing FATFS, the filesystem will get corrupted because FATFS library won't "know" that the low-level erase operation was performed. This function can be used if the application directly reads and writes SD card sector without using a filesystem.

dizcza, 2025-03-10:
@igrr no, I don't want to format the filesystem. I want to erase it so there is no chance of restoring old files after a (quick) format.

Formatting works fine but that's not what I want.

So what does sdmmc_full_erase do then?

dizcza, 2025-03-10:
I mean, I can read from sources that sdmmc_full_erase is just erasing all sectors of the SD card. But I don't understand what "erasing" of a sector means. I thought it would fill it with zero. Or at least make the sector "unrecoverable" - consider an attacker has physical access to it and wants to restore some files.

BitsForPeople, 2025-03-10:
SD cards do the erasing internally. The host does not need to write any data to erase a sector.
Erasing the whole card includes deleting any and all file systems on the card, so the card needs to be formatted again before subsequent use.
In fact, seeing that the file system is gone after the erase command indicates that the erase actually happened.
An SD card is not an MMC card, so MMC=0 is what you should get.

dizcza, 2025-03-10:
I'm using the SD card by itself with no adapter as shown here https://github.com/LilyGO/TTGO-T8-ESP32/blob/master/image/T81.7.jpg

> I think sdmmc_full_erase should do the job, we do have a test case for it which seems to be passing. The problem might indeed be that the filesystem layer doesn't expect the data to be erased by the low level function. We can prepare an example to show how to use this function correctly.

I cannot believe it filled 60 million sectors to either zero or 0xFF in a subsecond, while a (quick) format takes about 5 to 10 seconds to perform. The "Discard" and "Erase" options are also confusing, thanks for the clarification, I'd like to see it in the function description. However, I'm still not sure if "Discrad" will actually (instruct to) fill the sector with whatever constant value, so I'm not sure if I can use this function to make sure that the attacker won't be able to restore some data.

Either way, the is_mmc is not set to 1 and I don't know if that causes the sdmmc_full_erase to misbehave since I'm mounting the card in the MMC mode.

dizcza, 2025-03-10:
I just mounted the same SD card with SPI:
```c
    esp_vfs_fat_sdmmc_mount_config_t mount_config = SDCARD_MOUNT_CONFIG_DEFAULT();

    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    host.slot = spi_host;
    host.max_freq_khz = freq_speed;

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
    slot_config.gpio_cs = (gpio_num_t) cs_gpio;
    slot_config.host_id = host.slot;

    esp_err_t err = esp_vfs_fat_sdspi_mount(sdcard_mount_point, &host,
            &slot_config, &mount_config, &m_card);
```
and `sdmmc_full_erase` is doing some job for 1 minute, it's just stuck... maybe it does fill the slots with zeros in this mode. I guess I need to wait 10+ hours to erase a 30G SD card using SPI and one data pin, so I'm stopping it now.

I do not use SPI mount API because it's slow compared to native MMC. And it seems to be an issue with the `sdmmc_full_erase` using MMC. Please confirm.

BitsForPeople, 2025-03-10:
> I cannot believe it filled 60 million sectors to either zero or 0xFF in a subsecond, while a (quick) format takes about 5 to 10 seconds to perform.

Flash memory and its controllers work differently than you think. No use in comparing a "chip erase" hardware operation to writing the data structures of a file system.

> However, I'm still not sure if "Discrad" will actually (instruct to) fill the sector with whatever constant value, so I'm not sure if I can use this function to make sure that the attacker won't be able to restore some data.

Why? @igrr already wrote that "discard" may or may not do the same as erase; how (and if!) it's implemented is at the sole discretion of the memory controller inside each SD card and can vary from one card to the next.

> Either way, the is_mmc is not set to 1 and I don't know if that causes the sdmmc_full_erase to misbehave since I'm mounting the card in the MMC mode.

How does the card "misbehave"?

dizcza, 2025-03-10:
> Erasing the whole card implicitly wipes out any and all file systems on the card, so the card needs to be formatted again before subsequent use.

Ok, do I need to unmount the SD card before the erase operation? It looks like I don't need to. And I just need to format it after the erase is done as you suggest.

> After the erase, you should find 0xff or 0x00 in every byte of every sector you read from the card.

Hmm, that's easy to check... I cannot believe the "erase" command can do such a thing in 100 ms... But all right.

> An SD card is not an MMC card, so MMC=0 is what you should get.

So I misunderstood the meaning of the is_mmc field.

All questions answered and there seems to be no issues in sdmmc_full_erase and it's fast response is expected. I still need to check of all bytes are 0xFF or 0x00, till then I'm closing the issue.

closed 2025-03-10 by dizcza

dizcza, 2025-03-10:
Sorry, but mounting followed by formatting the SD card after the erase command is completed does not work: the SD card fails to open files.

I call `esp_vfs_fat_sdcard_unmount` and then `esp_vfs_fat_sdmmc_mount` which internally formats the SD card before mounting as it's seen in the logs. The SD card prints the info but then is unable to operate.

Restart helps. Any remedies save for restart?

```
I (21:21:34.999) sdcard: Erasing SD card...
I (21:21:35.046) sdcard: SD card erased
I (21:21:35.046) sdcard: sdcard_unmount OK
I (21:21:35.047) sdcard: Mounting SD MMC...
W (21:21:35.095) vfs_fat_sdmmc: failed to mount card (13)
W (21:21:35.096) vfs_fat_sdmmc: partitioning card
W (21:21:35.098) vfs_fat_sdmmc: formatting card, allocation unit size=16384
W (21:21:45.575) vfs_fat_sdmmc: mounting again
I (21:21:45.577) sdcard: SD card mounted at /sd
Name: SD32G
Type: SDHC/SDXC
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 29580MB
CSD: ver=2, sector_size=512, capacity=60579840 read_bl_len=9
SSR: bus_width=1
MMC: 0
Free: 29572MB
```