espressif/esp-idf #15546: Download behaves differently between clone and ZIP

carstenpoulsen, 2025-03-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

current master

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The expected behaviour is to get the same result whether I clone or download the ZIP.

### What is the actual behavior?

I download by GIT Clone and ZIP.
I am looking for esp-idf\examples\peripherals\touch_sensor
When I clone I find
- touch_element              
- touch_sens_basic           
- touch_sens_examples_common 
- README.md           When I download ZIP I find in the same path
- touch_element           
- touch_sensor_v1         
- touch_sensor_v2         
- touch_periph_version.txt
The ZIP contains esp-idf-5,.2.5

A brief examination reveals that it is the same problem with all other paths in the examples path.

### Steps to reproduce.

Download by GIT Clone and ZIP and compare the results for equality

### Build or installation Logs.

```plain
N/A
```

### More Information.

No

wanckl, 2025-03-10:
@carstenpoulsen 
Just checked the ZIP and clone of master is same.    
And find `touch_sensor_v1/v2`  in release version.

Please check your ZIP version

BitsForPeople, 2025-03-10:
The master branch normally includes changes the latest _release_ does not.

ginkgm, 2025-03-20:
@BitsForPeople  

so could you give the specific version (5.2.5?) that has the difference? or a link

and the specific folder that has the difference.

carstenpoulsen, 2025-03-20:
In the cloned repos I expect to find
esp-idf\examples\peripherals\touch_sensor
instead I find touch_sensor_v1

ginkgm, 2025-03-21:
@carstenpoulsen I still can't understand your question.. 

I download the zip from [esp-idf-v5.2.5.zip](https://github.com/espressif/esp-idf/releases/download/v5.2.5/esp-idf-v5.2.5.zip) 

and it looks the same as https://github.com/espressif/esp-idf/tree/v5.2.5/examples/peripherals/touch_sensor . Same for the version file. So I think it's exactly the archive of v5.2.5

what do you mean by `latest release/last stable`? You can specify which version to download.. Or do you mean the component downloaded by the example?

carstenpoulsen, 2025-03-21:
Thanks for your perseverance.

It is appreciated.

I think that the outcome of this is that I must be observant when I encounter this again …. If I do.

You and others have pointed out what I shall look for, so let’s leave it there.

 

Have a nice day

Carsten

closed 2025-04-17 by ginkgm