espressif/esp-idf #15547: Watchdog timeout in oc-cli example when configured with CONFIG_OPENTHREAD_CONSOLE_TYPE_USB_SERIAL_JTAG=y

nicolas-christe, 2025-03-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-690-gd4aa25a38e

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

I expect the example code to run without error

### What is the actual behavior?

The sample code fail with a watchdog timeout error

### Steps to reproduce.

1. go to esp-idf/examples/openthread/ot_cli
2. change the config have the openthread console on JTAG-USB (CONFIG_OPENTHREAD_CONSOLE_TYPE_USB_SERIAL_JTAG=y)
3. Run the sample
4. check the monitor log

### Debug Logs.

```plain
I (23) boot: ESP-IDF v5.4-690-gd4aa25a38e 2nd stage bootloader
I (23) boot: compile time Mar 10 2025 09:18:39
I (24) boot: chip revision: v0.0
I (24) boot: efuse block revision: v0.2
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (35) boot.esp32c6: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (59) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (65) boot:  2 factory          factory app      00 00 00010000 00140000
I (72) boot: End of partition table
I (75) esp_image: segment 0: paddr=00010020 vaddr=420d0020 size=36824h (223268) map
I (124) esp_image: segment 1: paddr=0004684c vaddr=40800000 size=017cch (  6092) load
I (126) esp_image: segment 2: paddr=00048020 vaddr=42000020 size=cc9e0h (838112) map
I (282) esp_image: segment 3: paddr=00114a08 vaddr=408017cc size=0f498h ( 62616) load
I (296) esp_image: segment 4: paddr=00123ea8 vaddr=40810c70 size=027d4h ( 10196) load
I (303) boot: Loaded app from partition at offset 0x10000
I (304) boot: Disabling RNG early entropy source...
I (314) cpu_start: Unicore app
I (323) cpu_start: Pro cpu start user code
I (323) cpu_start: cpu freq: 160000000 Hz
I (324) app_init: Application information:
I (324) app_init: Project name:     esp_ot_cli
I (327) app_init: App version:      v5.4-690-gd4aa25a38e
I (332) app_init: Compile time:     Mar 10 2025 09:22:04
I (337) app_init: ELF file SHA256:  6bad13968...
I (342) app_init: ESP-IDF:          v5.4-690-gd4aa25a38e
I (347) efuse_init: Min chip rev:     v0.0
I (351) efuse_init: Max chip rev:     v0.99
I (355) efuse_init: Chip rev:         v0.0
I (359) heap_init: Initializing. RAM available for dynamic allocation:
I (365) heap_init: At 4081BE70 len 000607A0 (385 KiB): RAM
I (370) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (375) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (381) spi_flash: detected chip: generic
I (384) spi_flash: flash io: dio
W (387) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (400) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (406) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (412) coexist: coex firmware version: e727207
I (417) coexist: coexist rom version 5b8dcfa
I (421) main_task: Started on CPU0
I (421) main_task: Calling app_main()
I (431) phy_init: phy_version 330,4af1121,Feb  6 2025,14:39:13
I (471) phy: libbtbb version: 99f6adc, Feb  6 2025, 14:39:27
I(481) OPENTHREAD:[I] CslTxScheduler: Bus TX Time: 0 usec, Latency: 0 usec. Calculated CSL Frame Request Ahead: 20000 usec
I(481) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
> I (481) OPENTHREAD: OpenThread attached to netif
I (501) main_task: Returned from app_main()
> E (5981) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5981) task_wdt:  - IDLE (CPU 0)
E (5981) task_wdt: Tasks currently running:
E (5981) task_wdt: CPU 0: ot_cli
E (5981) task_wdt: Print CPU 0 (current core) backtrace
esp_backtrace_print: Print CPU 0 (current core) registers
--- Stack dump detected
Core  0 register dump:
MEPC    : 0x40801e80  RA      : 0x40801e80  SP      : 0x40826cb0  GP      : 0x40811474
--- 0x40801e80: _lock_acquire_recursive at /Users/nicolas/tools/esp/v5.4/esp-idf/components/newlib/locks.c:163
0x40801e80: _lock_acquire_recursive at /Users/nicolas/tools/esp/v5.4/esp-idf/components/newlib/locks.c:163

TP      : 0x40826e00  T0      : 0x40028140  T1      : 0x40000b8c  T2      : 0x00000000
--- 0x40028140: __retarget_lock_acquire_recursive in ROM
0x40000b8c: __call_ieee80211_is_tx_allowed in ROM

S0/FP   : 0x4081c9bc  S1      : 0x40826e68  A0      : 0x00000000  A1      : 0xffffffff
A2      : 0x00000004  A3      : 0x00000004  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x42009940  A7      : 0x9c3c0000  S2      : 0x00000000  S3      : 0x408119bc
--- 0x42009940: uart_read at /Users/nicolas/tools/esp/v5.4/esp-idf/components/esp_driver_uart/src/uart_vfs.c:272

S4      : 0x420c1b6c  S5      : 0x00000002  S6      : 0x00000001  S7      : 0xffffffff
--- 0x420c1b6c: lflush at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/refill.c:28

S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000001  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001889  MTVEC   : 0x40800001  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de
--- 0x40800001: _vector_table at /Users/nicolas/tools/esp/v5.4/esp-idf/components/riscv/vectors_intc.S:54
```

### More Information.

Example work fine when configured to use the UART for the openthread console (CONFIG_OPENTHREAD_CONSOLE_TYPE_UART=y)

zwx1995esp, 2025-03-10:
Hi, @nicolas-christe  I don't think this is a bug. On the ESP32-C6 devkit boards, there are two ports on the devkit boards.

[image]

If you connect the `UART` port, `blue block` shown in the picture, you should enable `CONFIG_OPENTHREAD_CONSOLE_TYPE_UART`, and if you changed to use `USB` port, `green block` shown in the picture, the `CONFIG_OPENTHREAD_CONSOLE_TYPE_USB_SERIAL_JTAG` should be enabled.

nicolas-christe, 2025-03-10:
Hi @zwx1995esp,

This is exactly what I'm doing:

- if I compile `oc_cli` example with `CONFIG_OPENTHREAD_CONSOLE_TYPE_UART=y` and connect to the blue port everything works fine, I have the log and can interact with openthread command line.
- if I compile `oc_cli` example with `CONFIG_OPENTHREAD_CONSOLE_TYPE_USB_SERIAL_JTAG=y` and connect to the green port,  I can run the app, on the console I have the log and few seconds after starting the watchdog time out error.

It's not a big issue for me because I can work with the UART port, but I was expecting the sample to also work with the USB port.

zwx1995esp, 2025-03-11:
Ok, thanks for your clarification, we will reproduce the issue you encountered when using USB port.

tanyanquan, 2025-03-12:
Hi @nicolas-christe , we've tried to reproduce the issue you've encountered but it seems to be working on our end. Could you help us try these few steps to see if it fixes the issue:

1. Delete the `build` folder in `esp-idf/examples/openthread/ot_cli`
2. Run the following commands:
    - `idf.py set-target esp32c6`
    - `idf.py menuconfig`
3. In the configuration menu, go to: Component config → ESP System Settings → Channel for console output
4. Select USB Serial/JTAG Controller
5. Press "Q" then "Y" to confirm changes
6. Run the following commands:
    - `idf.py build`
    - `idf.py erase-flash flash monitor`

nicolas-christe, 2025-03-14:
I've restarted from a clean checkout of the example and I not producing the problem anymore. Sorry for the inconvenience.

closed 2025-03-14 by nicolas-christe