espressif/esp-idf #15548: Problems with the i80 controller example  ( st7789vi)

ManelCS, 2025-03-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the i80_controller example with the esp32s3, the NHD-2.4-240320AF-CSXP display and the ST7789VI driver and on the screen I get what is seen in the photo below.

I am using the i80 8 lines configuration.

```
#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     CONFIG_EXAMPLE_LCD_PIXEL_CLOCK_HZ   // (12MHz)

#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL  1
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL
#define EXAMPLE_PIN_NUM_DATA0          13
#define EXAMPLE_PIN_NUM_DATA1          14
#define EXAMPLE_PIN_NUM_DATA2          11
#define EXAMPLE_PIN_NUM_DATA3          12
#define EXAMPLE_PIN_NUM_DATA4          9
#define EXAMPLE_PIN_NUM_DATA5          10
#define EXAMPLE_PIN_NUM_DATA6          3
#define EXAMPLE_PIN_NUM_DATA7          46
#if CONFIG_EXAMPLE_LCD_I80_BUS_WIDTH > 8
#define EXAMPLE_PIN_NUM_DATA8          41
#define EXAMPLE_PIN_NUM_DATA9          40
#define EXAMPLE_PIN_NUM_DATA10         39
#define EXAMPLE_PIN_NUM_DATA11         38
#define EXAMPLE_PIN_NUM_DATA12         48
#define EXAMPLE_PIN_NUM_DATA13         45
#define EXAMPLE_PIN_NUM_DATA14         21
#define EXAMPLE_PIN_NUM_DATA15         47
#endif
#define EXAMPLE_PIN_NUM_PCLK           42         // WRX
#define EXAMPLE_PIN_NUM_CS             1
#define EXAMPLE_PIN_NUM_DC             2
#define EXAMPLE_PIN_NUM_RST            17  
#define EXAMPLE_PIN_NUM_BK_LIGHT       4
```

[image]

wanckl, 2025-03-10:
@Lzw655 👀 do you have any idea?

wanckl, 2025-03-10:
@ManelCS Have you tried decrase clock frequency?

ManelCS, 2025-03-10:
> [@ManelCS](https://github.com/ManelCS) Have you tried decrase clock frequency?
Yes I have tried up to 5 Mhz and it is still the same.

suda-morris, 2025-03-11:
@ManelCS How do you connect the LCD with ESP32S3? Have you connected the LCD's "RD" line to a high level rather than floating?

wanckl, 2025-03-11:
@ManelCS one more attention, on S3 using multi data line, you may need enhance GND connection, the more, the shorter, the wider, the better

ManelCS, 2025-03-11:
> [@ManelCS](https://github.com/ManelCS) How do you connect the LCD with ESP32S3? Have you connected the LCD's "RD" line to a high level rather than floating?

You're right, the RD line was disconnected.
> [@ManelCS](https://github.com/ManelCS) one more attention, on S3 using multi data line, you may need enhance GND connection, the more, the shorter, the wider, the better

I have improved the connections to GND.

It works correctly for me now. Thank you very much.

closed 2025-03-11 by ManelCS