espressif/esp-idf #15549: wifi_provisioning/manager.h failed to compile after menuconfig

eiffel31, 2025-03-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I modified many settings using idf.py menuconfig
I try to remove as many wifi options as possible to speed up wifi connection on local only wifi network at startup.

### What is the actual behavior?

idf.py build failed with the following error message:
In file included from /users/[...]/esp-idf_v5.4/components/wifi_provisioning/src/wifi_provisioning_priv.h:12,
                 from /users/[...]/esp-idf_v5.4/components/wifi_provisioning/src/manager.c:26:
/users/[...]/esp-idf_v5.4/components/wifi_provisioning/include/wifi_provisioning/manager.h:221:1: error: empty enum is invalid
  221 | } wifi_prov_security_t;
      | ^

### Steps to reproduce.

Create a new empty project and use provided sdkconfig and source file.

[bug_report.zip](https://github.com/user-attachments/files/19163566/bug_report.zip)

### Build or installation Logs.

```plain

```

hrushikesh430, 2025-03-17:
Hi @eiffel31, actually while building your application, it is pulling and trying to build all the components (even if they are not required). So in the `protocomm` component the at least one security level (`ESP_PROTOCOMM_SUPPORT_SECURITY_VERSION_0/1/2`) should be enabled but in your case no security level is enabled. To resolve this issue you can follow up one of the three methods - 
1. Remove the protocomm config from your sdkconfig (even if they are not set), and for those components which are not required for your application. So that application will pull the default values of the `CONFIGS`.
2. Enable at least one of the three security level. And then try to build your application.
3. Enable the minimal build application in the `CMakelists.txt` for pulling the components which are required for the application. There are two CMakelists.txt on the top project and inside the main component.

Top level CMakelists.txt
```
# The following five lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
idf_build_set_property(MINIMAL_BUILD ON)
project(my_new_project)
```

Main component CMakelists.txt (In this CMakelists.txt add the components which are required for the application)
```
set(COMPONENT_REQUIRES lwip esp_wifi nvs_flash)

idf_component_register(SRCS "my_new_project.c"
                    INCLUDE_DIRS "."
                    REQUIRES ${COMPONENT_REQUIRES})
```

eiffel31, 2025-03-17:
Hi @hrushikesh430 ,
thank you very much for your response. So there are actions I can do on my side to fix this problem.

The underlying question: is it the expected behavior for from menuconfig/espressif's code? In this case, changes using menuconfig results in failure while compiling the code in the library.

In my opinion, either the code should compile for all options that menuconfig can generate, or menuconfig should not generate an option set that is not supported by the library code.

hrushikesh430, 2025-03-19:
Hi @eiffel31, actually this case, the issue occurring due to you skdconfig file, which include the configurations of (wifi_provisoning as well as protocomm). Due to which it is pulling both the components and none of the security level is enabled. By default if you try to create the project using the command `idf.py create-project <project-name>`,  the default values of config will be pulled by the `idf.py set-target esp32` and you will have no build crash.

hrushikesh430, 2025-04-01:
Hi @eiffel31, Closing this issue due to no response, please feel free to reopen this issue for an y further queries.

closed 2025-04-01 by hrushikesh430