espressif/esp-idf #15552: Boot in wrong Ota Partiton after a Crash Software reset due to exception/panic (ESP_RST_PANIC)

filzek, 2025-03-10:
We are using the Release v5.4 branch with the commit
commit d4aa25a38e9c8659a5e5f4704d31ff532d807de0 (HEAD -> release/v5.4, origin/release/v5.4)

we are not sure when the problem start to happen but the result is that if a crash happen the system will reboot and load the firmware from the another OTA partition. 

But if so, we just tell the device to reboot with esp_restart it reboot and load the correct firmware partition set in the otadata.

Not sure why this behavior start.

filzek, 2025-03-10:
after this problem happen the board itself will always boot the other partition after an esp_restart

filzek, 2025-03-12:
maybe something related to old bootloaders in conflict with the new sdk when panic after ota happen? even without rollback should it set the partition to on using esp_ota_mark_app_valid_cancel_rollback();  ?

KonstantinKondrashov, 2025-03-12:
Hi @filzek!
Could you add more details about this case? sdkconfig, Do you use the rollback or anti-rollback features?
We did not observe such a behaviour.

For example, if you use the rollback feature and do not confirm that this app works (by calling esp_ota_mark_app_valid_cancel_rollback()) after the first new app boot, then after a crash you will see that the previous app is loaded. This is correct behavior.

During the panic and crash, the ota_data partition does not change, so it should not affect the selection of the running app.

> maybe something related to old bootloaders in conflict with the new sdk when panic after ota happen? even without rollback should it set the partition to on using esp_ota_mark_app_valid_cancel_rollback(); ?

I do not think so, the old bootloader has to support the rollback feature if it is this use case. The old bootloader has to work correctly anyway because it selects the app based on the ota_data partition. There is a tool that you can use to read the content out and investigate the issue, when and how it happened - https://github.com/espressif/esp-idf/blob/master/docs/en/api-reference/system/ota.rst

filzek, 2025-03-18:
even without the resource enable, the system is rebooting in another partition if its crashes, so, using the set to a valid solve the problem.

filzek, 2025-04-04:
> Hi @filzek!
> Could you add more details about this case? sdkconfig, Do you use the rollback or anti-rollback features?
> We did not observe such a behaviour.
>
> For example, if you use the rollback [...]

Still with problem, the current sdk 5.4.1 firmware once the firmware is loaded we set it as confirmed working.

Some field boards when have a software restart boots in the wrong partition.

So, this is a very erratic problem, can @KonstantinKondrashov you suggest anything else or want us to add any code?

filzek, 2025-04-04:
We will try to collect the bootloader version, is there any way to extract all other information needed?

KonstantinKondrashov, 2025-04-04:
Hi @filzek!
Could you provide the sdkconfig file? Are the bootloader and app built with the same IDF version?
Can you show the part of the code how you do OTA updates, is it a standard one?

> We will try to collect the bootloader version, is there any way to extract all other information needed?

At the start, the bootloader prints the idf version - ESP_EARLY_LOGI(TAG, "ESP-IDF %s 2nd stage bootloader", esp_bootloader_get_description()->idf_ver). You can try to read it out by app.

filzek, 2025-04-04:
@KonstantinKondrashov 

All the boot loaders in filed is prior to version 5, so will do not have metadata

filzek, 2025-04-04:
@KonstantinKondrashov  
major boards are with:
I (31) boot: ESP-IDF v4.2-dev-792-g6330b3345-dirty 2nd stage bootloader

filzek, 2025-04-11:
Bug still in place,

KonstantinKondrashov, 2025-04-11:
I still do not understand your case. May you share the sdkconfig files? Is the rollback feature used or not?

I guess, your case is:
1. ota_0 is running, OTA update of a new app, which goes to ota_1, ota_data is updated to point to the boot on ota_1, reboot.
2. ota_1 is running, a crash happens,
3. ota_0 is running gain.

When the crash happens there is no operation with the ota_data partition responsible for selecting the app.

Please share sdkconfig and partition table (csv) file.

filzek, 2025-04-11:
boot loader is old so we are not sure if it was enable or now. 
either way just in case after a update over OTA we do a esp_ota_mark_app_valid_cancel_rollback(); after the firmware is confirmed working.

[sdkconfig.zip](https://github.com/user-attachments/files/19707581/sdkconfig.zip)

filzek, 2025-04-11:
Let's explain everything again in a simple and clear way:

We have two OTA partitions, each with a size of 0x340000:
**ota_0
ota_1**

The OTA update process is performed normally and completes successfully. Devices reboot and run the new firmware without issues.

Once the new firmware boots, it logs its version to the cloud. This confirms that the device has fully completed the boot process and successfully performed the handshake—validating that the new firmware was properly uploaded and booted from the OTA partition.

After that, we call:

`esp_ota_mark_app_valid_cancel_rollback();  
`

This is meant to confirm the new firmware as valid and prevent the system from rolling back to the previous version.

However, all devices are configured to reboot at least once between 2 AM and 5 AM.

**The issue:** During this scheduled reboot window, some devices unexpectedly boot from the previous OTA partition, effectively reverting to the old firmware, even though **esp_ota_mark_app_valid_cancel_rollback()** was already called.

This behavior is unexpected and indicates that something may be going wrong with either:

- The OTA validation process,
- The rollback mechanism, or
- The partition boot logic.

We need to investigate what is causing this regression to the previous firmware version after a successful update and validation.

filzek, 2025-04-11:
so we check by:
void check_ota_status(void)
{
    const esp_partition_t *running = esp_ota_get_running_partition();
    esp_ota_img_states_t state;
    esp_err_t err = esp_ota_get_state_partition(running, &state);

    if (err == ESP_OK)
    {
        switch (state)
        {
        case ESP_OTA_IMG_NEW:
            debug_add("OTA CHECK: NEW Image not validated yet!");
            break;
        case ESP_OTA_IMG_PENDING_VERIFY:
            debug_add("OTA CHECK: Pending Verify Rollback possible!");
            {
                esp_err_t mark_err = esp_ota_mark_app_valid_cancel_rollback();
                if (mark_err != ESP_OK)
                {
                    debug_add("OTA CHECK: Failed to mark OTA as valid in PENDING_VERIFY state [%d] [%s]!",mark_err, esp_err_to_name(mark_err));
                }
            }
            break;
        case ESP_OTA_IMG_VALID:
            debug_add("OTA CHECK: VALID Rollback disabled!");
            break;
        case ESP_OTA_IMG_INVALID:
            debug_add("OTA CHECK: INVALID Image will not boot again!");
            break;
        case ESP_OTA_IMG_ABORTED:
            debug_add("OTA CHECK: ABORTED Update process aborted!");
            break;
        default:
            debug_add("OTA CHECK: UNKNOWN OTA state! - Will set it as VALID for next BOOT!");
            {
                esp_err_t mark_err = esp_ota_mark_app_valid_cancel_rollback();
                if (mark_err != ESP_OK)
                {
                    debug_add("OTA CHECK: Failed to mark OTA as valid in default case!  [%d] [%s]!",mark_err, esp_err_to_name(mark_err));
                }
            }
            break;
        }
    }
    else
    {
        debug_add("OTA CHECK: Failed to get OTA state!");
    }
}

in the first boot after the update the ota is in:
DEBUG READ
INITIAL DEBUG OK - MALLOC SUCESS! [1744383743] - FW:30000069
[1730233448] - NUMBER OF INCOMPLETE BOOTS:1
[1730233453] - WiFi STA Had Been Started
[1730233453] - WiFi STA Had Been Started
[1730233461] - SENTINELA - ******************** BOOT TIME **************************
[1730233461] - SENTINELA - BOOT: [382] FINISHED: [14106] SPENT: [13724]ms
[1730233461] - SENTINELA - ******************** BOOT TIME **************************
**[1730233461] - OTA CHECK: UNKNOWN OTA state! - Will set it as VALID for next BOOT!**
[1730233468] - IP has been taken
[1730233468] - STA - Global event_handler got ip:192.168.100.6
[1730233468] - STA - Global event_handler got Gateway ip:192.168.100.254

So, if reboot, it will show:
DEBUG READ
INITIAL DEBUG OK - MALLOC SUCESS! [1744383836] - FW:30000069
[1730233448] - NUMBER OF INCOMPLETE BOOTS:1
[1730233453] - WiFi STA Had Been Started
[1730233453] - WiFi STA Had Been Started
[1730233461] - SENTINELA - ******************** BOOT TIME **************************
[1730233461] - SENTINELA - BOOT: [382] FINISHED: [14121] SPENT: [13739]ms
[1730233461] - SENTINELA - ******************** BOOT TIME **************************
**[1730233461] - OTA CHECK: VALID Rollback disabled!**
[1730233467] - IP has been taken
[1730233467] - STA - Global event_handler got ip:192.168.100.6
[1730233467] - STA - Global event_handler got Gateway ip:192.168.100.254

So, in this case the boot was set correct.

The question is why even with this set correctly, the ota sometimes after a later restart boot to the wrong partition, so, we need to solve it.

filzek, 2025-04-23:
Hi @KonstantinKondrashov 

We were able to capture one log from when the problem happen.

entry 0x40080658
I (31) boot: ESP-IDF v4.2-dev-792-g6330b3345-dirty 2nd stage bootloader
I (31) boot: compile time 01:02:27
I (31) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 8MB
I (50) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) ЬѬլϬѬЬԬЬάӬάҬѬѬ֬ЬЬϬЬЬЬ֬ӬԬѬԬ֬ЬҬϬӬѬϬϬϬѬϬЬլЬҬ׬ЬЬϬϬѬӬϬ֬ѬҬԬЬŬϬҬЬЬ֬ϬѬϬЬլѬЬЬЬ׬ЬЬϬլѬЬլЬЬϬϬϬˬӬάլЬӬҬϬѬϬѬӬάҬϬЬЬѬլӬЬϬЬͬϬЬЬӬ׬ϬӬϬЬЬլЬӬЬլЬЬӬЬ֬ӬЬЬЬάЬ:  0 nvs              WiFi data        01 02 0000a000 00006000
I (92) ЬѬլϬѬЬԬЬάӬάҬѬѬ֬ЬЬϬЬЬЬ֬ӬԬѬԬ֬ЬҬϬӬѬϬϬϬѬϬЬլЬҬ׬ЬЬϬϬѬӬϬ֬ѬҬԬЬŬϬҬЬЬ֬ϬѬϬЬլѬЬЬЬ׬ЬЬϬլѬЬլЬЬϬϬϬˬӬάլЬӬҬϬѬϬѬӬάҬϬЬЬѬլӬЬϬЬͬϬЬЬӬ׬ϬӬϬЬЬլЬӬЬլЬЬӬЬ֬ӬЬЬЬάЬЬլϬЬЬӬ۬ӬӬͬЬԬѬӬѬҬլӬԬϬЬԬլլӬЬϬЬӬѬϬѬЬӬ׬ϬѬӬϬҬЬЬЬӬԬЬѬЬЬЬլӬϬЬҬ֬լӬάЬլҬլЬЬЬӬЬϬϬЬӬӬϬάЬЬЬά��Ь��ѬѬϬϬϬѬӬѬЬϬЬЬѬЬάլЬЬҬЬҬӬЬϬϬЬӬӬҬϬϬϬѬЬЬϬЬЬЬάάЬϬլӬάϬЬѬҬҬϬЬڬӬҬЬЬӬԬԬЬЬЬլլѬϬϬЬѬӬѬЬЬѬѬάЬЬϬ֬ѬЬЬЬլӬϬЬЬӬ֬լҬЬѬ֬ӬӬЬѬӬҬԬЬЬլլӬϬͬЬЬլϬͬάѬլϬϬ��Ь��ҬѬϬЬЬѬЬЬЬЬҬӬѬЬЬЬԬӬάѬӬѬҬϬЬ��Ѭ��ЬЬѬЬЬЬЬϬЬЬϬͬЬϬЬ֬ЬϬЬЬѬҬѬЬҬլӬԬЬЬ׬ѬϬЬЬѬ֬ѬЬάѬѬЬҬϬЬЬҬЬϬЬЬӬӬЬԬЬѬԬ֬ϬͬӬӬ׬ҬѬѬլӬӬЬЬԬӬլЬЬЬ֬ӬЬϬͬҬҬϬѬͬЬӬӬ̬ЬЬЬӬѬЬϬӬӬԬϬϬЬӬ٬֬ѬЬЬҬѬͬЬѬЬӬѬЬЬӬӬЬЬ��Ь��ЬϬϬάЬӬάϬϬҬҬЬѬϬϬЬҬЬЬЬϬҬӬЬЬӬЬҬѬЬЬЬЬӬϬάѬѬլϬϬЬ��Ѭ��ϬϬЬЬѬϬЬϬӬլϬͬЬЬЬӬҬϬЬЬҬѬϬѬЬ۬ϬϬЬЬԬӬЬd) %s: Error in write_otadata operation. err = 0x%x
:  1 otadata          OTA data         01 00 00010000 00002000
I (203) ϬЬլӬҬϬЬӬѬҬЬЬЬ֬ѬЬάѬЬЬլϬЬЬѬԬլЬɬЬլѬӬЬЬҬӬѬЬӬЬѬӬЬЬԬЬԬӬЬЬЬҬҬάϬլӬլѬЬάЬЬ׬ЬϬլԬЬӬЬЬլҬ��Ҭ��ԬԬѬЬЬӬЬӬѬЬάӬϬЬάЬάլլͬЬЬЬЬҬɬЬҬЬЬЬЬЬլѬЬЬЬЬԬӬϬЬѬѬ֬ѬѬЬҬҬЬϬլЬԬѬϬЬЬЬЬЬϬЬѬѬϬԬͬЬӬҬϬЬѬҬҬԬЬЬ׬ӬլϬЬЬѬ٬ϬЬѬѬѬӬάЬѬЬѬЬЬЬӬҬЬϬͬЬҬ֬ԬЬѬӬӬѬϬЬѬլӬլѬЬҬӬЬЬҬЬլլЬϬ֬ЬЬϬˬάѬҬӬϬЬҬѬѬЬ��Ѭ��ЬլЬάլլϬӬЬЬҬҬЬϬϬЬѬҬЬϬЬҬЬϬЬҬӬլЬϬЬЬЬҬϬάЬЬͬѬϬάѬѬЬЬϬӬҬҬϬЬЬ׬ԬӬЬάЬҬϬϬѬͬЬѬͬͬЬЬӬѬЬϬЬѬӬͬˬЬӬ׬ѬҬЬѬӬӬάͬѬѬլ֬ѬϬӬѬѬϬЬЬЬլάάЬЬҬЬͬЬҬҬЬЬЬЬѬѬЬЬӬӬЬլլάЬլӬҬͬЬѬ֬ӬϬϬЬЬ֬ϬЬϬЬЬϬЬϬԬӬϬϬ��Ϭ��֬ԬͬϬѬѬЬЬЬѬάϬҬάάЬЬάϬЬЬԬѬЬͬѬѬѬϬϬϬѬЬЬϬЬЬլҬϬάЬѬԬҬЬͬլҬԬӬЬѬ٬ѬЬϬЬҬѬЬЬϬЬҬҬӬͬͬѬѬϬѬЬάҬЬӬӬͬѬЬЬ֬ҬϬЬϬѬӬӬЬЬЬӬЬЬЬѬӬӬάάЬϬϬѬϬάЬЬͬЬЬЬӬӬЬЬϬҬӬլάѬЬ٬ҬЬЬϬЬӬͬЬѬЬ֬լЬЬӬЬѬϬЬЬЬӬӬͬάЬӬ֬ѬЬЬҬӬҬЬЬҬӬЬ֬ҬЬԬѬ��ά��ѬЬӬϬͬҬլѬЬάЬӬd) %s: Error in write_otadata operation. err = 0x%x
:  2 phy_init         RF data          01 01 00012000 00001000
I (313) Ҭ׬ϬЬѬЬ֬άҬϬϬլѬάЬѬҬѬЬϬЬӬҬѬЬЬҬӬլ׬̬ЬҬӬӬάЬҬҬҬЬЬЬլӬϬϬЬЬӬϬҬάЬӬҬάϬЬӬѬӬЬϬЬЬլѬЬҬЬЬӬЬЬӬЬϬ̬ѬЬѬԬάάϬЬҬͬϬϬЬ¬ЬϬάͬӬӬάЬЬЬӬԬϬˬ׬ѬӬѬϬѬլӬЬЬЬӬլ¬ϬЬЬЬӬӬЬάЬѬϬҬЬάլЬЬЬϬԬӬϬЬЬԬլӬҬЬЬլѬҬլЬҬլѬάϬӬѬӬЬЬЬϬլϬЬϬЬӬЬϬϬЬЬԬҬϬЬԬѬѬЬϬϬѬӬЬЬЬЬӬҬάЬЬЬҬЬάҬЬѬͬЬϬЬЬϬϬϬЬЬЬϬϬάЬӬԬЬϬЬѬѬѬͬѬҬӬԬϬЬҬЬЬϬϬЬҬЬάͬάЬ��Ӭ��̬ϬЬѬѬϬЬѬӬЬԬͬЬѬ׬ЬͬӬЬԬ׬ЬЬӬӬЬЬЬѬѬҬЬϬЬҬӬЬϬάάЬЬԬϬϬլӬͬЬЬЬӬЬҬЬѬӬӬѬЬЬҬլӬЬЬͬլЬϬͬЬϬ֬ѬϬЬҬҬάЬЬЬӬЬЬϬϬάԬͬϬϬЬЬҬЬϬϬѬЬѬЬЬЬҬҬЬЬ׬ϬӬϬϬЬӬѬϬЬάЬЬЬάЬϬЬҬѬϬάѬӬϬӬЬͬլӬҬͬЬЬЬҬͬЬЬЬЬ��Ϭ��ѬЬЬЬѬѬЬѬҬӬЬϬЬЬЬҬЬϬ��Ь��ӬάѬЬЬӬϬϬҬӬӬͬЬЬЬЬϬЬάϬϬά̬ϬάƬӬѬϬϬЬӬѬЬЬЬӬѬԬЬϬӬЬϬάЬЬӬЬЬˬЬЬӬѬϬЬЬҬЬЬЬϬҬЬϬӬЬЬӬӬάά֬ЬҬЬЬҬԬҬլЬЬҬҬӬЬЬЬ֬լЬϬϬЬѬϬҬϬЬ֬ӬϬЬѬѬЬЬЬάӬӬҬϬάάӬԬҬϬЬϬҬѬ̬ϬЬάѬЬͬЬЬҬЬͬѬЬլϬͬͬϬЬӬϬ̬d) %s: Error in write_otadata operation. err = 0x%x
:  3 nvs_keys         NVS keys         01 04 00013000 00006000
I (424) ϬЬҬЬѬϬЬӬЬЬЬЬҬӬҬ̬ѬЬЬլӬάϬЬѬЬϬѬάЬѬάЬЬЬӬЬάϬЬӬάѬͬάѬӬͬЬάϬӬӬЬͬҬӬҬӬͬѬЬլӬϬЬάЬ֬άάЬЬѬԬϬϬЬЬϬάάͬҬӬѬάˬЬѬѬЬͬѬլЬլͬЬЬӬЬլЬЬӬӬЬϬլЬӬҬ��ά��ѬҬϬϬЬѬЬЬЬЬЬЬҬЬͬԬҬάӬάάӬѬϬѬʬЬѬѬЬϬЬЬѬЬϬЬЬЬϬάЬάӬӬϬάѬЬͬҬάϬӬѬЬѬϬϬҬӬϬϬЬӬѬЬ̬ͬЬѬӬҬѬϬЬӬͬϬЬЬ׬ЬЬЬЬЬجЬϬЬЬԬӬЬͬЬլҬҬЬլӬӬҬϬϬЬҬѬӬЬЬЬӬӬˬЬЬѬլϬЬѬЬЬЬάϬҬӬЬЬάЬӬѬѬͬЬӬӬӬҬϬЬӬЬѬЬ��Ҭ��ЬЬάЬӬӬЬϬЬЬ̬ͬЬͬЬѬЬϬЬͬҬҬЬϬϬŬЬӬЬϬӬЬЬӬϬЬҬЬЬάҬѬѬЬϬЬϬӬЬϬЬϬЬլЬϬϬϬҬԬϬάЬЬӬѬͬϬҬѬԬЬЬѬլӬϬЬЬӬլڬЬϬЬЬάϬЬլЬЬҬӬѬӬЬЬҬ֬ЬӬѬѬЬӬЬͬЬЬӬѬϬϬЬЬЬϬЬϬӬլϬͬЬϬЬҬЬϬӬѬҬҬЬ̬ӬլЬѬЬլ֬ЬϬϬάЬӬѬЬάЬҬϬͬЬˬҬҬЬЬЬЬӬЬϬЬάѬӬìάͬӬЬҬѬЬЬѬѬͬάҬѬԬҬЬϬЬ֬֬ЬЬϬЬӬάŬЬϬҬѬάЬҬѬЬЬϬЬѬѬЬЬЬ֬ԬҬҬ̬ЬӬӬѬάӬЬѬЬЬϬЬѬӬϬάЬϬѬͬЬϬЬԬӬЬάϬԬӬѬάЬլѬҬЬϬЬЬЬӬЬЬЬЬάάϬЬЬҬϬάϬЬЬάͬѬЬլЬϬЬͬԬ֬ЬϬЬѬ֬֬ЬάլЬҬҬάѬҬd) %s: Error in write_otadata operation. err = 0x%x
:  4 factory_nvs      WiFi data        01 02 00019000 00006000
I (534) լЬЬЬЬìҬӬϬѬЬЬԬάЬҬҬѬЬϬϬѬЬϬϬϬЬԬԬͬϬЬЬҬЬάάЬҬӬЬάҬӬլЬЬЬЬӬլЬЬ׬ЬլҬЬЬլЬЬϬЬѬҬӬЬˬЬҬЬЬЬ̬ӬլάլЬЬӬӬͬҬԬЬլӬЬЬҬѬӬЬЬЬӬӬϬѬѬͬӬ֬ͬάϬЬЬѬЬϬӬЬͬϬЬЬҬЬϬЬϬҬӬϬϬЬЬ��Ӭ��ЬϬѬҬϬ̬ѬЬҬЬЬͬԬϬάϬ̬ЬѬ֬ϬάͬЬӬ¬άЬѬҬҬӬϬϬҬӬЬЬҬͬӬҬͬЬϬЬЬѬάЬЬЬѬѬάϬҬҬЬѬЬϬЬӬϬͬ֬ЬЬӬϬЬѬ֬ѬҬЬЬҬѬѬЬҬЬѬԬϬͬӬЬЬϬϬͬӬԬϬͬάЬҬЬЬϬҬЬЬѬϬҬլӬЬӬЬЬԬҬάˬѬѬӬѬЬάάѬЬϬϬϬҬЬЬϬЬЬЬҬˬͬЬЬάѬͬЬЬѬˬЬѬЬìѬЬϬЬҬҬϬϬѬЬ֬άϬϬЬЬìӬӬϬѬЬЬӬЬЬѬӬҬЬЬЬάҬլԬЬϬЬլլϬЬЬʬѬҬάЬЬӬϬЬϬЬҬЬЬϬѬЬ֬ЬάЬЬӬѬϬͬЬѬլЬЬЬϬҬѬЬЬЬЬЬάϬϬ֬ЬϬ̬άЬѬլϬϬͬѬѬ׬ϬϬѬѬЬ׬ЬЬӬ֬Ҭά֬Ь֬ԬάЬЬҬҬӬϬЬҬЬЬЬЬЬӬӬϬϬЬЬЬӬάάԬӬϬҬάѬѬҬЬЬѬЬӬѬϬЬӬЬҬӬЬЬӬЬϬάѬЬ֬ЬϬЬ��Ҭ��ҬЬЬҬЬЬЬϬѬԬӬˬЬЬЬѬЬ̬άЬЬЬϬЬϬǬѬͬϬϬϬѬˬˬЬЬЬάѬˬάЬӬάЬЬЬӬӬЬϬӬӬӬլάѬЬլӬϬѬЬЬӬͬЬЬϬҬҬЬϬҬϬЬЬЬЬӬӬӬϬͬЬլҬӬЬҬ֬ҬҬϬЬЬլЬԬѬЬ֬ӬЬͬլЬҬӬѬd) %s: Error in write_otadata operation. err = 0x%x
:  5 ota_0            OTA app          00 10 00020000 00340000
I (645) ЬѬЬҬԬЬЬҬҬҬϬЬЬЬԬά̬ЬЬլάͬŬЬЬЬѬЬЬҬҬЬЬѬӬЬҬѬϬЬ֬ԬҬϬЬѬլҬϬЬѬѬӬЬϬЬЬЬЬЬάլԬάͬЬͬӬլӬЬϬЬѬЬЬϬѬ׬ЬԬЬЬѬѬЬϬЬЬجҬЬͬҬѬѬͬϬЬЬլѬЬЬЬԬЬϬάЬЬӬҬЬά׬ѬҬѬЬЬլѬѬϬЬԬҬӬϬЬϬ׬ԬѬЬͬЬѬάԬЬάӬѬЬЬҬӬҬЬ��Ϭ��ӬԬЬЬҬӬӬӬЬЬҬӬӬͬЬԬӬҬЬϬЬѬӬάЬŬЬҬϬϬϬѬլӬЬάЬӬѬЬЬάЬҬլЬЬҬЬЬԬЬЬլЬӬϬάѬЬլϬͬάҬЬЬ̬ЬЬЬҬϬЬϬѬӬЬϬЬЬ׬ӬѬͬЬЬԬӬϬӬӬҬͬЬҬӬЬѬЬЬҬլϬͬ֬ЬӬҬάЬӬЬԬѬЬѬҬҬЬͬЬѬӬѬЬѬЬЬЬѬάЬլԬάѬЬӬӬЬϬЬӬӬӬҬЬЬӬԬԬϬЬЬҬѬЬϬЬϬѬҬˬЬЬӬάЬϬЬլЬϬЬάЬԬЬϬϬϬլҬҬ̬ϬЬӬѬЬЬϬҬլЬЬҬЬìϬάЬϬЬάάάЬӬլͬЬάЬѬҬЬϬѬҬЬԬЬЬҬӬЬϬЬЬЬѬЬЬЬҬѬҬϬЬѬЬЬϬЬЬ׬ԬЬϬЬЬЬѬѬάѬӬѬӬЬЬӬЬЬЬҬάԬӬάάҬЬҬѬϬͬҬЬϬάӬЬѬҬЬЬЬլЬЬϬЬլ֬άӬͬϬӬլЬҬЬѬլѬͬϬӬѬҬЬЬЬЬѬϬϬЬЬլ��ͬ��ЬЬάӬϬЬЬҬϬѬЬάӬ֬ЬЬϬӬլլάЬЬ֬լ��Ь��ӬӬϬϬѬϬҬЬͬЬѬϬЬϬ̬ЬѬӬҬͬϬЬӬӬӬϬѬӬѬӬЬЬѬӬЬҬѬЬլҬЬͬЬЬѬլͬάЬЬѬϬЬЬЬӬЬϬϬάѬӬЬͬЬd) %s: Error in write_otadata operation. err = 0x%x
:  6 ota_1            OTA app          00 11 00360000 00340000
I (755) ЬҬѬϬЬѬѬѬЬϬЬҬӬҬЬЬլլԬӬͬЬӬӬԬͬѬѬѬϬάЬϬЬҬϬЬϬЬЬϬЬά��Ҭ��ЬϬϬԬѬӬͬϬӬҬԬϬάѬϬЬӬϬЬҬЬϬάЬЬЬҬͬϬάѬլ̬ϬЬЬӬЬЬϬͬլԬЬͬЬЬլҬϬάլѬ֬ѬϬЬҬӬЬͬϬѬҬӬϬЬϬЬӬӬЬ̬ЬѬάЬѬάӬЬЬЬЬЬҬЬЬЬӬӬլЬЬѬҬҬӬҬЬѬ֬ѬϬϬЬӬϬЬЬЬЬҬάάͬϬӬάάϬЬѬѬЬάЬҬӬѬЬϬЬҬҬѬͬӬЬլӬϬϬѬѬӬӬάЬЬҬѬͬϬѬЬѬϬͬάЬЬϬЬάѬլЬάЬά��Ӭ��ˬЬЬҬѬЬЬѬ֬Ь֬άЬӬЬϬάЬЬ֬լЬͬЬѬƬϬϬЬЬӬЬЬЬάԬЬϬϬϬЬլԬЬάլЬҬӬЬЬӬҬӬЬϬѬѬЬЬЬЬԬլЬЬάЬҬάѬЬϬ֬ҬЬЬҬЬѬЬЬϬӬԬӬϬϬЬӬԬլЬЬѬҬѬϬЬЬҬҬЬϬЬҬլѬЬѬЬҬЬЬάЬ׬լͬϬЬլЬѬЬάЬҬլѬЬӬЬѬӬϬά׬ЬҬͬͬЬѬѬϬϬЬЬЬ��ά��ЬЬЬЬЬάӬӬϬάЬЬ֬ӬѬάѬЬӬҬϬѬӬҬЬЬϬЬӬЬЬάЬѬլӬЬͬЬЬάҬЬЬѬӬЬЬϬҬլЬЬӬҬЬҬҬЬҬѬѬӬѬӬѬӬѬϬάЬӬӬҬЬЬЬӬЬЬЬЬԬլЬѬЬѬӬѬͬЬЬҬЬЬЬЬլӬլЬЬЬԬ [... cut ...] ҬݬҬ׬ͬͬЬǬϬͬЬЬάѬӬҬѬѬѬͬҬЬάѬʬЬʬͬЬˬЬˬʬʬЬ��Ҭ��٬۬׬٬ϬլЬˬѬάЬЬЬҬЬ٬ӬѬѬѬѬլ̬ЬϬ��Ь��ЬЬҬͬЬЬЬԬӬЬʬԬάˬЬʬЬˬͬάǬάd) Fatal exception (28): LoadProhibited
epc1=0x40008044, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000037f, depc=0x00000000

seens that something related to the system got a very weirdo behavior while the system is booting right before the app_main could be called.

So, maybe something in esp-idf plataform code is messing with the system memory while 2nd stage and fw boot is hapening at the same time?

KonstantinKondrashov, 2025-04-23:
@filzek This log is from the 2nd stage bootloader. It tries to read the partition table and output the partition's summary. 

```
I (76) boot: 2 phy_init RF data 01 01 00012000 00001000
I (83) ҬЬ: 3 nvs_keys NVS keys 01 04 00013000 00006000
I (89) Ϭάάլ̬ѬѬάѬѬЬЬЬѬ̬ЬˬЬЬϬʬЬϬάƬͬǬϬѬѬӬӬԬݬѬլЬͬάЬЬѬϬˬ֬ӬҬ�: 4 factory_nvs WiFi data 01 02 00019000 00006000
```
Suddenly, the bootloader outputs an incorrect TAG in log messages. The TAG is a pointer to a string (the garbage is printed until zero is met). For some reason, it points to the incorrect place. Seems to me something is not reset properly especially if after the following reboot it starts normaly. I may think of an issue with the cache/MMU due to missconfigurations.

The bootloader is v4.2-dev-792-g6330b3345-dirty
The app is v5.4.1-155-g4354c3080a-dirty

I see that the secure boot v1 and flash encryption is enabled. It means that the ROM bootloader was able to read out from the flash the 2nd stage bootloder and verify its signature. So the 2nd stage bootloader was loaded correctly.

It is no longer the OTA upate (rollback) issue. I suggest to open a new issue for that. Please provide the skconfigs for bootloader and app.

KonstantinKondrashov, 2025-04-23:
> do you mean open an issue with the cache/MMU bootloader?

Sorry, it's just a guess. I do not know what it might be. I suggest opening a new issue for this https://github.com/espressif/esp-idf/issues/15552#issuecomment-2822793200. The colleagues who support it will have a look.

Have you tried, for example, v5.3 to run? Which version works well?

filzek, 2025-04-25:
Got some more info on this:

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6608
ho 0 tail 12 room 4
load:0x40078000,len:16140
load:0x40080400,len:4048
entry 0x40080658
I (31) boot: ESP-IDF v4.2-dev-792-g6330b3345-dirty 2nd stage bootloader
I (31) boot: compile time 01:02:27
I (31) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) ▒6▒▒ow▒V▒▒ow▒V▒O▒6▒▒o▒O▒V▒▒ow▒V▒O▒6▒▒o▒O▒6▒▒ow▒V▒O▒6w▒o▒O▒6▒▒ow▒V▒O▒6▒V▒O▒6▒▒ow▒V▒▒6w▒V▒O▒6▒▒ow▒V▒▒o▒V▒O▒6▒▒o▒O▒V▒▒o▒V▒▒o▒v▒                                                                                V▒O▒6▒▒o▒O▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒                                                                                o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒V                                                                                ▒O▒6▒▒o▒v▒V▒▒6▒v▒V▒O7▒▒o▒v▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒O▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o                                                                                ▒v▒V▒O▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒6▒v▒V▒O                                                                                ▒6▒▒o▒v▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒V▒O▒6▒▒o▒v▒V▒▒6▒v▒V▒O7▒▒o▒v▒V▒▒o▒v▒V▒O▒6▒▒o▒O▒V▒▒o▒v▒                                                                                V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒O▒6▒▒o▒O▒6▒▒o▒v▒V▒O▒6▒v▒o▒O▒6▒▒o▒v▒V▒OI (126) boot: Partition Table:
I (129) boot: ## Label            Usage          Type ST Offset   Length
I (135) boot:  0 nvs              WiFi data        01 02 0000a000 00006000
I (142) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (148) boot:  2 phy_init         RF data          01 01 00012000 00001000
I (155) boot:  3 nvs_keys         NVS keys         01 04 00013000 00006000
I (162) boot:  4 factory_nvs      WiFi data        01 02 00019000 00006000
I (168) boot:  5 ota_0            OTA app          00 10 00020000 00340000
I (175) boot:  6 ota_1            OTA app          00 11 00360000 00340000
I (181) boot:  7 storage          Unknown data     01 81 006a0000 00090000
I (188) boot:  8 blockchain       Unknown data     01 81 00730000 00001000
I (195) boot:  9 block_nvs        WiFi data        01 02 00731000 00010000
I (201) boot: 10 fctry            WiFi data        01 02 00741000 00006000
I (208) boot: End of partition table
I (211) boot_comm: chip revision: 3, min. application chip revision: 1
I (217) esp_image: segment 0: paddr=0x00360020 vaddr=0x3f400020 size=0x59a38 (367160) map
I (381) esp_image: segment 1: paddr=0x003b9a60 vaddr=0x3ff80000 size=0x0001c (    28) load
I (381) esp_image: segment 2: paddr=0x003b9a84 vaddr=0x3ffbdb60 size=0x054f8 ( 21752) load
I (396) esp_image: segment 3: paddr=0x003bef84 vaddr=0x40080000 size=0x01094 (  4244) load
I (398) esp_image: segment 4: paddr=0x003c0020 vaddr=0x400d0020 size=0x159d3c (1416508) map
I (1003) esp_image: segment 5: paddr=0x00519d64 vaddr=0x40081094 size=0x1d88c (120972) load
I (1061) esp_image: segment 6: paddr=0x005375f8 vaddr=0x00000000 size=0x08988 ( 35208)
I (1094) boot: Loaded app from partition at offset 0x360000
I (1095) boot: Checking flash encryption...
▒oI (1095) boot: Disabling RNG early entropy source...
E (1993) spi_flash: Detected size(8192k) smaller than the size in the binary image header(261120k). Probe failed.

assert failed: __esp_system_init_fn_init_flash startup_funcs.c:95 (flash_ret == ESP_OK)

Backtrace: 0x400821b4:0x3ffe3a50 0x400989f1:0x3ffe3a70 0x4009c681:0x3ffe3a90 0x400d627a:0x3ffe3bc0 0x400d6b43:0x3ffe3be0 0x400d6b76                                                                                :0x3ffe3c00 0x40081a12:0x3ffe3c30 0x40079961:0x3ffe3c80 |<-CORRUPTED

ELF file SHA256: 759269b1f

Rebooting...

So, this seens that some operations done before the esp_restart() is corrupting the memory wich the restart happens so, maybe something triggers the otadata problem or something more dangerous

AxelLin, 2025-05-28:
I hit the same issue with v5.4.1-743-gaed8bdc8dd using ESP32-WROOM-32E-N8R2 module.