espressif/esp-idf #15558: Ability to scan multiple access point SSIDs at once

bryghtlabs-richard, 2025-03-11:
### Describe the solution you'd like.

I'd like the ability to specify multiple SSIDs in a `wifi_scan_config_t`.

### Describe alternatives you've considered.

Currently to connect we do something like this pseudocode:

```
for each (SSID, Password) in CustomWifiManager's Credential List:
    if WiFiScan(SSID) is found:
        if Connect(SSID,Password) is okay:
            break
```

But, for SSIDs near the end of the list, we have to perform multiple scans first, which greatly lengthens the connection time.

### Additional context.

We have a few product requirements to balance:

- Must support connecting to loudest AP in a mesh networks - for this we perform an all-channel scan before connecting to identify the best node to connect with. **This is needed for pre-802.11kr access points.**
- Must support multiple [SSID,password] pairs of credentials.
- Must support connecting to hidden SSIDs. **To support meshes with hidden SSIDs, this currently requires a separate scan per SSID**.
- Should support connecting to the loudest SSID if multiple are available.
- Should have fast connect time

mhdong, 2025-03-17:
hi @bryghtlabs-richard 
To scan for a specific SSID, we include the SSID field in the probe request. However, this field can only carry one SSID at a time, making it impossible to scan multiple AP SSIDs at once.
If hidden APs are not a concern, a full-channel scan can be performed first, followed by filtering the desired SSIDs. However, if hidden SSIDs need to be considered, each SSID must still be scanned individually.

bryghtlabs-richard, 2025-03-17:
>To scan for a specific SSID, we include the SSID field in the probe request. However, this field can only carry one SSID at a time, making it impossible to scan multiple AP SSIDs at once.

@mhdong, 802.11 has supported "SSID List" probe requests since at least 2012.

Please consider the following steps:

1. Allow sending multiple probe requests on a single channel during the scan so that `wifi_scan_config_t.scan_time` only applies per-channel instead of per-channel, per-SSID.
2. Allow coalescing multiple SSIDs into fewer `SSID List` probe request packets. If the SSID List is too long for a single probe request, a few may be required. The relevant section in 802.11-2012, "8.3.3.9 Probe Request frame format", "Table 8-26—Probe Request frame body", frame field/order 10. In 802.11ax-2021, it's "Table 9-38—Probe Request frame body"

mhdong, 2025-03-18:
hi @bryghtlabs-richard 
Thank you very much for your correction. We will discuss internally whether to add support for this feature and how to implement it.

bryghtlabs-richard, 2025-03-18:
Thank you for your consideration @mhdong .

rsaxvc, 2026-10-04:
It looks like "SSID list" was added after 2007. I've wondered how to handle probe compatibility with older APs, but I think I have an answer.

Before the pseudocode above, do an additional scan like this: SSID:wildcard + SSID list: all SSIDs of interest.

* AP with SsidList support would appear in the scan list results regardless of public or private SSID
* AP without SsidList support but with public broadcast it would appear in the scan results.
* AP without SsidList support and private SSID would not appear,but could still be detected by fallback loop above.