espressif/esp-idf #15559: NVS blob does is not mutually exclusive with other NVS types

robotman2412, 2025-03-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

a6c3a9cbbb5e0a99e0d30e58271f27a0d4e9ae90

### Espressif SoC revision.

ESP32-P4 1.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

USB

### What is the expected behavior?

I expected NVS entries with the same key/namespace pair to be mutually exclusive between the types; only one type is can exist for such a key/namespace pair.

### What is the actual behavior?

Writing a blob to NVS after writing something else (e.g. string) will not delete the earlier value but still add the blob.

### Steps to reproduce.

1. Write a string to NVS
2. Write a blob to the same namespace and key
3. List the NVS entries
4. Observe that both exist

### Debug Logs.

```plain
Robot tools$ ./badgelink.py nvs list
namespace | key      | type  
----------+----------+-------
testns    | testkey2 | string
testns    | testkey  | i32   
Robot tools$ ./badgelink.py nvs write testns testkey string "abcd"
Robot tools$ ./badgelink.py nvs list
namespace | key      | type  
----------+----------+-------
testns    | testkey2 | string
testns    | testkey  | string
Robot tools$ ./badgelink.py nvs write testns testkey blob "abcd"
Robot tools$ ./badgelink.py nvs list
namespace | key      | type  
----------+----------+-------
testns    | testkey2 | string
testns    | testkey  | string
testns    | testkey  | blob  
Robot tools$ ./badgelink.py nvs write testns testkey i32 123
Robot tools$ ./badgelink.py nvs list
namespace | key      | type  
----------+----------+-------
testns    | testkey2 | string
testns    | testkey  | blob  
testns    | testkey  | i32   
Robot tools$
```

KaeLL, 2025-03-12:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_partition_gen.html#caveats

robotman2412, 2025-03-17:
> https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_partition_gen.html#caveats

This does not use the NVS generator; the tool you see is communicating with an ESP32-P4 over USB, and the ESP32-P4 itself interacts with NVS in response to these requests.

rrtandler, 2025-03-17:
Hi @robotman2412,
Could you share the c-code which interacts with ESP IDF NVS library functions ? Especially the code fragments related to the "write" and "list" commands.

robotman2412, 2025-03-17:
> Hi [@robotman2412](https://github.com/robotman2412), Could you share the c-code which interacts with ESP IDF NVS library functions ? Especially the code fragments related to the "write" and "list" commands.

Unfortunately, I cannot post that publicly for another couple months, but I can tell you this.

Badgelink is a protocol we made to communicate over USB (the ESP32 is the "badge"). It has a request that handles NVS accesses. This NVS request is implemented by opening the namespace with NVS_READONLY for list and read, and NVS_READWRITE for write and delete. After that, the NVS list function will create an iterator like so:
```c
    // Try to open NVS.
    uint32_t       offset = req->list_offset;
    nvs_iterator_t iter;
    esp_err_t ec = nvs_entry_find(NVS_DEFAULT_PART_NAME, req->namespc[0] ? req->namespc : NULL, NVS_TYPE_ANY, &iter);
```

In this case, the namespace is not NULL but the behaviour doesn't depend on that.
After opening NVS in this way, it will use `nvs_entry_next` followed `nvs_entry_info` until the former returns `ESP_ERR_NVS_NOT_FOUND`. All the entries are formatted for transmission over USB and sent back to the host. I am very confident that the bug is not in our code because we use protobuf for it and we also double-check the CRC32 of the entire message and we don't do any caching.

rrtandler, 2025-03-20:
Hi @robotman2412 ,

I have reproduced the behaviour you are describing in the host test and working on the fix. I'll come back as soon as the fix is merged to the master.

rrtandler, 2025-04-14:
Hi @robotman2412,
The fix is now merged into the master branch od IDF. Commit SHA 0b8f661c081

closed 2025-04-14 by rrtandler