espressif/esp-idf #15560: LittleFS plus USB

diplfranzhoepfinger, 2025-03-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

DRAFT

tore-espressif, 2025-03-14:
@diplfranzhoepfinger Please provide more information about your issue. Eg. logs, reproducible code, configuration etc

> this combination I feel is not healthy.
it seems the USB Package draws fatfs Package and makes weird things.

This is not enough information for us to start debugging

Alvin1Zhang, 2025-03-26:
@diplfranzhoepfinger Thanks for reporting, would you please help share more details for the issue? Thanks.

diplfranzhoepfinger, 2025-03-26:
Yes. Understand. 
My issue is gone with the removal of USB. Soni must re-start a Sample which is shareable and reproduceable.

adokitkat, 2025-05-02:
Hello, what is the conclusion here? How did the instability you have seen manifest?

diplfranzhoepfinger, 2025-11-24:
i guess that with https://github.com/espressif/esp-idf/releases/tag/v6.0-beta1 this instability might be removed. 

5.5.1 was drawing odd packaged, like FATFS or Driver even not needed. 

6.0 seems to improve this. 

i will test again soon.