espressif/esp-idf #15562: Task WDT on FreeRTOS UniCore reset by task on CPU1

FnxQT, 2025-03-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.8

### Espressif SoC revision.

ESP32-D0WD-V3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I configured my ESP-IDF with FreeRTOS in unicore mode:

```
#
# FreeRTOS
#
CONFIG_FREERTOS_UNICORE=y
CONFIG_FREERTOS_CORETIMER_0=y
# CONFIG_FREERTOS_CORETIMER_1 is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION=y
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_NONE is not set
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_PTRVAL is not set
CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY=y
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y
CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER=y
# CONFIG_FREERTOS_RUN_TIME_STATS_USING_CPU_CLK is not set
CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID=y
```

and with a task watchdog timer:

```
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=30
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=y

```

So all my tasks are running on core 0. I could verify that. I have four tasks monitored by the task WDT:

```
I (190908) task_wdt: Tasks monitored by the Task WDT:
I (190908) task_wdt: CPU: CPU 0 - IDLE
I (190908) task_wdt: CPU: CPU 0 - out
I (190908) task_wdt: CPU: CPU 0 - bcmd
I (190908) task_wdt: CPU: CPU 0 - main
```

All on core 0. In that case, something happening on CPU1 should not bother my system at all.

### What is the actual behavior?

The system is reset because an unknown task on CPU1 has reset the watchdog time.

### Steps to reproduce.

That's the bad part, I cannot replicate it easily elsewhere. I don't want it treated as a bug immediately but to understand if this behaviour is known or even "correct" ? The task on CPU 1 do not even have a name.

### Debug Logs.

```plain
That's what I see from the Task WDT reset:

(804763) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

(804763) task_wdt:  -  (CPU 1)

(804763) task_wdt: Tasks currently running:

(804763) task_wdt: CPU 0: IDLE

(804763) task_wdt: Aborting.

abort() was called at PC 0x401581a9 on core 0
```

o-marshmallow, 2025-03-14:
Hello @FnxQT ,

By default, idle tasks for both CPUs are registered to the task watchdog here: https://github.com/espressif/esp-idf/blob/v4.4.8/components/freertos/port/port_common.c#L137

Since you are executing on a single core, this should not be the case for CPU1, make sure that `CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1` is **not** part of your `sdkconfig`.

It seems like you added some tasks to watch to the watchdog via the function `esp_task_wdt_add`, I suspect that one of tasks registered got deleted or freed at some point which could lead to this corruption. A corruption would explain why you don't always get this problem, it's random.
The fact that ` -  (CPU 1)` is shown means that the (inexistant) task's affinity is not 0 nor `tskNO_AFFINITY`, but it doesn't necessarily means it's 1: https://github.com/espressif/esp-idf/blob/v4.4.8/components/esp_system/task_wdt.c#L158

You can check (log) the addresses of all the Tasks you create and the ones that are part of `twdt_config->list` above, maybe it will help you see which one was corrupted (if it is a corruption)

FnxQT, 2025-03-14:
Hello @o-marshmallow 

Thank you very much your help! It pointed me into the right direction. What I didn't saw from my logs was that one of the task monitored by the watchdog was terminated:

```
(504765) BCOMMAND: Select error

(504766) BCOMMAND: Task exit
```

And that connects directly with what you said : a task must have been corrupted, which is this one since it doesn't exist anymore. Everything is clear now. Thank you very much.

closed 2025-03-14 by FnxQT

o-marshmallow, 2025-03-17:
Awesome, thanks for coming back with the explanation! 😄