espressif/esp-idf #15563: duplicated Namespace in one nvs partitioin.

junyoun-kim, 2025-03-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm developing ESP32 with release/v5.0 branch.

And now I'm using default nvs partition with namespace named "stdk".

But when I'm testing it, I realize that whenever boot up, and open "stdk" namespace, it open different data set randomly.

It looks like there are two different data set under same name "stdk" namespace.

So I dumped default nvs partition and found out there were actually two "stdk" namespaces with different index.

Here is some of nvs partition dump,

.........
Page no. 96, Status: Active, Version: 2, CRC32: 61b0294b, Page address: 0x1000
 Entry state bitmap: aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa fe ff ff ff ff ff
 000. Written, Namespace Index: 000, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 1e47c9d6 | stdk : 1
 001. Written, Namespace Index: 001, Type: string    , Span: 002, Chunk Index: 255, CRC32: 3d5b87ae | PKType : Size=8, CRC32=e26df9c2
      0x000  45 44 32 35 35 31 39 00 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ED25519.ÿÿÿÿÿÿÿÿ ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
 003. Written, Namespace Index: 001, Type: string    , Span: 003, Chunk Index: 255, CRC32: 85634d2b | PublicKey : Size=45, CRC32=712fc5a8
      0x000  58 76 52 44 32 48 79 6e 6a 33 47 68 34 69 4c 45  72 4a 79 30 6b 67 54 49 76 4b 75 6f 55 7a 71 4f  XvRD2Hynj3Gh4iLE rJy0kgTIvKuoUzqO
............
Page no. 95, Status: Full, Version: 2, CRC32:  318a3c5, Page address: 0x2000
 Entry state bitmap: aa aa aa aa aa aa aa aa aa aa aa aa aa 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f0 ff
 000. Written, Namespace Index: 000, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 70242b5f | misc : 1
 001. Written, Namespace Index: 000, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 8a20d492 | nvs.net80211 : 2
 002. Written, Namespace Index: 002, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 9caf9616 | ap.sndchan : 1
 003. Written, Namespace Index: 000, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 5c62ceab | stdk : 3
 004. Written, Namespace Index: 000, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 69d6725e | phy : 4
 005. Written, Namespace Index: 004, Type: blob_data , Span: 002, Chunk Index: 000, CRC32: c9e17970 | cal_mac : Size=6, CRC32=acc4e21f
 .........

Is it expected behavior? or bug? Could you please explain my situation?

Please let me if you need more information.

RathiSonika, 2025-03-12:
Could you try erasing flash before writing data using the command `idf.py erase-flash`, and then re-flash the chip with `idf.py flash`?

junyoun-kim, 2025-03-13:
@RathiSonika I can do that and fix the problem. but I'm afraid this could happen again. Is it known issue?

rrtandler, 2025-03-17:
Hi @junyoun-kim,
This is not a known issue. 
To me it looks like a change of the "nvs" labelled partition offset in the flash memory without respective flash erase sometimes in the past. 
Have you modified the partition.csv file or offset of the partition table beginning (via CONFIG_PARTITION_TABLE_OFFSET) during your application development ?

junyoun-kim, 2025-03-17:
@rrtandler Thank you for reply. But I don't think I changed offset of partition table.

And also even assuming I did changed the offset of partition table, selecting random data set among two same namespaces looks like inconsistant behavior.

I don't know exact internal initialization process of nvs partitioin, but does it load flash data randomly? So it loads the namespace data it encounters first?

I'm guessing there sholud be some error like duplicate namespace encontered or consistant behavior like last namespace data set is selected.

rrtandler, 2025-03-20:
Hi @junyoun-kim ,

The presence of double records binding the namespace "stdk" to two different values of index (1 and 3) indicates that during the opening of the namespace "stdk", the library was not aware of it's existence. This let the library create the namespace index for a second time.
The shifted beginning of the real NVS flash space was the first reason that came to my mind as a reason why to miss the existing namespace record.

The initialisation of NVS reads all the NVS pages from the partition and keeps all the namespace name - index pairs for opening of them later on. It doesn't attempt to detect duplicities in namespace indexes in order to keep initialisation duration reasonably fast, primarily, because namespace records are maintained internally only and there are no delete or update operations foreseen for namespaces.
 
The erase flash before flashing should help you in your case.

If you encounter the same issue again, please try to gather and share details helping us to reproduce the case.

junyoun-kim, 2025-03-21:
@rrtandler Okay, I'll let you know when this occurs again. And as I said, it might be helpful that during nvs initialization, it detects duplicate namespace and let developer know. So it can reduce debugging time.