espressif/esp-idf #15565: Implementing Mesh Network with ESPNOW in IDF Example

gajananmaske1234, 2025-03-12:
### Is your feature request related to a problem?

 Hi,

I am currently using the ESPNOW code from the IDF example and want to implement a mesh network scenario. In this setup, I have three devices: Slave 1, Slave 2, and Master. The issue is that Slave 2 is out of range from the Master, but in range with Slave 1.

My goal is to have Slave 2 send a message through Slave 1 to the Master, so that even though Slave 2 is out of range from the Master, it can still communicate via Slave 1.

Could you please guide me on how to modify the code to implement this functionality, where Slave 1 forwards the message from Slave 2 to the Master?

Thanks in advance for your assistance!

nopnop2002, 2025-03-15:
Here is the esp-mesh-lite user guide.
This includes the ESPNOW Usage Guide.
Perhaps it is possible with esp-mesh-lite.

https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide.md#difference-between-esp-mesh-lite-and-esp-mesh

zhangyanjiaoesp, 2025-03-19:
@gajananmaske1234 you can refer to this :
https://github.com/espressif/esp-mesh-lite/blob/master/examples/rainmaker/led_light/components/group_control/app_espnow.c

gajananmaske1234, 2025-03-26:
Thanks for your response,

[image]

As shown in above diagram I am using mesh lite example and espnow and send receive API.

Now using esp lite example i am expecting master to layer 2 device should communicate through Slave 1 device but not working as expected.

slave to connecting with slave1 as a child confirmed in logs.

Want to know can i do communication as above if yes then how? is there any example please guide.

nopnop2002, 2025-06-10:
>ESPNOW can still use them to add peers and send packets.

Oh!!
I didn't know.
Thank you.