espressif/esp-idf #15566: Does ESP-IDF 5.3.2 support old I2C driver?

mike-2020, 2025-03-12:
ESP32S3
ESP-IDF 5.3.2
ADF 2.7

Received the following errors during booting time:

`I (1033) spi_flash: flash io: qio
W (1037) spi_flash: Detected size(32768k) larger than the size in the binary image header(16384k). Using the size in the binary image header.
E (1050) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

abort() was called at PC 0x420da8eb on core 0
--- 0x420da8eb: check_i2c_driver_conflict at /home/user/SDK/esp-idf-v5.3.2/components/driver/i2c/i2c.c:1719`

There are two I2C usages in the project:
ADF codec: but I have enabled the option "Enable backward compatibility for the I2c driver (force use of the old I2c_driver above v5.3)"
MPU6050 driver: it uses I2CDev driver which is based on the old I2C driver.

Why the errors above? Does ESP-IDF 5.3.2 support old I2C driver? (there was no such issue when I used IDF 5.2.1)
How can I fix this problem?

mythbuster5, 2025-03-13:
It supports the old driver but you can't use a component with new driver and a component with old driver together. And we are updating the component to new driver step by step. But seems the MPU6050 is still use the old one.

mike-2020, 2025-03-13:
> It supports the old driver but you can't use a component with new driver and a component with old driver together. And we are updating the component to new driver step by step. But seems the MPU6050 is still use the old one.
> 
> If you force use old driver in 5.3 in ADF codec, you need to check whether other part still use new i2c and it has been linked. You can just disable build the new driver and see where has a compiling error

Thanks! but how to disable the new driver in build?

greenaddress, 2025-03-13:
@AxelLin do you know if the esp_lcd_touch_ft5x06 driver in the esp idf components repository is still using the old i2c driver and do you know when it will use the new i2c driver? on 5.4 you can only use the new driver and touch seems the only thing not updated from the official packages.

https://components.espressif.com/components/espressif/esp_lcd_touch_ft5x06/versions/1.0.7

thanks!

mike-2020, 2025-03-14:
Revert to IDF v5.2.5, with ADF v2.7, everything works fine now.

It seems there is issue with either IDF v5.3 or ADF v2.7.

mythbuster5, 2025-04-11:
Thanks, we will gratually replace legacy i2c to new i2c in 3rd library maintained by espressif. Other 3rd component will rely on community. i will close this issue now.

closed 2025-04-11 by mythbuster5