espressif/esp-idf #15568: [esp32-p4] openthread border router

chegewara, 2025-03-12:
Hi team,
is there any chance to have border router support on P4.
What i mean is precompiled library `libopenthread_br.a`.

Thanks

chshu, 2025-03-13:
Yes, it's definitely possible. This will be a P4 + C6 + H2 combination solution, stay tuned.

Feel free to share your project requirements and timeline, and we may adjust our priorities accordingly.

chegewara, 2025-03-13:
Hi,
there is no project yet, just i thought i can finally test br, so its very low priority from my side and no rush.

Thanks

FnxQT, 2025-03-28:
Hi @chshu,

I'm capitalizing on this topic. As a company we are very interested in the capacity of the P4 to be used as a Thread Border Router. I was looking for example on how to set this up on ESP32-P4 DevKit since it seems to have everything required (P4 as main chip for Ethernet + C6 as the Thread Chip).

However as I understand there is no solution yet and we cannot run a Thread Border Router on the P4 ?

When would this be possible ? Do you think it will ever be possible on ESP32-P4 DevKit hardware ?

chshu, 2025-03-29:
@FnxQT It's possible to enable Thread BR on P4, but the P4 Thread BR library is not available yet.

We will provide the P4 based Thread BR solution with two configurations:

- **Ethernet based**: P4 + Ethernet Module + H2 (RCP)
- **Wi-Fi based**: P4 + C6 (Hosted Wi-Fi) + H2 (RCP)

Plan to launch the demo example by the end of May.

chshu, 2025-03-29:
Follow up in https://github.com/espressif/esp-thread-br/issues/136

closed 2025-03-29 by chshu