espressif/esp-idf #15570: BT BLE MESH 组件 menuconfig中个别配置不可显示以配置

ShaoMingya, 2025-03-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.3.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

https://github.com/espressif/esp-idf/blob/1f46216a7214e27c1e351b4310acf7fcd77029d4/components/bt/esp_ble_mesh/Kconfig.in#L1596

为啥这个不（能）允许在menuconfig中配置？

[image]

### What is the actual behavior?

无法显示配置选项。

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

forx157, 2025-03-31:
Because currently we do not recommend changing this configuration.

ShaoMingya, 2025-04-01:
> Because currently we do not recommend changing this configuration.

目前ESP32使用vendor server，手机发送自定义协议消息。有时会报出警告“Duplicate SDU from src。。。。”导致ESP32漏掉一条消息。目前我把那个配置改成了 default n。不太清楚触发那个警告的原因？

forx157, 2025-04-01:
It is possible that because you used your phone to broadcast two consecutive segment messages to esp32, the second segment message was received by esp32 before the first segment message was determined to have failed to receive the packet, at which point esp32 would discard the first segment message that did not receive complete, and instead start receiving the latest packetized message (which is the second packetized message).

You can share the sdkconfig and log with me after you change the mesh log to info level and I can help you determine the problem more accurately.