espressif/esp-idf #15571: Bug - nvs_partition_gen.py V5.3.1 doesn't encrypt NVS but V5.2.1 does

akEmbeddedSW, 2025-03-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

nvs_partition_gen.py V5.3.1 generates plain NVS but can not create encrypted NVS using the same csv.
However, nvs_partition_gen.py V5.2.1 works fine.
Setup was the same for all of the commands.

[image]

akEmbeddedSW, 2025-03-17:
Test with idf V5.4 and still the same issue.

adokitkat, 2025-03-17:
Hello, can you please run `pip show esp-idf-nvs-partition-gen | grep Version:` and tell me what version you're using? Thank you.

adokitkat, 2025-03-17:
I suspect you're on version 0.1.4 or 0.1.5. Please update the package with `pip install esp-idf-nvs-partition-gen==0.1.8` and try to run your commands again please.

akEmbeddedSW, 2025-03-17:
Hi, thanks for the response. Update did solve the issue

Version: 0.1.4
                           Version 2.0, January 2004
   Licensed under the Apache License, Version 2.0 (the "License");

I installed IDF using VS Code extension. Shouldn't it install the latest versions of tools?

adokitkat, 2025-03-17:
I will take a look into it. When did you install the IDF via the extension?

akEmbeddedSW, 2025-03-17:
Brilliant. I installed version 5.4 today.