espressif/esp-idf #15573: Its posible work wirh AC servo PMSM 110-220v

artayasa, 2025-03-14:
### Is your feature request related to a problem?

Good day, very i terst with your project. It is posible or suport to make for ac servo pmsm 110-220v. So i will coustem or used ex ac servo drive for the driver and power. So esp32 as the controler with 6pwm right, and add some code to make control by step/dir mate. Thank u, any idea?? Because i have plenty ex ac servo drive, buth with outh driver, just have some driver but no motor sir. 
The motor with encoder incremental 2000ppr or 20bit incremental sir.

suda-morris, 2025-03-17:
ESP32 features the MCPWM peripheral, so it's possible to drive the PMSM with SVPWM. There's a simple example about the SVPWM: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_foc_svpwm_open_loop

WRT the step/dir motor control, I'm not sure what's the exact feature are you asking for. But there's an example that ESP32 can generate the step/dir signals to drive a stepper motor: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/stepper_motor

artayasa, 2025-03-17:
> ESP32 features the MCPWM peripheral, so it's possible to drive the PMSM with SVPWM. There's a simple example about the SVPWM: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_foc_svpwm_open_loop

Its posible also with close loop?? With 2000ppr encoder or 20bit encoder ABZ??? 

> WRT the step/dir motor control, I'm not sure what's the exact feature are you asking for. But there's an example that ESP32 can generate the step/dir signals to drive a stepper motor:

I want control that pmsm motor closeloop by strp/dir signal mate. So the esp32 recive step and dir signal from PLC. And run the motor pmsm as per step/dir. ( if u know AC servo drive control with step/dir signal) i want like that mate. Couse i have plenty as servo pmsm without driver. So i want to make the controler from esp32. I have the powerstage can control by HIN and LIN signal mate also lowside or inline shunt resistor. But till now not yet find controler that can run pmsm by step/dir. STMBL are ok but no more part for it, the igbt and other very difficult to find or buy mate. Thank u

suda-morris, 2025-03-17:
> Its posible also with close loop?? With 2000ppr encoder or 20bit encoder ABZ???

Yes, it's possible. BTW, you may be interested in [this example](https://github.com/espressif/esp-iot-solution/tree/master/examples/motor/foc_velocity_control)

artayasa, 2025-03-17:
> Yes, it's possible. BTW, you may be interested in

Do u have example for position control closed loop?? I want to control position the pmsm motor by step/dir ( like stepper motor) very"interestes with this project. Its example using shunt resistor?? Voltage devider for each phasa and vbus sir?

artayasa, 2025-03-18:
> No, we don't have an example that uses the "step/dir" interface to do the position control of a PMSM

Not for step/dir interface. Just position control with quadature encoder ABZ. With shunt resistor?

artayasa, 2025-03-18:
> No, we don't have an example that uses the "step/dir" interface to do the position control of a PMSM.

Or if posible make a example to this mate, i think some people need it.

suda-morris, 2025-03-19:
@artayasa I think this is the closest example that can match your requirement. https://github.com/espressif/esp-iot-solution/blob/master/examples/motor/foc_velocity_control/main/main.cpp If you have more related question about PMSM motor FOC control, feel free to ask them in https://github.com/espressif/esp-iot-solution/issues Thanks.

artayasa, 2025-03-19:
> If you have more related question about PMSM motor FOC control, feel free to ask them in https://github.com/espressif/esp-iot-solution/issues Thanks.

Okee i will ask there mate🙏🙏

closed 2025-03-20 by Alvin1Zhang