espressif/esp-idf #15575: idf.py size-files is reporting false information

X-Ryl669, 2025-03-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Running the command `idf.py size-files` returns a list of files with their supposed impact on the final binary. However, it reports dumb or wrong results.

For example, here's what it gives as output:
```
                                                                                   Per-file contributions to ELF file
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━┳━━━━━━┳━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━┓
┃ Object File                              ┃ Total Size ┃ DIRAM ┃ .bss ┃ .data ┃ .text ┃ Flash Code ┃ .rodata ┃ .appdesc ┃ .text ┃ LP SRAM ┃ .rtc_reserved ┃ .rtc_noinit ┃ .data ┃ .force_fast ┃ .text ┃
┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━╇━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━┩
│ esp_app_desc.c.obj                       │     102529 │    10 │   10 │     0 │     0 │     102519 │  101729 │      256 │   534 │       0 │             0 │           0 │     0 │           0 │     0 │
│ main.cpp.obj                             │      57159 │   738 │  698 │    40 │     0 │      55165 │    3033 │        0 │ 52132 │    1256 │             0 │        1256 │     0 │           0 │     0 │
│ ecp_curves.c.obj                         │      34946 │     0 │    0 │     0 │     0 │      34946 │   29540 │        0 │  5406 │       0 │             0 │           0 │     0 │           0 │     0 │

--truncated--
```

Strange, the first listed item is something that should only contain the firmware version and few other information, so let's dig in:

```
$ ls -l build/esp-idf/esp_app_format/libesp_app_format.a
-rw-r--r-- 1 mine mine 27596 14 march  18:10 build/esp-idf/esp_app_format/libesp_app_format.a
```
Ok, a library containing the esp_app_desc.c.obj is smaller than the reported file size contribution... 
Let's check this better:
```
$ riscv32-esp-elf-readelf -sW build/esp-idf/esp_app_format/libesp_app_format.a

File: build/esp-idf/esp_app_format/libesp_app_format.a(esp_app_desc.c.obj)

Symbol table '.symtab' contains 386 entries:
   Num:    Value  Size Type    Bind   Vis      Ndx Name
     0: 00000000     0 NOTYPE  LOCAL  DEFAULT  UND
     1: 00000000     0 FILE    LOCAL  DEFAULT  ABS esp_app_desc.c
     2: 00000000     0 SECTION LOCAL  DEFAULT    1 .text
     3: 00000000     0 SECTION LOCAL  DEFAULT    2 .data
     4: 00000000     0 SECTION LOCAL  DEFAULT    3 .bss
     5: 00000000     0 SECTION LOCAL  DEFAULT    4 .text.esp_app_format_init_elf_sha256.part.0
     6: 00000000    90 FUNC    LOCAL  DEFAULT    4 esp_app_format_init_elf_sha256.part.0
     7: 00000000     0 NOTYPE  LOCAL  DEFAULT    4 .L0
     8: 00000000     0 NOTYPE  LOCAL  DEFAULT    4 .L0
     9: 00000000     0 NOTYPE  LOCAL  DEFAULT    4 $xrv32i2p1_m2p0_a2p1_c2p0_zicsr2p0_zifencei2p0_zmmul1p0_zaamo1p0_zalrsc1p0
    10: 00000000     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    11: 00000004     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    12: 0000000c     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    13: 00000012     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    14: 00000016     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    15: 00000018     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    16: 0000001a     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    17: 0000001a     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    18: 0000001e     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    19: 00000026     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    20: 00000026     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    21: 00000026     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    22: 00000026     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    23: 00000026     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    24: 00000028     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    25: 00000028     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    26: 0000002c     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    27: 00000030     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    28: 00000034     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    29: 00000038     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    30: 0000003c     0 NOTYPE  LOCAL  DEFAULT    4 .L0
    31: 0000003c     0 NOTYPE  LOCAL  DEFAULT    4 .L0
[... 334 lines cut ...]
   366: 00000026     0 NOTYPE  LOCAL  DEFAULT    4 .LBB23
   367: 0000003c     0 NOTYPE  LOCAL  DEFAULT    4 .LBE23
   368: 0000003c     0 NOTYPE  LOCAL  DEFAULT    4 .LBB24
   369: 0000004c     0 NOTYPE  LOCAL  DEFAULT    4 .LBE24
   370: 00000018     0 NOTYPE  LOCAL  DEFAULT   11 .LBE53
   371: 00000020     0 NOTYPE  LOCAL  DEFAULT   11 .LBB56
   372: 0000002a     0 NOTYPE  LOCAL  DEFAULT   11 .LBE56
   373: 00000000     0 SECTION LOCAL  DEFAULT   29 .comment
   374: 00000000     0 SECTION LOCAL  DEFAULT   31 .debug_frame
   375: 00000000     0 SECTION LOCAL  DEFAULT   33 .riscv.attributes
   376: 00000000    10 OBJECT  GLOBAL DEFAULT   15 app_elf_sha256_str
   377: 00000000   256 OBJECT  WEAK   DEFAULT   16 esp_app_desc
   378: 00000000    10 FUNC    GLOBAL DEFAULT    6 esp_app_get_description
   379: 00000000    90 FUNC    GLOBAL DEFAULT    8 esp_app_get_elf_sha256
   380: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND memcpy
   381: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND __stack_chk_guard
   382: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND esp_log_default_level
   383: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND esp_log_timestamp
   384: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND esp_rom_printf
   385: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND __stack_chk_fail
$ riscv32-esp-elf-readelf -sW build/esp-idf/esp_app_format/libesp_app_format.a | awk '{SUM+=$3;}END{print SUM;}'
840
```
So the actual size impact for this file should be 840 bytes and not 102529 byte as reported by the `size-files` tool. 
How can we trust the tool report then?

fhrbata, 2025-03-14:
Hello @X-Ryl669,

thank you for reporting. I believe this is expected and related to the linker relaxations(e.g. string deduplication and tail string elimination). There is a note about this in the [idf.py size documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-size.html) .

> The size of the .rodata section in the Flash Data memory type may appear very large for a single archive. This occurs due to linker relaxations. The linker may attempt to combine object file sections with MERGE and STRINGS flags from all archives into one to perform tail string optimization. Consequently, one archive may end up with a very large .rodata section, containing string literals from other archives. This is evident in the .rodata section of the libesp_app_format.a archive. The specific compiler behavior here can be turned off by enabling CONFIG_COMPILER_NO_MERGE_CONSTANTS option (only for GCC toolchain), please read help for more details.

X-Ryl669, 2025-03-15:
Ok, that's likely the case. I'll disable it to have a better understanding of where the size is spent. Thanks!

closed 2025-03-15 by X-Ryl669