espressif/esp-idf #15577: Request to Update /components/openthread/openthread Submodule to the Latest Version

aaa147147, 2025-03-15:
### Is your feature request related to a problem?

**Dear Community Maintainers,**

During recent development, I noticed that the /components/openthread/openthread submodule is significantly behind the latest version of the upstream repository. Here are the details:

**​Current Local Commit:**
005c5cefc22aaf0396e4327ee7f2e0ad32a7733b
(Commit Date: October 29, 2024)

**​Latest Remote Commit:**
c8fa6e3b8f7613665dba2d3b8874905eacc20b19
(Commit Date: March 13, 2025)

**​Commit Difference:**
The local version is ​278 commits behind the latest version.

**Issue Description**
I attempted to update the submodule to the latest version independently, but encountered errors during build.

**Request**
​Update the Submodule:
Request the community to update the /components/openthread/openthread submodule to the latest version (c8fa6e3b8f7613665dba2d3b8874905eacc20b19) and ensure compatibility with other ESP-IDF components.

**​Resolve Compatibility Issues:**
Since updating independently resulted in runtime errors, I kindly ask the community to test and resolve any potential compatibility issues to ensure the updated submodule works stably.

**Additional Notes**
Updating the /components/openthread/openthread submodule to the latest version will not only address known issues but also introduce new features and optimizations from the upstream, further enhancing the stability and performance of ESP-IDF. I hope the community can prioritize this request.

Thank you for your efforts and support! If you need any further clarification, please feel free to contact me.

Best regards,

zwx1995esp, 2025-03-17:
Hi, @aaa147147 thanks for your suggestion! We've already prepared the upstream update MRs in our local branch, but we encountered some issues. We've submitted a fix PR to the OpenThread repo, which is currently under review. Once it's merged, we'll run regression tests and merge the update MRs ASAP.

aaa147147, 2025-03-26:
@zwx1995esp Thank you for your reply. Is there any update now? Do you have an estimated time?

zwx1995esp, 2025-03-26:
Hi, the PR for the Openthread upstream has been merged and the regression tests are under testing, I think our internal MR will be ready in one or two weeks. And also need some time to sync our internal branch to Github.

chshu, 2025-04-24:
The submodule update has been synced to github, closing the issue.

BTW, we will trigger another update in mid-May.

closed 2025-04-24 by chshu