espressif/esp-idf #15578: [NimBLE][ESP32-S3] Unable to detect BLE Gatt Server in iOS / macOS settings app

Pr47h4m, 2025-03-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Everyone,

I have created BLE gatt server using Nimble Stack 
I am unable to discover device inside iOS / macOS settings app. However I am still able to discover device inside Android settings app

<details>
<summary>code</summary>

```
#include "host/ble_hs.h"
#include "host/ble_uuid.h"
#include "host/util/util.h"
#include "nimble/nimble_port.h"
#include "nimble/nimble_port_freertos.h"
#include "services/gap/ble_svc_gap.h"

#include "ble_app.h"
#include "settings.h"
#include "ble_prph.h"
#include "utils.h"
#include "sim7x00.h"

static const char TAG[] = "ble_app";

static uint8_t own_addr_type;
static uint16_t conn_handle;

static QueueHandle_t events_queue;

void ble_store_config_init(void);

static void ble_app_advertise(void);

static int gap_event(struct ble_gap_event *event, void *arg);

void host_task(void *param) {
    ESP_LOGI(TAG, "start host tsk");
    /* This function will return only when nimble_port_stop() is executed */
    nimble_port_run();

    nimble_port_freertos_deinit();
}

static void on_sync() {
    rc_chk(ble_svc_gap_device_name_set(settings.device_name), "ble_svc_gap_device_name_set", ble_err_to_name);
    rc_chk(ble_svc_gap_device_appearance_set(0x0341), "ble_svc_gap_device_appearance_set", ble_err_to_name);
    rc_chk(ble_hs_util_ensure_addr(0), "ble_hs_util_ensure_addr", ble_err_to_name);
    rc_chk(ble_hs_id_infer_auto(0, &own_addr_type), "ble_hs_id_infer_auto", ble_err_to_name);

#ifdef USE_2M_PHY
    uint8_t prefered_phy = BLE_HCI_LE_PHY_2M_PREF_MASK; // | BLE_HCI_LE_PHY_1M_PREF_MASK | BLE_HCI_LE_PHY_CODED_PREF_MASK;
    rc_chk(ble_gap_set_prefered_default_le_phy(prefered_phy, prefered_phy), "ble_gap_set_prefered_default_le_phy",
           ble_err_to_name);
#endif

    ble_app_advertise();
}

static void on_reset(int reason) {
    ESP_LOGI(TAG, "on_rst: %d", reason);
}

[... 208 lines cut ...]

    ble_hs_cfg.reset_cb = on_reset;
    ble_hs_cfg.sync_cb = on_sync;
    ble_hs_cfg.gatts_register_cb = gatt_svr_register_cb;
    ble_hs_cfg.store_status_cb = ble_store_util_status_rr;
    ble_hs_cfg.sm_io_cap = BLE_SM_IO_CAP_NO_IO;
    ble_hs_cfg.sm_sc = 1;
    ble_hs_cfg.sm_bonding = 1;
    ble_hs_cfg.sm_our_key_dist |= BLE_SM_PAIR_KEY_DIST_ENC;
    ble_hs_cfg.sm_their_key_dist |= BLE_SM_PAIR_KEY_DIST_ENC;

    ble_store_config_init();

    nimble_port_freertos_init(host_task);
}

void ble_app_send(msg_t *msg) {
    gatt_svr_notify(conn_handle, msg);
}

QueueHandle_t ble_app_events_queue(void) {
    return events_queue;
}
```
</details>

rahult-github, 2025-03-15:
Hi @Pr47h4m ,

There exists a  bleprph example under examples/bluetooth/nimble folder. You can refer to it and check your application

Pr47h4m, 2025-03-17:
Hi @rahult-github,

Thank you for replying

I ran bleprph example project in ESP32 and ESP32-S3 (N8R8)
 
Peripheral name is discovered and shown in Bluetooth settings in android
However, it is not shown in Bluetooth settings in iOS / macOS

I am able to discover it using NRF Connect or swift code.

rahult-github, 2025-03-17:
Hi @Pr47h4m ,

This is iphone behaviour. Per apple [support](https://forums.developer.apple.com/thread/71627),  this is how iphone behaves. 

You need other apps on iphone to find ble devices ( e.g. nrfconnect , that works for you too) .

Pr47h4m, 2025-03-17:
> Hi [@Pr47h4m](https://github.com/Pr47h4m) ,
> 
> This is iphone behaviour. Per apple [support](https://forums.developer.apple.com/thread/71627), this is how iphone behaves.
> 
> You need other apps on iphone to find ble devices ( e.g. nrfconnect , that works for you too) .

Thanks for the clarification!

I was wondering if it’s possible to configure the advertisement parameters in ESP-IDF to meet Apple’s requirements, so the device can be discovered and displayed in the iOS Bluetooth settings app. For example, nRF Bluetooth modules are visible in both Android and iOS Bluetooth settings, so I’m curious if we can achieve the same behavior with ESP devices.

Thank you for your quick help!

Pr47h4m, 2025-03-17:
Since other Bluetooth modules (like nRF) are visible in the iOS Bluetooth settings app, would it be okay to keep this issue open? This way, we can investigate their advertisement patterns or events to understand how they achieve this and possibly implement similar functionality in ESP-IDF.

rahult-github, 2025-03-17:
Hi @Pr47h4m  , is nRF module BLE only ?  If yes, can you help share one OTA or log that outlines what adv data is being advertised ?

Pr47h4m, 2025-03-17:
> Hi [@Pr47h4m](https://github.com/Pr47h4m) , is nRF module BLE only ? If yes, can you help share one OTA or log that outlines what adv data is being advertised ?

Yes, nRF module is BLE only

[image]
<br />
iOS Settings App
<br />
<br />

[image]
<br />
NRF Connect App
<br />
<br />

[image]
<br />
NRF Connect App
<br />
<br />

<details>
<summary>NRF Connect Logs</summary>

```
[17:32:47.8780] Normal: -44 dBm
[17:32:47.8780] Normal: 3919.30 ms
[17:32:48.9130] Verbose: Connecting...
[17:32:48.9130] Debug: cbCentralManager.connect()
[17:32:49.3530] Normal: -42 dBm
[17:32:49.3530] Normal: 1474.76 ms
[17:32:49.6220] Debug: [Callback] centralManager(central, didConnect: peripheral)
[17:32:49.6220] Normal: Connected.
[17:32:49.6220] Verbose: Discovering Services...
[17:32:49.6220] Debug: peripheral.discoverServices(nil)
[17:32:49.8010] Debug: [Callback] peripheral(peripheral, didDiscoverServices: nil)
[17:32:49.8010] Normal: Discovered Nordic UART Service, Secure DFU Service, Device Information, and Battery Service Services.
[17:32:49.8010] Verbose: Discovering Characteristics for Nordic UART Service...
[17:32:49.8010] Debug: peripheral.discoverCharacteristics(nil, for: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E)
[17:32:49.8010] Verbose: Discovering Characteristics for Secure DFU Service...
[17:32:49.8010] Debug: peripheral.discoverCharacteristics(nil, for: FE59)
[17:32:49.8010] Verbose: Discovering Characteristics for Device Information...
[17:32:49.8010] Debug: peripheral.discoverCharacteristics(nil, for: 180A)
[17:32:49.8010] Verbose: Discovering Characteristics for Battery Service...
[17:32:49.8010] Debug: peripheral.discoverCharacteristics(nil, for: 180F)
[17:32:49.8330] Debug: [Callback] peripheral(peripheral, didDiscoverCharacteristicsFor: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E, error: nil)
[17:32:49.8330] Verbose: Discovering Descriptors for Characteristic UART RX Characteristic...
[17:32:49.8330] Debug: peripheral.discoverDescriptors(for: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E)
[17:32:49.8330] Verbose: Discovering Descriptors for Characteristic UART TX Characteristic...
[17:32:49.8330] Debug: peripheral.discoverDescriptors(for: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E)
[17:32:49.8340] Normal: Discovered UART RX Characteristic and UART TX Characteristic Characteristics for Service Nordic UART Service.
[17:32:49.8620] Debug: [Callback] peripheral(peripheral, didDiscoverCharacteristicsFor: FE59, error: nil)
[17:32:49.8620] Verbose: Discovering Descriptors for Characteristic Buttonless DFU Without Bonds...
[17:32:49.8620] Debug: peripheral.discoverDescriptors(for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
[17:32:49.8630] Normal: Discovered Buttonless DFU Without Bonds Characteristics for Service Secure DFU Service.
[17:32:49.8920] Debug: [Callback] peripheral(peripheral, didDiscoverCharacteristicsFor: 180A, error: nil)
[17:32:49.8920] Verbose: Discovering Descriptors for Characteristic Manufacturer Name String...
[17:32:49.8920] Debug: peripheral.discoverDescriptors(for: 2A29)
[17:32:49.8920] Verbose: Discovering Descriptors for Characteristic PnP ID...
[17:32:49.8920] Debug: peripheral.discoverDescriptors(for: 2A50)
[17:32:49.8930] Normal: Discovered Manufacturer Name String and PnP ID Characteristics for Service Device Information.
[17:32:49.9520] Debug: [Callback] peripheral(peripheral, didDiscoverCharacteristicsFor: 180F, error: nil)
[17:32:49.9520] Verbose: Discovering Descriptors for Characteristic Battery Level...
[17:32:49.9520] Debug: peripheral.discoverDescriptors(for: 2A19)
[17:32:49.9530] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, error: nil)
[17:32:49.9530] Normal: Discovered Battery Level Characteristics for Service Battery Service.
[17:32:49.9530] Normal: UART RX Characteristic has no Descriptors.
[17:32:49.9820] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, error: nil)
[17:32:49.9830] Normal: Discovered Client Characteristic Configuration Descriptors for Characteristic UART TX Characteristic
[17:32:50.0120] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: 8EC90003-F315-4F60-9FB8-838830DAEA50, error: nil)
[17:32:50.0120] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: 2A29, error: nil)
[17:32:50.0130] Normal: Discovered Client Characteristic Configuration Descriptors for Characteristic Buttonless DFU Without Bonds
[17:32:50.0130] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: 2A50, error: nil)
[17:32:50.0130] Normal: Manufacturer Name String has no Descriptors.
[17:32:50.0140] Normal: PnP ID has no Descriptors.
[17:32:50.0720] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: 2A19, error: nil)
[17:32:50.0730] Normal: Discovered Client Characteristic Configuration Descriptors for Characteristic Battery Level
[17:33:07.3340] Normal: Scanner Off.
```
</details>

rahult-github, 2025-03-18:
Hi @Pr47h4m ,  the device is showing up only if HID information is present.  I suggest to refer to [this](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_hid_device/main/esp_hid_gap.c#L749) where HID uuid is included in the adv data. 

But again, this will show up only as HID device.

Pr47h4m, 2025-03-20:
> > Hi [@Pr47h4m](https://github.com/Pr47h4m) , the device is showing up only if HID information is present. I suggest to refer to [this](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_hid_device/main/esp_hid_gap.c#L749) where HID uuid is included in the adv data.
> > But again, this will show up only as HID device.
> 
> Hi [@rahult-github](https://github.com/rahult-github)
> 
> To ensure discoverability while still using custom services, I’ll include the HID information in the advertisement. Then, I can proceed with communication using custom service characteristics (read, write, and notify), similar to the existing nRF module device I shared, which also communicates over custom characteristics despite being discoverable.
> 
> I will try this and update here.
> 
> Thanks

Hello,

I will require some more time to implement and test... 

Thanks

closed 2025-03-24 by Pr47h4m