espressif/esp-idf #15581: undefined reference to nd6_get_next_hop_addr_or_queue

dannybackx, 2025-03-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Compile correctly

### What is the actual behavior?

`
/home/danny/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lwip/liblwip.a(ethip6.c.obj):(.literal.ethip6_output+0x8): undefined reference to `nd6_get_next_hop_addr_or_queue'
/home/danny/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/lwip/liblwip.a(ethip6.c.obj): in function `ethip6_output':
/home/danny/src/github/esp32/esp-idf-v5.4/components/lwip/lwip/src/core/ipv6/ethip6.c:108:(.text.ethip6_output+0x53): undefined reference to `nd6_get_next_hop_addr_or_queue'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
`

### Steps to reproduce.

Compile your app with or without CONFIG_LWIP_ND6

'
victus: {2089} diff -c sdkconfig~ sdkconfig.spotpear | grep ND6
! CONFIG_LWIP_ND6=y
! # CONFIG_LWIP_ND6 is not set
victus: {2090} 
'

### Build or installation Logs.

```plain

```

espressif-abhikroy, 2025-03-24:
@dannybackx Thanks for reporting this issue! I tried to reproduce it with the provided configuration, but the application compiles successfully on my end. Could you please provide more details on any configuration changes in your sdkconfig?
Could you attach your entire sdkconfig file?
If possible, run the following command and share the output to highlight any differences:

`diff -u sdkconfig sdkconfig.old`

Have there been any other modifications to ESP-IDF or the lwip component?
This will help me investigate the issue further. Thanks!

espressif-abhikroy, 2025-03-25:
Got it. Thanks for the information. It seems that you are disabling `CONFIG_LWIP_ND6` while `CONFIG_LWIP_IPV6` is enabled. As a result, the compiler does not compile `nd6.c`, which contains the `nd6_get_next_hop_addr_or_queue()` function. This condition needs to be handled in `ethip6.c`. I will create a fix to address this issue. However, as a temporary workaround, please ensure that `CONFIG_LWIP_ND6` remains enabled if you are using IPv6.

espressif-abhikroy, 2025-03-27:
@dannybackx Also, could you clarify the specific scenario in which you want to use IPv6 without ND6 in an Ethernet network? Understanding the use case will help ensure that the fix aligns with your requirements.

dannybackx, 2025-03-27:
I don't have a use scenario. I turned it off and got build errors, this shouldn't happen.
I'm not sure if there is a use case for IPv6 with ND6 so I'd suggest not making it possible to do this.