espressif/esp-idf #15583: esp_driver_i2c: i2c_master_multi_buffer_transmit causes array access out of bound

mzx2689, 2025-03-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.3)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

-

### Power Supply used.

USB

### What is the expected behavior?

-

### What is the actual behavior?

In function `i2c_master_multi_buffer_transmit`:

https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/components/esp_driver_i2c/i2c_master.c#L1160

The `array_size` check is ok when it **= 5**. (`SOC_I2C_CMD_REG_NUM` = 8 for ESP32-C3)
It will create an array `i2c_ops` (size = 7 in this situation). Then it will call `s_i2c_synchronous_transaction`.

In function `s_i2c_synchronous_transaction`:

https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/components/esp_driver_i2c/i2c_master.c#L906

It will copy the array into `i2c_dev->master_bus->i2c_ops` (size **= 6**). This causes array access out of bound.

https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/components/esp_driver_i2c/i2c_private.h#L64
https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/components/esp_driver_i2c/i2c_private.h#L144

### Steps to reproduce.

1. New an I2C master bus with `i2c_new_master_bus` and add a device with `i2c_master_bus_add_device`.
2. Create a `buffer_info_array` (size = 5) and call `s_i2c_synchronous_transaction`.

```c
    uint8_t data[1];
    i2c_master_transmit_multi_buffer_info_t buffer_info[5] = {
        {.write_buffer = data, .buffer_size = 1},
        {.write_buffer = data, .buffer_size = 1},
        {.write_buffer = data, .buffer_size = 1},
        {.write_buffer = data, .buffer_size = 1},
        {.write_buffer = data, .buffer_size = 1},
    };
    i2c_master_multi_buffer_transmit(dev_handle, buffer_info, 5, -1);

```

### Debug Logs.

```plain

```

BitsForPeople, 2025-03-16:
So the issue is that i2c_master_multi_buffer_transmit() can use up to (SOC_I2C_CMD_REG_NUM - 2) entries, plus 2 for START+STOP, which is equal to SOC_I2C_CMD_REG_NUM, but I2C_STATIC_OPERATION_ARRAY_MAX is less than SOC_I2C_CMD_REG_NUM?

Seems like the fix would be to just define I2C_STATIC_OPERATION_ARRAY_MAX to SOC_I2C_CMD_REG_NUM.

ginkgm, 2025-03-17:
Thanks for reporting. Will look into this issue

mythbuster5, 2025-03-17:
Correct, I forget why I set this to 6. Will fix this soon,.

mzx2689, 2025-03-17:
> Correct, I forget why I set this to 6. Will fix this soon,.

I think it's because `i2c_master_transmit_receive` uses 6 i2c operations and it uses the most operations when `I2C_STATIC_OPERATION_ARRAY_MAX` was defined.

closed 2025-03-24 by espressif-bot