espressif/esp-idf #15586: NVS: what happening here

diplfranzhoepfinger, 2025-03-17:
```
VT Instance: 128   LoginState:  70 
I (14427) main: Core id: 0, prio: 1,  free heap: 5000436 bytes., minimum free heap: 4990272 bytes., minimum free stack: 1528 bytes.
SOFTKEY ACTIVATION: 0x1388       1            1   Time:    19910
SOFTKEY ACTIVATION: 0x1388       1            0   Time:    20031
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556d5  PS      : 0x00060130  A0      : 0x8211e07f  A1      : 0x3fce48f0  
--- 0x400556d5: r_rwbt_isr in ROM

A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x00000001  A12     : 0xffffffc2  A13     : 0x00000004  
A14     : 0x25d3906e  A15     : 0x3fce4910  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
--- 0x400556d5: r_rwbt_isr in ROM
0x400556e5: r_rwbt_isr in ROM

Backtrace: 0x400556d2:0x3fce48f0 0x4211e07c:0x3fce4900 0x4211cc7a:0x3fce4970 0x4211d2a3:0x3fce49e0 0x4211d6c1:0x3fce4a00 0x421d3d3f:0x3fce4a30 0x420adab5:0x3fce4a60 0x420ade4b:0x3fce4ab0 0x4203fe73:0x3fce4ad0 0x4203fe80:0x3fce4af0 0x420dcffb:0x3fce4b10
--- 0x400556d2: r_rwbt_isr in ROM
0x4211e07c: nvs::Page::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char) at /home/franz/esp-idf-v5.4/components/nvs_flash/src/nvs_page.cpp:173
0x4211cc7a: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /home/franz/esp-idf-v5.4/components/nvs_flash/src/nvs_storage.cpp:475
 (inlined by) nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /home/franz/esp-idf-v5.4/components/nvs_flash/src/nvs_storage.cpp:375
0x4211d2a3: nvs::NVSHandleSimple::set_typed_item(nvs::ItemType, char const*, void const*, unsigned int) at /home/franz/esp-idf-v5.4/components/nvs_flash/src/nvs_handle_simple.cpp:21
 (inlined by) nvs::NVSHandleSimple::set_typed_item(nvs::ItemType, char const*, void const*, unsigned int) at /home/franz/esp-idf-v5.4/components/nvs_flash/src/nvs_handle_simple.cpp:16
0x4211d6c1: nvs::NVSHandleLocked::set_typed_item(nvs::ItemType, char const*, void const*, unsigned int) at /home/franz/esp-idf-v5.4/components/nvs_flash/src/nvs_handle_locked.cpp:70
0x421d3d3f: int nvs::NVSHandle::set_item<unsigned long>(char const*, unsigned long) at ??:?
0x420adab5: FORTE_NVS::executeSet() at ??:?
0x420ade4b: FORTE_NVS::executeEvent(unsigned int, CEventChainExecutionThread*) at ??:?
0x4203fe73: CEventChainExecutionThread::mainRun() at ??:?
0x4203fe80: CEventChainExecutionThread::run() at ??:?
0x420dcffb: CFreeRTOSThread::threadFunction(void*) at ??:?

ELF file SHA256: cbfbd241b

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038050a
--- 0x4038050a: esp_cpu_wait_for_intr at /home/franz/esp-idf-v5.4/components/esp_hw_support/cpu.c:64

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2810,len:0x1658
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe78
load:0x403cb700,len:0x2f7c
entry 0x403c8958
```

adokitkat, 2025-03-20:
Hello. Could you please provide the code which caused this? Or even better a reproducible example? 

cc @rrtandler

diplfranzhoepfinger, 2025-03-20:
It is solved. 
Therefore I referenced the other issue in 4diac. Will follow-up there. 

Espressif side: would be good to have a better Error Handling if Handle is not initialized. 

And I can provide a minimum Example yes.

diplfranzhoepfinger, 2025-03-20:
> Could you try to enable this setting `CONFIG_SPIRAM_RODATA` in `idf.py menuconfig` to see if it makes a difference? (https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/external-ram.html#move-read-only-data-in-flash-to-psram)

This is enabled.

adokitkat, 2025-04-03:
Hello. Can you confirm your problem was solved? What was the problem?

diplfranzhoepfinger, 2025-04-03:
only the "handle" was not initialized.

closed 2025-04-03 by diplfranzhoepfinger